No0ne/ps2x2pico #1: Mouse Issues

serisman, 2022-11-10:
First of all, awesome project!

It was super easy to get a Logitech wireless keyboard/mouse (K345/M275) combo up and running on an old DOS PC with only PS2 ports.

I haven't found any issues with the keyboard at all, but the mouse code seems buggy.

I am testing with both the CTMOUSE.EXE driver as well as Microsoft MOUSE.EXE driver.  With both drivers, by default, the mouse cursor jumps around pretty randomly and it seems like button presses are being toggled without actually pressing anything.  No amount of tweaking the driver's resolution seems to help things.  Eventually, I found that the /O switch on CTMOUSE (which enables wheel support) seems to work... about half the time.  Other times it just reports 'Error: device not found".

I think what is happening is that this code is always reporting 'Mouse with scroll wheel' and sending 4-byte 'Intellimouse' data packets, even when the computer hasn't asked for them (yet).  If I'm reading the documentation at [()](https://wiki.osdev.org/PS/2_Mouse) correctly, it seems like by default the mouse is supposed to report back type 0 'Standard PS/2 mouse' and send 3-byte data packets until a specific initialization sequence is requested to put the mouse into enhanced mode at which point it will report back type 3 'Mouse with scroll wheel' and send 4-byte data packets.

serisman, 2022-11-10:
I did some further testing, and my assumption was indeed true.  I made some code changes and now it works much better.  I'll clean it up and submit a pull request sometime tomorrow for you to review.

No0ne, 2022-11-10:
Hi, thanks for your feedback!
Yes you are right, the mouse support is still very sparse.

I've currently only tested it on Windows 2000 as there are still some issues left. I need to port the PS/2 protocol part to the Pico PIOs so that both channels can send/receive independently. At the moment it is just a port from [/No0ne/ps2pico] where CPU based bitbanging was enough since theres only the keyboard ps/2 channel. In my other repo [/No0ne/ps2x2pico-rs] I already tried PIO support with Rust, but there USB support is more of a problem.

I also found out that all my USB mice with the PixArt chip still do native PS/2 using the green passive adapter!
E.g. the official Raspberry Pi Mouse or the Microsoft Basic Optical Mouse

I hope I'll finish the PIO implementation this year.

serisman, 2022-11-11:
I'm using this with a Logitech wireless keyboard/mouse receiver, so those green passive adapters won't work for me.  :(  That's why I was excited to find your project to begin with.

I just submitted Pull Request #2 with some better mouse code for you to review.  It still isn't a 100% complete mouse driver, but does the main stuff better than before (i.e. switching between the different mouse types and honoring the corresponding data packet format).

I have tested this in MS DOS with the CTMOUSE.EXE and MS MOUSE.EXE drivers as well as with Windows 3.11, and all seems well.  I did run into a weird race condition when trying to run `ctmouse /o` where the 'key up' event of the ENTER key from running the command takes time away from the mouse initialization and causes weird issues.  If I just use `ctmouse` or put the command in a batch file with a pre-delay, everything works correctly.  I suppose what you mention about using PIO is probably the real solution that is needed.

No0ne, 2022-11-11:
Thanks for your PR!
Quick sidenote: I did compile the UF2s with Pico-SDK develop branch because of this https://github.com/raspberrypi/pico-sdk/issues/483
Will do a new release with your enhancements soon!

serisman, 2022-11-11:
Yeah, I had to do something similar.  I think I just updated the tinyusb git submodule to the latest.  Without doing that, only the keyboard was working.  Took a while to track that one down.  Finally realized I could get some debugging info by connecting a USB TTL serial to the UART!

serisman, 2022-12-05:
That looks pretty promising!  I'll have to check it out in more detail later.

I had one other thought for optimization, although I'm not sure how useful it would be once the PIO work is done.  Anyway, might be helpful to have all the serial logging controlled by a debug flag and produce two different binaries (one for run-time, one for debug).  The serial output is very useful for debugging, but it does steal some time from the PS/2 code.  During normal operation it seems kinda wasteful for all that serial processing to continue to happen.  It's like shouting into the void with no one listening.

No0ne, 2022-12-07:
Please try the new binary from the pio-testing branch: https://github.com/No0ne/ps2x2pico/raw/pio-testing/ps2x2pico.uf2
I've only tested it on Windows 2000 so far, but there still seams to be an issue if you drag the typematic sliders in the keyboard preference panel. The mouse starts jumping vertically. Also the PIO code is still messy right now.

serisman, 2022-12-16:
@No0ne - I finally had a chance to do some testing of your pio-testing branch, and everything seems to be working fine on my side.  I did some testing with both MS-DOS 6.22 as well as Windows 3.11 and didn't notice any issues.  Good job!

No0ne, 2022-12-17:
Cool! I still have issues on Windows 2000, especially when changing key repeat settings in the system control panel. I think I need to capture some data between real PS/2 devices and different PCs and mimic the timings as close as possible.

No0ne, 2022-12-18:
The PIO code is much more refined right now, but still not working 100% of the time.
Please try if this is still happening with the latest build [5f96329]:
> I did run into a weird race condition when trying to run `ctmouse /o` where the 'key up' event of the ENTER key from running the command takes time away from the mouse initialization and causes weird issues.

Thanks!

serisman, 2022-12-23:
> The PIO code is much more refined right now, but still not working 100% of the time. Please try if this is still happening with the latest build [...]

Sorry, but with the current version the mouse isn't working at all.  It works again if I rollback to the Dec 7th version as long as I have ctmouse in the autoexec.bat, but not if I just type it on the command line.

nanoant, 2023-08-20:
@No0ne @serisman Can you guys share Dec 7th version? (the link above is broken) The latest beta 0.6 does not work at all on my Toshiba 430CDT using my build described at #16. I see lot of communications in oscilloscope, but in the end laptop is not taking any external mouse events.

No0ne, 2023-08-20:
I'm currently rewriting the whole PIO code as part of integrating to the PiKVM pico hid project. I hope this will fix all mouse related issues, please give me some to finish.

No0ne, 2023-08-27:
@serisman if you still have the board around you could test the new pre-release: https://github.com/No0ne/ps2x2pico/releases/download/beta-0.7/ps2x2pico.uf2

serisman, 2023-08-30:
@No0ne - I've mostly moved on to the HIDman-mini / HIDman-micro boards that I designed (https://github.com/serisman/HIDman-mini), but yes, I do still have a ps2x2pico board around I could do some tests with.  Can you remind me what pins need to be swapped to try the new firmware?

No0ne, 2023-08-30:
Cool, please see https://github.com/No0ne/ps2x2pico/releases/tag/beta-0.7

serisman, 2023-08-30:
Ok, I've done some additional testing with DOS 6.22 (CTMouse driver) and WFW 3.11, and overall it looks pretty good.  My original mouse issues seem to be resolved!

I did notice two potential issues, although I have no idea how I got into this state.
1. Once, when exiting Windows 3.11, the keyboard stopped working until I hard rebooted.
2. Once, when messing with the CTMouse driver, the system got into a state where it wouldn't recognize the mouse at all until I hard rebooted.

The only other thing I noticed is that the mouse acceleration seems pretty high.  Not sure if that can be dialed down at all?

No0ne, 2023-08-30:
Cool! Finally the PIO version makes progress over CPU bitbanging =)

I'll WFW 3.11 my self, hope I can reproduce it.

I believe the problem is the mouse rate. Currently it sends mouse packets as fast as possible/as fast as the usb reports come in. I'll have to implement proper mouse polling rates, maybe this fixes the acceleration too?

nanoant, 2023-09-02:
Hi @No0ne. I confirm that 0.7 works also on my Toshiba 430CDT, sort of...

1. The ps2x2pico mouse (cursor) in Windows 98 SE is laggy comparing to PS/2 cable mouse.
2. After a while of using ps2x2pico, mouse breaks and starts sending random clicks and cursor jumps over the screen as if Toshiba started receiving random garbage on PS/2.

Anything I can possibly do to help here? I have couple of different mice and portable oscilloscope.

[image]

No0ne, 2023-09-03:
Thanks for testing! I think its almost certainly the polling rate issue. Should soon be fixed with beta-0.8, please test again after release.

nanoant, 2023-09-09:
Hi @No0ne. I tested 0.8 today and it is almost perfect:
1. No lag observed anymore
2. The mouse craziness (random cursor jumps and clicks) is still there, but it is rare, occurs mostly when I try to use mouse after long pause (e.g. working with keyboard for 5 minutes, then trying to do something with mouse) and goes away (previously in 0.7 it didn't)

Thanks for this great work. Let me know if I can support you anyhow. Best, Adam.

No0ne, 2023-09-10:
Cool, thanks for testing again! I think I'm getting there, but its a lot harder than i thought 😅 Will do further refinement, afterwards please test the new release.

seritools, 2023-12-10:
Hi there,

thank you for this awesome project! I've build it up as well and have been testing.

Setup:

- i440BX motherboard with a Pentium 3, Win98SE
- WASD Keyboard, Logitech G502 mouse

I've encountered these issues so far while testing with beta-0.8:

- if the mouse is set to a polling rate of more than 125 Hz, the cursor is laggy and slower than at 125 Hz
- even at 125 Hz, pressing keys at the same time as moving the mouse causes the system to lock up

Please let me know if you'd like more information or want me to test something!

No0ne, 2023-12-11:
Yeah already discoverd this problem my self. Currently the USB stack directly starts the PS/2 transmit process and I'll need to decouple this and sum up intermediate movements. I hope I have some time for this after christmas. I'll let you know if theres something new to test.

No0ne, 2024-01-27:
[ps2x2pico-locking-test.zip](https://github.com/No0ne/ps2x2pico/files/14074183/ps2x2pico-locking-test.zip)

@nanoant could you please retest with this version an report if the issue 2. mouse craziness is fix?
Thank you!

nanoant, 2024-01-28:
> @nanoant could you please retest with this version an report if the issue "2. mouse craziness" is fix? Thank you!

Hi. Unfortunately after around 5-10 of perfectly stable behavior the mouse started jumping around generating random clicks, same Toshiba 430CDT setup as before.

No0ne, 2024-02-17:
While implementing https://github.com/pikvm/pikvm/issues/476 I discovered I can at least reproduce the "mouse jumping around" now if I move the KVM mouse and passthru mouse at the same time.

No0ne, 2024-03-02:
I've implemented a sample rate timer now.

@nanoant could you please retry with this:
https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-boot.uf2
https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-report.uf2

The first one runs the mouse in boot mode (as it always was).
The second one runs the mouse in report mode (I discovered my wheel only works in report mode).

No0ne, 2024-03-02:
@seritools please also try the two files above with more than 125 Hz, thanks!

seritools, 2024-03-02:
Tests with just the mouse:

`-report.uf2`:

Latency issue is gone, but moving the mouse to the right moves the cursor down, left moves it up, top/down don't work at all.

`-boot.uf2`:

Movement works without latency, however fast movements seem to cause the cursor to jump in the inverse direction, like an overflow in the variable used to sum up the movements? Let me know if you want a video showcasing the issue. With the G502 the mouse wheel seems to still work in boot mode.

Test with kb+mouse and `-boot.uf2`:

The freeze seems to be gone! 💜

seritools, 2024-03-02:
Ah yeah, maybe these need to be bigger than u8: https://github.com/No0ne/ps2x2pico/commit/0ca3e6403c90902dacea0244f2d82c2e9d8a08a4#diff-e25424adb57c069078b41d0ac8adbc71d89db4d8a01f32ab5164dd27e21ce164R37

No0ne, 2024-03-02:
Ah hm, this sounds like overflow yes. I'll have to cap it at 0xff since I can't send anything larger than 8 bits.

No0ne, 2024-03-05:
I think I finally fixed it! Did forget that PS/2 movements are 9-bit signed, so I did implement that extra bit. And if the accumulated values are larger/smaller than 255/-255 the remainder will get transmitted in the next mouse packet.

@seritools please try again, thanks! https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-boot.uf2

seritools, 2024-03-07:
Just tested again, seems to work flawlessly now ❤️

No0ne, 2024-03-07:
@nanoant if you can confirm please, I'll make the 1.0 release finally 🙏🏻💪🏻

No0ne, 2024-03-07:
@serisman if you could also test it, that would be nice: https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-boot.uf2

serisman, 2024-03-08:
Just did some quick testing on my hardware.

I had some issues where both the keyboard and mouse would completely freeze or not be recognized without a hard reboot.  Once or twice I had it happen after entering the BIOS.  One or twice the mouse wasn't recognized when the ctmouse driver tried to load it in autoexec.bat.  Once, they both froze up after re-entering Win 3.11 after entering/exiting it a few times.

But, as of right now, they are both working flawlessly both in DOS and also Win 3.11.  So, not sure if there is still a code problem, or just some weird issue with my hardware config.

Either way, don't hold up a release because of issues on my end.  I usually use my custom HIDman-micro board these days anyway.

serisman, 2024-03-08:
It just froze up again.  It was sitting at the DOS prompt while I typed that reply.  After I was done, I re-entered Win 3.11 and everything was working.  I exited to DOS again, and at least the keyboard was still working.  Then I re-entered Win 3.11 and both the keyboard and mouse are frozen.  No way to get them working again that I can think of without performing a hard reboot of the computer.  (rebooting the Pi Pico doesn't help).

nanoant, 2024-03-10:
@No0ne I am sorry but [ps2x2pico-boot.uf2](https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-boot.uf2) make mouse completely dead after Win98 Toshiba laptop boot or it emits completely random events on the other boot. With [ps2x2pico-report.uf2](https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-report.uf2) mouse moves only in one direction but mouse buttons function fine (8-bit overflow?).

[ps2x2pico-locking-test.zip](https://github.com/No0ne/ps2x2pico/files/14074183/ps2x2pico-locking-test.zip) works still much better albeit it starts to jump after a while (as previously reported).

Also in Win98 I neither see `Microsoft PS/2 Mouse` nor rate settings. I see something like `WheelMouseIM` device and `Toshiba` as manufacturer of the driver. But my physical PeriMice-209 PS/2 mouse works flawlessly there.

No0ne, 2024-03-15:
Hey everyone, if you could please test again that would be nice, thanks!
https://github.com/No0ne/ps2x2pico/releases/download/1.0/ps2x2pico.uf2

nvllsvm, 2024-03-16:
V1.0 alpha works well with my G305 @ 500hz, though I needed to use PS2Rate to change the rate in Windows 98. The default was very choppy compared to using the same mouse with the [Minicom PS2 to USB adapter](https://www.amazon.com/gp/product/B00404YD8W/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1). 

There also might be some small latency issues, though I'll need to test more.

No0ne, 2024-03-18:
Just checked and yes, Win98 default sets the rate to 0x28 = 40hz which means 25ms between transmits. I currently apply the sample rate requested by the host but if you want we could also hardcode it. Please explain if you have more details about stuttering, thanks!

[image]

seritools, 2024-03-24:
Just tested again with 1.0, works flawlessly over here. 🎉 

And wow, PS2Rate makes a huge difference, TIL Win9x sets it to such a low value. Maybe having a second FW version or a HW switch through another GPIO for forcing a sample rate might make sense?

nvllsvm, 2024-03-24:
I've been testing a modification which forces a constant sample rate without any issues. 
https://github.com/nvllsvm/ps2x2pico/commit/f867858986a37cafbccab6e7d057983de4874b4e

Both 100hz and 200hz are working well enough for me to comfortably replace the Minicom adapter I was using. Playing Unreal on Windows 98 feels fantastic and responsive.
 
[ps2x2pico_mouse_100hz.uf2.tar.gz](https://github.com/No0ne/ps2x2pico/files/14736769/ps2x2pico_mouse_100hz.uf2.tar.gz)
[ps2x2pico_mouse_200hz.uf2.tar.gz](https://github.com/No0ne/ps2x2pico/files/14736770/ps2x2pico_mouse_200hz.uf2.tar.gz)

No0ne, 2024-03-28:
Just checked that 200hz with mouse wheel enabled doesn't work without overclocking the ps/2 port.
The fourth byte is around the 5ms mark which the next first byte should be.
With a 2x overclock the transmission is still a bit choppy but with over 20kHz clock this exeeds the proposed max 16kHz.

[image]
[image]

No0ne, 2024-03-28:
Correction: it does start to work above around 14.5kHz so I could release a 200Hz version with a higher clock.

No0ne, 2024-03-28:
5628f15 is now optimized, 200Hz mouse movement should work now.
PS/2 clock is at 15.625kHz and theres a little bit more than one byte of time space between mouse packets 🥳

nvllsvm, 2024-03-28:
> [5628f15](https://github.com/No0ne/ps2x2pico/commit/5628f15e4999bbc766244dda96cc425ffdc37cec) is now optimized, 200Hz mouse movement should work now. PS/2 clock is at 15.625kHz and theres a little bit more than one byte of time space between mouse packets 🥳

I wasn't having any issues with the wheel previously, but I've confirmed this change works on my end (ABit AV8 motherboard).

ssokolow, 2024-03-28:
Once the pack of level shifters I've ordered arrives in about a month, I should be able to test on an HP t5530 thin client that's been repurposed as a Windows 98 SE machine, a cheap Chinese USB/VGA KVM switch that doubles as a hub, a Rosewill RK-9000I mechanical keyboard,  and a Logitech G203 Prodigy mouse.

(The thin client has USB but the aftermarket USB 2.0 drivers for Windows 98 SE crash with Fatal Error 0E if you unplug the USB keyboard or mouse while running and I want something I can KVM without having to find and pay for a KVM switch which emulates permanently connected screen, keyboard, and mouse to minimize switching latencies.)

No0ne, 2024-03-29:
> I wasn't having any issues with the wheel previously, but I've confirmed this change works on my end (ABit AV8 motherboard).

Thats true but it wasn't 200Hz either as it missed every other send slot. On the first screenshot https://github.com/No0ne/ps2x2pico/issues/1#issuecomment-2025216369 the marker 2 is on 5ms where the next mouse packet should have started. Raising the clock made each bit shorter to make more room.

nanoant, 2024-03-30:
I was testing [ps2x2pico_mouse_200hz.uf2.tar.gz](https://github.com/No0ne/ps2x2pico/files/14736770/ps2x2pico_mouse_200hz.uf2.tar.gz) attached previously by @nvllsvm on my 430CDT and it was almost perfect. It glitched once for a short while but recovered itself.

Does it correspond to [5628f15](https://github.com/No0ne/ps2x2pico/commit/5628f15e4999bbc766244dda96cc425ffdc37cec) with optimized 200Hz? Or shall I build the latest commit myself?

nvllsvm, 2024-03-30:
@nanoant It was for an older commit. Here's a build for https://github.com/No0ne/ps2x2pico/commit/5628f15e4999bbc766244dda96cc425ffdc37cec

[ps2x2pico_5628f15_200hz.uf2.tar.gz](https://github.com/No0ne/ps2x2pico/files/14813101/ps2x2pico_5628f15_200hz.uf2.tar.gz)

No0ne, 2024-04-19:
Here is the 1.0 release: https://github.com/No0ne/ps2x2pico/releases/tag/1.0

Thanks everyone in this thread for contributing.
Please open a new issue if you have further mouse releated issues/updates.

closed 2024-04-19 by No0ne