No0ne/ps2x2pico #3: Inverted Y-Axis

anderssonbilly, 2022-11-17:
Hi.

Just finished building the adapter and it all works great except I got some issues with the mouse.
My test system is a Pentium III WinME computer and a Razor DeathAdder USB mouse.
The keyboard works fine, but the mouse Y-Axis is inverted.

No0ne, 2022-12-04:
Only the Y-Axis is inverted and everything else correct?
Will address the "Mouse Issues #1" first and afterwards add DIP-switches or jumpers for inverting axis.

anderssonbilly, 2022-12-04:
With the mouse, yes, only the Y-Axis is inverted other than that it works great.
A DIP-switch to invert the axis could be really useful in some DOS games.
Curious thou why my mouse is inverted.

If there is anything I could do to help troubleshoot just tell me.

anderssonbilly, 2022-12-05:
`HID device address = 1, instance = 0 is mounted
HID Interface Protocol = Mouse
00 00 00 00
00 00 00 00
00 01 01 00
00 00 01 00
00 00 01 00
00 00 01 00
00 00 01 00
00 00 01 00
00 00 02 00
00 00 01 00
00 00 01 00
00 fd 02 00
00 00 01 00
00 00 01 00
00 ff 02 00
00 00 01 00
00 ff 02 00
00 00 01 00
00 00 01 00
00 00 02 00
00 00 01 00
00 00 01 00
00 00 01 00
00 ff 02 00
00 fe 01 00
00 00 01 00
00 00 01 00
00 ff 02 00
00 00 01 00
00 00 01 00
00 00 02 00
00 00 02 00
00 00 01 00
00 ff 01 00
00 00 02 00
00 00 02 00
00 ff 02 00
00 00 01 00
00 00 04 00
00 00 01 00
00 00 02 00
00 00 02 00
00 ff 02 00
00 00 02 00
00 00 02 00
00 02 02 00
00 01 01 00
00 00 03 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 03 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 01 00
00 00 02 00
00 00 02 00
00 00 02 00
00 00 01 00
00 00 01 00
00 00 02 00
00 fd 01 00
00 00 01 00
00 00 01 00
00 00 01 00
00 ff 00 00
00 00 fd 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 00 fe 00
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 00 ff 00
00 02 fe 00
00 00 fe 00
00 01 ff 00
00 00 ff 00
00 00 fe 00
00 01 ff 00
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 01 fe 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 01 ff 00
00 00 fe 00
00 00 ff 00
00 00 ff 00
00 01 00 00
00 00 fe 00
00 00 ff 00
00 00 ff 00
00 01 fe 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 01 ff 00
00 00 fe 00
00 00 ff 00
00 01 fe 00
00 00 ff 00
00 00 ff 00
00 01 fe 00
00 00 fe 00
00 00 ff 00
00 01 fe 00
00 00 fe 00
00 00 fe 00
00 00 fe 00
00 00 ff 00
00 01 fe 00
00 00 fe 00
00 01 fe 00
00 00 fe 00
00 00 fe 00
00 01 ff 00
00 00 fe 00
00 00 ff 00
00 02 fe 00
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 00 ff 00
00 00 fe 00
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 00 ff 00
00 01 fe 00
00 00 fe 00
00 00 ff 00
00 01 fe 00
00 00 ff 00
00 00 fe 00
00 01 ff 00
00 00 fe 00
00 00 ff 00
00 01 fe 00
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 00 fe 00
00 00 ff 00
00 01 ff 00
00 00 ff 00
00 00 fe 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 fd 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 ff 00 00
00 02 00 00
00 01 00 00
00 01 00 00
00 02 00 00
00 01 00 00
00 01 00 00
00 01 00 00
00 04 00 00
00 00 02 00
00 01 00 00
00 01 00 00
00 01 01 00
00 01 00 00
00 02 00 00
00 01 00 00
00 00 01 00
00 01 00 00
00 01 00 00
00 00 01 00
00 02 00 00
00 01 01 00
00 01 00 00
00 01 01 00
00 02 01 00
00 01 00 00
00 01 00 00
00 01 01 00
00 01 01 00
00 01 01 00
00 02 00 00
00 01 01 00
00 01 00 00
00 02 01 00
00 02 00 00
00 00 01 00
00 02 00 00
00 01 01 00
00 02 01 00
00 01 01 00
00 03 01 00
00 00 01 00
00 02 01 00
00 02 01 00
00 01 01 00
00 02 00 00
00 02 01 00
00 00 01 00
00 03 01 00
00 02 01 00
00 02 00 00
00 02 01 00
00 02 01 00
00 02 01 00
00 01 01 00
00 02 00 00
00 02 01 00
00 02 01 00
00 02 01 00
00 01 00 00
00 00 01 00
00 02 01 00
00 02 00 00
00 01 01 00
00 02 00 00
00 02 01 00
00 00 01 00
00 03 00 00
00 01 00 00
00 01 01 00
00 02 01 00
00 01 00 00
00 01 00 00
00 00 01 00
00 02 00 00
00 01 01 00
00 02 00 00
00 01 00 00
00 02 01 00
00 01 00 00
00 01 00 00
00 02 01 00
00 01 00 00
00 00 01 00
00 01 00 00
00 00 01 00
00 02 00 00
00 01 01 00
00 01 00 00
00 01 00 00
00 01 00 00
00 01 01 00
00 01 01 00
00 01 00 00
00 01 00 00
00 00 01 00
00 01 00 00
00 01 00 00
00 01 00 00
00 00 01 00
00 00 01 00
00 01 00 00
00 01 00 00
00 01 01 00
00 01 00 00
00 01 00 00
00 00 01 00
00 00 01 00
00 02 00 00
00 01 01 00
00 01 00 00
00 00 01 00
00 01 00 00
00 01 00 00
00 01 01 00
00 01 00 00
00 01 00 00
00 00 01 00
00 01 01 00
00 01 00 00
00 01 01 00
00 01 00 00
00 00 01 00
00 01 00 00
00 01 00 00
00 01 00 00
00 01 00 00
00 00 01 00
00 01 00 00
`

Here I got some data from my Razor mouse.
I moved up then down, left and right.

serisman, 2022-12-05:
> Sorry for no newline, how do I markdown that in github? Thought it was four spaces.

Triple backticks

https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

anderssonbilly, 2022-12-06:
I used the latest release but I'll try to compile latest my self and report the result as soon as I got some time over.

No0ne, 2022-12-07:
@serisman The additional info is missing because the mouse port was not in streaming mode.

No0ne, 2022-12-14:
hm @anderssonbilly your mouse is indeed repoting the y-axis inverted 🤔
here is mine:
```
UP
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 01 ff 00
00 00 ff 00
00 00 ff 00
00 00 fe 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 00 ff 00
00 01 ff 00
00 00 ff 00
```

```
DOWN
00 00 01 00
00 00 01 00
00 00 02 00
00 00 02 00
00 00 02 00
00 01 03 00
00 01 03 00
00 01 03 00
00 00 03 00
00 01 04 00
00 00 04 00
00 01 04 00
00 00 04 00
00 01 04 00
00 01 03 00
00 00 02 00
```

```
LEFT
00 fd 00 00
00 fd 00 00
00 fd 00 00
00 fd 00 00
00 fe 00 00
00 fe 00 00
00 fe 00 00
00 fe 00 00
00 fe 00 00
00 fe 00 00
00 ff 00 00
00 ff 00 00
```

```
RIGHT
00 01 00 00
00 01 00 00
00 01 00 00
00 02 01 00
00 02 00 00
00 02 00 00
00 02 01 00
00 02 00 00
00 02 00 00
00 03 01 00
00 03 00 00
00 03 01 00
00 03 00 00
00 04 01 00
00 03 00 00
00 03 01 00
00 03 00 00
00 04 01 00
00 03 00 00
```

No0ne, 2023-01-28:
@anderssonbilly sorry for the delay (have no permissions to send PMs on vogons)
I'm currently out of time, may be I'll continue work in spring or summer.
You could also look into https://github.com/rasteri/HIDman which made good progress I think.

No0ne, 2023-09-20:
Please try the [fork by anteo](/anteo/ps2pico4zx) which runs the mouse in report mode. I will likely do report mode in my repo too.

No0ne, 2023-10-23:
Recently I came across a Razor which also had inverted Y-Axis in the BIOS. So I will be testing with that soon.

anderssonbilly, 2023-11-14:
@No0ne sorry for the late reply, have had way to little time for hobby projects latley.
Did you get anywhere with the Razor mouse?

No0ne, 2023-11-15:
Sorry no, didn't have time atm. Could you test with the [fork by anteo](https://github.com/anteo/ps2pico4zx)?

anderssonbilly, 2023-11-23:
Link seems to be dead. Do you have a backup?

No0ne, 2023-11-23:
Fork seems to be gone 🤨

No0ne, 2024-03-02:
@anderssonbilly could you please retry if the inverted axis is fixed with one of these two updates?
https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-boot.uf2
https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-report.uf2

No0ne, 2024-06-21:
Also got a Razor DeathAdder Model RZ01-03850 but I can't reproduce the problem with the recent version 0.9-1.1!
Maybe the TinyUSB library upgrade fixed the problem.

closed 2024-06-21 by No0ne

No0ne, 2024-11-16:
Please try the new pre-release: https://github.com/No0ne/ps2x2pico/releases/tag/pre-1.3