No0ne/ps2x2pico #9: Add Buffer option

AbhishekAKA47, 2023-03-16:
i need it to delay the mouse movement by some time. 
can you edit it so it just buffers the mouse movement received by 1 second.
just buffer all of it dosent have to modify it in any way just buffer the mouse data
and specify at which line is that so its editable later on by someone not smart?

No0ne, 2023-07-23:
The mouse support is still not really finished. I'll have to calculate how much ram buffering one second of mouse frames would take. I also don't know how the polling rate would affect this. I will try a test implementation!
The last question about the usb-a cable I didn't understand, please rephrase.

No0ne, 2023-10-23:
Didn't hear anything back from you.

closed 2023-10-23 by No0ne