No0ne/ps2x2pico #12: Chinese mechanical keyboards not being initialized

MalvareRJ, 2023-07-19:
Thank you so much for your work on this! This is a life saver!

I have tried two chinese mechanical keyboards (Machenike K500 and Redragon Dark Avenger K568WP) and they are not initialized. BIOS complains about missing keyboard.
However, if I turn the computer on (and consequently the adapter as well) using another USB keyboard (have used Dells and HPs without any glitches) and then switch back to any of the two chinese keyboards, they work perfectly, until I switch the computer off and on again. Resets wouldn't affect this at all.
Interestingly, I can also make these keyboards work if I initialize the computer using a USB mouse, then switch it with any of the chinese keyboards.
Seems like there's something missing in these keyboards at initialization time.

Thanks in advance for any support in this matter!

No0ne, 2023-07-23:
Please retry with this version: https://github.com/No0ne/ps2x2pico/raw/main/ps2x2pico.uf2
It is compiled with the newest pico sdk and tinyusb release.

closed 2023-07-27 by MalvareRJ

No0ne, 2023-07-27:
Cool, thanks! The mouse support isn't finished yet anyway.