No0ne/ps2x2pico #13: not producing UF2 output

TsvetanUsunov, 2023-07-27:
Your config do not produce UF2 output

No0ne, 2023-07-27:
🤔 
Please post the output of:
```
git status
git rev-parse HEAD
grep set\(PICO_SDK_VERSION_ $PICO_SDK_PATH/pico_sdk_version.cmake
ls -l *.uf2
```

TsvetanUsunov, 2023-07-27:
I deleted .uf2 files in both build and ps2pico directory as I saw that they are not updated when execute make
they are never build after this

the output is
```
ceco@ceco-ThinkCentre-M920q:~/git/ps2x2pico$ git status
git rev-parse HEAD
grep set\(PICO_SDK_VERSION_ $PICO_SDK_PATH/pico_sdk_version.cmake
pwd
ls -l *.uf2
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	deleted:    ps2x2pico.uf2

no changes added to commit (use "git add" and/or "git commit -a")
b874f5c6dcdb057d60abf4c91afb27174010161d
set(PICO_SDK_VERSION_MAJOR 1)
set(PICO_SDK_VERSION_MINOR 5)
set(PICO_SDK_VERSION_REVISION 1)
#set(PICO_SDK_VERSION_PRE_RELEASE_ID develop)
set(PICO_SDK_VERSION_STRING "${PICO_SDK_VERSION_MAJOR}.${PICO_SDK_VERSION_MINOR}.${PICO_SDK_VERSION_REVISION}")
    set(PICO_SDK_VERSION_STRING "${PICO_SDK_VERSION_STRING}-${PICO_SDK_VERSION_PRE_RELEASE_ID}")
/home/ceco/git/ps2x2pico
ls: cannot access '*.uf2': No such file or directory
ceco@ceco-ThinkCentre-M920q:~/git/ps2x2pico$ 

```

No0ne, 2023-07-27:
Please try this and post the output:
```
cd build
elf2uf2/elf2uf2 -v ps2x2pico.elf ps2x2pico.uf2
```

TsvetanUsunov, 2023-07-27:
thanks! this did the job :)

closed 2023-07-27 by TsvetanUsunov

No0ne, 2023-07-27:
Cool, hm but I don't know why cmake/make didn't do that 🤷

No0ne, 2023-07-30:
I saw you are integrating it into the Neo6502, just ordered one myself! 👍

TsvetanUsunov, 2023-07-30:
yes, I was study to add in Neo6502 (where PS2 interface is not necessary)  but the USB keyboards/mouse fits great the FabGL PS2 interface for the further revision of  AgonLight2 ;) PS2 keyboards and mouses are exotic today

No0ne, 2023-07-30:
> PS2 keyboards and mouses are exotic today
True! I ran into the same issue using newer keyboards on my NuXTv2