No0ne/ps2x2pico #15: PS/2 as library

mdevaev, 2023-08-02:
Hello from [PiKVM Project](https://github.com/pikvm/pikvm)! :)

I am very impressed with your work on PS/2 and especially with the fact that unlike other projects, you got rid of `sleep()` calls.

My project uses an Arduino-based firmware to emulate a USB keyboard and mouse, and a PS/2 keyboard. Right now I'm porting to RP2040, and it would be great if I could reuse your work to emulate PS/2 mouse and keyboard, because PS/2 support seems pretty tricky to me.

Do you think you could make a part of your project in the form of a library? I would be happy to donate if you did that.

No0ne, 2023-08-03:
Hi!
I'm running a PiKVM on a Zero 2 myself, thank you, its awesome!

I'm just getting started working on the project again since I had not enough time available this year so far.
Currently digging into the pull request from [hoffman373](https://github.com/hoffman373).
I hope I'll get the mouse finally working. This seems to be the most difficult part, don't know why.
The keyboard part is working stable hence the 1.0/1.1 release on the other ps2pico repo.

How do you interface the pico with the pi, do you have schematics?
I think you only need the ps2 output stage, not the usb input, right?

I don't know how to build a library yet, but maybe your codebase helps.
Do you have your repository for the port online already?

mdevaev, 2023-08-04:
> I'm running a PiKVM on a Zero 2 myself, thank you, its awesome!

Thank you sir! Glad you like it :)

> I hope I'll get the mouse finally working. This seems to be the most difficult part, don't know why.

I have a protocol analyzer and a host with PS/2, so I can check how everything works. I hope this helps.

> How do you interface the pico with the pi, do you have schematics?

This time, unlike the old Arduino HID, I use SPI so as not to occupy the UART on the Raspberry. We are currently only preparing the documentation, but here the schematics. It's super-easy:

[image]

> I think you only need the ps2 output stage, not the usb input, right?

Yes, you're right.

> I don't know how to build a library yet, but maybe your codebase helps.

In the simplest case, you can make a separate directory with c/h-files, where there will be all functions for working with the ps/2 keyboard/mouse and there will be no `main()` function. I would already be able to take advantage of this by simply adding a submodule. You can separately create something like `main.c` at a higher level and use your functions from the directory.

Here the source code. All you need is run `make` in this directory. It will download own sdk, tinyusb and build the firmware.
https://github.com/pikvm/kvmd/tree/master/hid/pico

Ideally I would like an API like:
```c
void ps2_kbd_init(...); // Pass the keyboard pinout
void ps2_kbd_send(uint8_t key, bool state); // [key] is a USB keycode, [state] is true if key pressed, else if released
uint8_t ps2_kbd_get_leds(void); // Leds state like on USB

void ps2_mouse_init(...);
void ps2_mouse_send(uint8_t buttons, int8_t x, int8_t y); // Buttons is a current mask of the state
// Something for wheel also, etc. Maybe make separate functions for the each event type.

void ps2_task(void); // I will call it if you need some periodic processing in the main loop
```

mdevaev, 2023-08-07:
Sup. I've made some documentation for the new hid: https://docs.pikvm.org/pico_hid/

PS: BTW how do you like my API ^^^ idea?

No0ne, 2023-08-08:
Thanks for the docs, like it!
I'll make a pull request to your repo after I finished debugging the mouse support.
Give me a couple of weekends and I'll come back to this topic!

mdevaev, 2023-08-08:
Great! I'll put "TODO: PS2" tags for you, so you can just do grep and place calls to your code in the right places

If you need help, let me know. I will arrange for you to access a machine with a PS/2 protocol analyzer.

mdevaev, 2023-08-08:
Here, I've just made some stubs with comments for you: https://github.com/pikvm/kvmd/commit/895ec1cb737a655a9b8ccdc6b6a911a9bfdd8e74

mdevaev, 2023-08-08:
Note: for PS/2 on Pico HID you can use GPIO10-17.

No0ne, 2023-08-08:
GPIO11-15 would be perfect, then a standard level converter fits like this:
```
                   _________________
                  |                 |
Pico GPIO11 ______| LV1         HV1 |______ PS/2 keyboard data
Pico GPIO12 ______| LV2         HV2 |______ PS/2 keyboard clock
Pico GPIO13 ______| LV          HV  |______ PS/2 5V + Pico VBUS
Pico    GND ______| GND         GND |______ PS/2 GND
Pico GPIO14 ______| LV3         HV3 |______ PS/2 mouse data
Pico GPIO15 ______| LV4         HV4 |______ PS/2 mouse clock
                  |_________________|
```
[image]

No0ne, 2023-08-08:
Ah, if you leave out `+ Pico VBUS` you can also leave out the diode and the pico is solo powered by the pi.

mdevaev, 2023-08-08:
This diode is in the other direction. It does not allow Raspberry to be powered by Pico.

No0ne, 2023-08-08:
Yes, I meant you can leave out violet cable from the photo. So 5V from the PS2 ports ending at the HV pin.

No0ne, 2023-08-08:
Hm, but if you leave it in, the pico can initialize the PS2 port even if the raspberry pi is not powered.

mdevaev, 2023-08-08:
I don't see anything wrong with that. In my opinion, HID should work independently of Pi. It doesn't matter to you whether the user is at the keyboard or not, it is always initialized.

No0ne, 2023-08-08:
Yes! just ignore what I said: 🤪
> Ah, if you leave out `+ Pico VBUS` you can also leave out the diode and the pico is solo powered by the pi.

No0ne, 2023-08-10:
making progress:
[image]

mdevaev, 2023-08-10:
Oh, so you have analyzer :D

No0ne, 2023-09-08:
Should be ready for testing, added note to [README.md](https://github.com/No0ne/ps2x2pico/blob/main/README.md)!

closed 2023-09-08 by No0ne

mdevaev, 2023-09-16:
Could you give me your paypal? I promised donation for you :)