No0ne/ps2x2pico #17: Freezing on KVM

yarodin, 2023-08-28:
I have KVM with 4 computers attached, ps2pico plugged to PC4 KVM USB port. 
ps/2 ports on ZX Evolution <-> ps2pico <-> PC4 USB port on KVM / Common USB port <->  USB  Logitech unifying receiver -> Logitech Mouse & Keyboard

If I switching on ZX Evolution with KVM setted to PC4 - all works perfectly untlil I have switched to PC1/2/3 computer - after several second ps2pico stop acting. Same problem if I switching on ZX Evo with KVM setted to PC1/2/3.

No0ne, 2023-08-28:
Which hardware do you use? ps2pico or ps2x2pico?
Which firmware version did you download?

yarodin, 2023-08-28:
ps2x2pico
beta-0.5

No0ne, 2023-08-28:
please try out the new pre-releas beta-0.7
(remember to swap each data/clock pin pair on the 5V side)

yarodin, 2023-08-29:
I updated firmware to beta-0.7.

"If I switching on ZX Evolution with KVM setted to PC4 - all works perfectly untlil I have switched to PC1/2/3 computer - after several second ps2pico stop acting" - this case fixed, thanks a lot.

"Same problem if I switching on ZX Evo with KVM setted to PC1/2/3." - this is not fixed.

No0ne, 2023-08-29:
Did you connect mouse and keyboard to the KVM or keyboard only?

yarodin, 2023-08-29:
I have one USB Logitech unifying receiver plugged to KVM and mouse and keyboard connected through it.

No0ne, 2023-08-29:
Ok, and both stop working? or only the mouse?

yarodin, 2023-08-29:
if I switching on ZX Evo with KVM setted to PC1/2/3 and then setting KVM to PC4 mouse and keyboard not working on PC4, but still normaly working on PC1/2/3.

Seems, ps2x2pico isn't working if mouse and keyboard is not plugged to USB (or switched off by KVM) when it power on.

No0ne, 2023-08-29:
Ah yes, if theres nothing connected to usb it reports an error on the ps/2 side! I think i could add a jumper setting to disable that an always report OK on ps/2.

No0ne, 2023-09-16:
[ps2x2pico.zip](https://github.com/No0ne/ps2x2pico/files/12641676/ps2x2pico.zip)
Please retry with this binary, its currently hardcoded to always answer OK.

yarodin, 2023-09-17:
This version is working in all cases with KVM. You can close issue. Thanks.

No0ne, 2023-09-17:
Thank you for the feedback!

closed 2023-09-17 by No0ne