No0ne/ps2x2pico #20: could CDC be added?

robert-rozee, 2023-12-23:
hi,
    how practical would it be to add CDC device support as a third device option, with keyboard and mouse input (instead of sent out 2x PS/2 ports) being translated to VT escape sequences and piped back out to the USB CDC serial port? my usage case is with a VGA picomite ([https://geoffg.net/picomitevga.html](url)) which can NOT talk to USB devices - it can only talk to a serial terminal emulator running on an attached host. the picomite would then be attached to a USB hub (as a device) along with the keyboard and mouse, and the "ps2x2pico+CDC" would sit there as a host translating between keyboard+mouse and picomite VGA.

see the following discussion thread:
[https://www.thebackshed.com/forum/ViewTopic.php?FID=16&TID=16529](url)

cheers,
rob   :-)

No0ne, 2023-12-24:
Hi, yes I think this should be possible. But since this doesn't have to do anything with PS/2 I'll start another project for this.
As freetime is rare right now I don't know when I can get working on this.

robert-rozee, 2023-12-25:
that would be brilliant! if there is anything i can do to help, let me know. a few folks on the backshed forum (https://www.thebackshed.com/forum/ViewTopic.php?FID=16) have already started playing around with the ps2x2pico just to do USB->PS/2 keyboard translation - finding a PS/2 keyboard these days is starting to get a tad hit-and-miss.

cheers,
rob   :-)