No0ne/ps2x2pico #21: Keyboard not reconnecting after power cycle

peek82, 2023-12-30:
I'm using a ps2x2pico w/ a Picomite (keyboard only) and have wired up two versions, one using the level shifter board as prescribed, and one using direct wiring from pico to pico with 4k7 'pullups' going to GP13 of the ps2x2pico. Both versions work well until a power cycle, after which I need to reset the ps2x2pico by grounding the run pin before it will reconnect to the keyboard again. As a workaround, I am toggling the run pin on boot with an extra GPIO on the picomite. Love being able to use a USB keyboard for input, much thanks!

[image]

No0ne, 2024-01-04:
I'm currently doing another project for the picomite: https://github.com/No0ne/hid2cdc
It is hooked up like this: https://www.thebackshed.com/forum/ViewTopic.php?TID=16545&P=1
But its not finished yet, F-keys are not working.

For debugging the powercycle bug, an you hook up a serial console to GPIO 0 and post the output?
It is sending at 115200n8

peek82, 2024-01-09:
Thank you for the reply but when I moved from the breadboard to soldered connections the problem went away, so I'm chalking it up to a buggered up breadboard / jumper wires. It's functioning perfectly now, I really appreciate your work on this & the hid2cdc project. My current use case requires the ps2 version but I'll definitely be looking at using hid2cdc in the future.

No0ne, 2024-01-09:
Cool, thanks!

closed 2024-01-09 by No0ne