No0ne/ps2x2pico #22: Does not work with YD RP2040

Chandler-Kluser, 2024-01-03:
This is an interesting issue (in my opinion)...

First of all, thanks for sharing this great project!!! It worked as expected with my original Raspberry Pi Pico Board using a micro USB OTG adapter to my USB keyboard...

Then I used this [YD-RP2040](https://github.com/initdc/YD-RP2040/blob/master/YD-2040-2022-V1.1-SCH.pdf) board:

[image]

with an USB Type C OTG adapter, but it does not work... the board is completely pin compatible with the Raspberry Pi Pico...

Do you have any clue? Thanks in advance :smile:

No0ne, 2024-01-04:
Can you hook up a serial console to GPIO 0 and post the output?

Chandler-Kluser, 2024-01-05:
it is not outputing anything, when I turned on the first time it showed:

[image]

Are you sure the configuration is 115200 8-N-1?

No0ne, 2024-01-08:
🤔 hm you should get at least this string: "ps2x2pico-0.8 DEBUG=false"

Chandler-Kluser, 2024-01-08:
I am so sorry :sweat_smile: 

I had trouble with the UART interface while using another USB device, which were in conflict...

About the YD-RP2040, it actually worked with the built in .uf2 file in releases :smile: 

Thanks and I am sorry for bothering!!

closed 2024-01-08 by Chandler-Kluser

Chandler-Kluser, 2024-01-09:
I have actually find out that it can be these QSPI flashes, some of these boards have lower memory size.

Another thing could be that this YD board (right side on the picture) have RGB light, I don't know if it is using any GPIO for ps2x2pico...

[image]

I took this left board from WeAct Studio and it worked flawlessly as well as the Raspberry Pi Pico (Raspberry Pi Foundation Official Board).

Thanks one more time for this great project!!

No0ne, 2024-04-19:
Seems to the that Pin 20 aka Vout aka VBUS is connected after the diode on the YD-RP2040, see: https://github.com/No0ne/ps2x2pico/issues/33#issuecomment-2066438989