No0ne/ps2x2pico #24: PS2 input repeating

Gouster4, 2024-02-09:
Hello. I would like to use this project on my KVM which is almost older than me and has only PS2 port. But i would also like to keep my RKM working, which has ps2 output.

Can you please make optional PS2 input, so it can repeat PS2 input and also convert USB to PS2? Thank you.

No0ne, 2024-02-10:
Yes! In fact I'm already working on it because of this related issue: https://github.com/pikvm/pikvm/issues/476

I'm experimenting with an 8-channel level shifter, but I think I got the wrong part wich isn't suitable for open-drain circuits. But I can always use two 4-channel ones. I'll keep you updated if I made progress!

Gouster4, 2024-02-10:
Thats awesome. Thank you.

Now bit unrelatwd question, but dont want to open new issue report.
I would also like to know if this project can run on rp2040 zero with 2M flash. Thank you, and sorry for unrelaedd question.

No0ne, 2024-02-10:
I think this should work with any rp2040 board.

Gouster4, 2024-02-12:
Awesome. I would rather re-route cables from controller to level converter to keep gpio pins same if possible. I would like to prepare my hardware for PS2 passtrought, do you have already plans which gpio pins you will use for ps2 input once you implement ps2 passtrought? Are there also some specific requirements for pin capabilities for data or clock? Or basically any gpio pins can be used?

Thank you.

Gouster4, 2024-02-13:
Can you comfortably use pins between gpio0 and gpio10? Because those are present and easy accessable also on zero version. Basically 0-15&26-29 are easy accesable, 17-25 are still accessable but requires precise soldering on small solder pads with small gaps between, but still good enought. GPIO 16 is missing. I think, if usable, then requires microscopic soldering to chip/soc itself.

If it wouldnt be possible to confortably use gpio0-15 pins, there is still way to chagne gpio and compile for myself. But its easier to be up-to-date when using releases.

gpio 11-15 used as ps2 output are easy accessable on zero version without any issues. So thats perfect. Luck for users with zero version, because i cannot imagine changing existing pins in this project when lot of people have already soldered those exact pins to support releases instead of compiling themself.

Just let me know how did you decided. Its not a big deal for me to change gpio and compile, but if it would be possible to use gpio0-10 and still keep it good looking on non-zero type of board, it would be way easier for all users with zero version. 

Thank you

No0ne, 2024-02-13:
Sure, I corrected this for PS/2 input:
```
                   _________________
                  |                 |
Pico GPIO3 ______| LV1         HV1 |______ PS/2 keyboard data
Pico GPIO4 ______| LV2         HV2 |______ PS/2 keyboard clock
Pico GPIO5 ______| LV          HV  |______ PS/2 5V + Pico VBUS
Pico    GND ______| GND         GND |______ PS/2 GND
Pico GPIO6 ______| LV3         HV3 |______ PS/2 mouse data
Pico GPIO7 ______| LV4         HV4 |______ PS/2 mouse clock
                  |_________________|
```

No0ne, 2024-02-16:
I've cobbled together the first prototype: only keyboard is working at the moment and also only one data direction, meaning the leds on the keyboard don't get set.

Please try this: https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico.uf2

I also added two 3.3k pull-up resistors from HV1 to HV and HV2 to HV as the built-in 10k from the levelshifter might be too weak for your keyboard, you'll have to try it out.

Gouster4, 2024-02-17:
Thank you. Im still waiting for levelshifters. But maybe i can prototype my own temporary level shifter from pocally available components to test it.

No0ne, 2024-02-17:
You can try to run your keyboard on 3.3 Volts, maybe it works. Then you don't need a level shifter, only two pull-up resistors.

No0ne, 2024-04-07:
Did you have time to try it out?

Gouster4, 2024-04-07:
> Did you have time to try it out?

Sorry. I went to work trip, and im out of my tech for a while. Only access i have is remote services running at my server.

But in about 2 weeks i will be at home.
And i already recived level shofters, which my wife picked up. So, once i get home im ready to test it.

Gouster4, 2024-04-28:
Finally got home, and put that together.
Keyboard is working exaclty like you described.
Do you plan to add mouse support, and keyboard leds support.

Thank you

No0ne, 2024-04-28:
Thats cool to hear, yes I'll add mouse passthru support as I also need it for the PiKVM integration.

For the other issue, please post a diagram which devices are connected how.

Gouster4, 2024-04-28:
> lease post a diagram which devices are connected how.

Awesome. My plan is also to use PiKVM, but old X86 unofficial release of PiKVM as VM on my main server, with USB-Uart adapter (keyboard+mouse from PiKVM VM to KVM switch) adapter and Video Grabber passed trought to VM.

Currently, im using Belkin´s Matrix2 KVM, HP RKM and bunch of Servers.
On that HP rkm, i replaced ball mouse to some random touchpad from some broken laptop.

HP RKM(with replaced ps2 touchpad) -> Belkin´s Matrix2 KVM -> HP DL380p Server.

Before touchpad replacement, it was working.
Now im using ps2 mouse instead, and i have connected touchpad with PS2->USB converter directly to Server, as KVM doesnt like it.

For future, i would like to integrate ps2x2pico into KVM case like cutting wires on original ps2 connector of keyboard/mouse input, and connecting it trought ps2x2pico, and add 3 usb ports (2 TypeA connected to USB hub which connects to ps2x2pico, and 3rd typeC (for PiKVMx86) which would be connected to USB hub, where video grabber card and usb-uard bridge would be connected.)

No0ne, 2024-04-29:
Ah yes now I understand. It will most certainly fix the touchpad problem as the PS2 input and output are isolated as in no bytes travel directly from in out. The PS2 input rather controls the virtual mouse created on the PS2 output same as the USB input events do.

Gouster4, 2024-04-29:
> It will most certainly fix the touchpad problem as the PS2 input and output are isolated, as in no bytes travel directly from in to out. The PS2 input rather controls the virtual mouse created on the PS2 output, same as the USB input events do.

Thats awesome. Thank you. Looking forward for mouse support. 👍

No0ne, 2024-06-14:
Sorry, took me a bit longer to find time for this 😯
Try this, now keyboard and mouse should work: 
[ps2input.zip](https://github.com/user-attachments/files/15841002/ps2input.zip)

Gouster4, 2024-06-14:
> Sorry, took me a bit longer to find time for this 😯 Try this, now keyboard and mouse should work: [ps2input.zip](https://github.com/user-attachments/files/15841002/ps2input.zip)

Thank you so much. I'm on work trip again, but I will be at home in about two weeks, so will definitely test this out. Looking forward to test it.

Is this with PiKVM support on RX&TX pins included or without?

No0ne, 2024-06-14:
I currently have a pull request open for this: https://github.com/pikvm/kvmd/pull/171
Yes I tested this with the pico hid bridge over usb and it should also work over spi and serial.

No0ne, 2024-06-14:
But the pinout is different (uses 26+27 and 16+27) so for rp2040 zero we'll have to compile a custom pico hid bridge.

Gouster4, 2024-06-14:
> I currently have a pull request open for this: [pikvm/kvmd#171](https://github.com/pikvm/kvmd/pull/171) Yes I tested this with the pico hid bridge over usb and it should also work over spi and serial.

Nice. I will then make a fork, try to merge it, and keep current pin Configuration and build.

Gouster4, 2024-07-14:
https://github.com/No0ne/kvmd/commit/6c23bef9987d79c502e03830f7f64fa05967ca85

This is my attemp to get it working with same pins used as i tested here before.

No0ne, 2024-07-14:
Looks good, but you also need to disable and hardcode the jumper configurations since they share the same GPIOs like this:
```
diff --git a/hid/pico/src/ph_outputs.c b/hid/pico/src/ph_outputs.c
index 4ec40de5..472635a6 100644
--- a/hid/pico/src/ph_outputs.c
+++ b/hid/pico/src/ph_outputs.c
@@ -53,29 +53,29 @@ static int _read_outputs(void);
 
 void ph_outputs_init(void) {
 #	define INIT_SWITCH(x_pin) { gpio_init(x_pin); gpio_set_dir(x_pin, GPIO_IN); gpio_pull_up(x_pin); }
-	INIT_SWITCH(_PS2_ENABLED_PIN);
-	INIT_SWITCH(_PS2_SET_KBD_PIN);
-	INIT_SWITCH(_PS2_SET_MOUSE_PIN);
+	//INIT_SWITCH(_PS2_ENABLED_PIN);
+	//INIT_SWITCH(_PS2_SET_KBD_PIN);
+	//INIT_SWITCH(_PS2_SET_MOUSE_PIN);
 
-	INIT_SWITCH(_BRIDGE_MODE_PIN);
+	//INIT_SWITCH(_BRIDGE_MODE_PIN);
 
-	INIT_SWITCH(_USB_DISABLED_PIN);
-	INIT_SWITCH(_USB_ENABLE_W98_PIN);
-	INIT_SWITCH(_USB_SET_MOUSE_REL_PIN);
-	INIT_SWITCH(_USB_SET_MOUSE_W98_PIN);
+	//INIT_SWITCH(_USB_DISABLED_PIN);
+	//INIT_SWITCH(_USB_ENABLE_W98_PIN);
+	//INIT_SWITCH(_USB_SET_MOUSE_REL_PIN);
+	//INIT_SWITCH(_USB_SET_MOUSE_W98_PIN);
 #	undef INIT_SWITCH
 	sleep_ms(10); // Нужен небольшой слип для активации pull-up
 
-	const bool o_ps2_enabled = !gpio_get(_PS2_ENABLED_PIN); // Note: all pins are pulled up!
-	const bool o_ps2_kbd = !gpio_get(_PS2_SET_KBD_PIN);
-	const bool o_ps2_mouse = !gpio_get(_PS2_SET_MOUSE_PIN);
+	const bool o_ps2_enabled = true; //!gpio_get(_PS2_ENABLED_PIN); // Note: all pins are pulled up!
+	const bool o_ps2_kbd = true; //!gpio_get(_PS2_SET_KBD_PIN);
+	const bool o_ps2_mouse = true; //!gpio_get(_PS2_SET_MOUSE_PIN);
 
-	ph_g_is_bridge = !gpio_get(_BRIDGE_MODE_PIN);
+	ph_g_is_bridge = true; //!gpio_get(_BRIDGE_MODE_PIN);
 
-	const bool o_usb_disabled = (ph_g_is_bridge || !gpio_get(_USB_DISABLED_PIN));
-	const bool o_usb_enabled_w98 = !gpio_get(_USB_ENABLE_W98_PIN);
-	const bool o_usb_mouse_rel = !gpio_get(_USB_SET_MOUSE_REL_PIN);
-	const bool o_usb_mouse_w98 = !gpio_get(_USB_SET_MOUSE_W98_PIN);
+	const bool o_usb_disabled = (ph_g_is_bridge); // || !gpio_get(_USB_DISABLED_PIN));
+	const bool o_usb_enabled_w98 = false; //!gpio_get(_USB_ENABLE_W98_PIN);
+	const bool o_usb_mouse_rel = false; //!gpio_get(_USB_SET_MOUSE_REL_PIN);
+	const bool o_usb_mouse_w98 = false; //!gpio_get(_USB_SET_MOUSE_W98_PIN);
 
 	int outputs = _read_outputs();
 	if (outputs < 0) {
```
The boolean **ph_g_is_bridge** decides if the Pico is connected via USB (true) or SPI (false) to the PiKVM.

No0ne, 2024-07-14:
I think you missed GPIO 7.

Gouster4, 2024-07-14:
Thanx. that makes perfectly sense. Now i leaved w98 mouse mode pins active, as i have those pins accessable, and it should break functionality i guess.
https://github.com/No0ne/kvmd/commit/c974a6a6189cbd97f6ea086bd5aa3bc7d2ff66e3

#define _MOUSE_IN_DATA_PIN 6
gpio6 is on pin number 7 i guess.

#define _USB_ENABLE_W98_PIN 7
should be gpio 7, so pin number 8, right?

Then it shouldnt be conflicting.

Gouster4, 2024-07-14:
does this also support usb input (USB->PS/2) like ps2x2pico?

No0ne, 2024-07-14:
8 and 9 are also used for _USB_SET_MOUSE_REL_PIN and _USB_SET_MOUSE_W98_PIN.

No, direct USB Input is not supported by the Pico HID, only via a PiKVM connection.

Gouster4, 2024-07-14:
> put is not supported by the Pico HID, only via a PiKVM connection.
So, ideally i should daisy-chain two raspberries pico, to get both USB and PiKVM input?
Or is there any other option?

No0ne, 2024-07-14:
Yes this should work. But I thought you had PS/2 keyboard and a touchpad in the first place?

Theres also the software route of enabling USB input on the Pico HID, but the hardware solution is definitly easier for now.

Gouster4, 2024-07-14:
> Yes this should work. But I thought you had PS/2 keyboard and a touchpad in the first place?
> 
> Theres also the software route of enabling USB input on the Pico HID, but the hardware solution is definitly easier for now.

Yes, I have PS2 input from RKM, but I would like this KVM to accept also USB input, as it supports also USB output. Not sure what I would connect there in future.

But PS2 input is what I'm currently using, and would use for long time.

Gouster4, 2024-07-14:
Or USB is used to emulate mouse for host?
If so, then those:
+	//INIT_SWITCH(_USB_ENABLE_W98_PIN);
+	//INIT_SWITCH(_USB_SET_MOUSE_REL_PIN);
+	//INIT_SWITCH(_USB_SET_MOUSE_W98_PIN);
are used to set mouse mode for USB mouse emulation?

No0ne, 2024-07-15:
Correct, the boolean ph_g_is_bridge globally enables/disables USB emulation: https://docs.pikvm.org/pico_hid_bridge/

In the bridge configuration USB emulation is disables as its used to connect to the PiKVM.

No0ne, 2024-07-15:
I think the latency will not be noticeable.

Gouster4, 2024-07-15:
Can you please include kvmd/hid compatible input on uart in ps2x2pico? Or it would be too complex to do? No matter if it would be New "The Pico HID Bridge" or old implementation "The Pico HID" used for PiKVM V1.

No0ne, 2024-07-15:
Huh, no, I think thats out of the scope of that project and too complex.

No0ne, 2024-07-15:
Is the PS/2 touchpad routed through the input now working?

No0ne, 2024-11-15:
documented: https://github.com/No0ne/ps2x2pico?tab=readme-ov-file#ps2-passthru

closed 2024-11-15 by No0ne