No0ne/ps2x2pico #25: [feature request] Touchscreen support?

Gouster4, 2024-02-13:
Can you also add support for USB and RS232->USB touchscreen?
Thank you.

Gouster4, 2024-02-17:
Basically it wouldnt require any difference in connection.

PS2 output would be connected to PC/KVM or other device, and touscreen can be connected to usb.
Some touch screen controllers already comes with usb output, some with rs232, which can be used with USB-Serial adapter on USB input of raspberry.

But in not sure if it would be possible, because some touchscreens needs special drivers. But if there are some plug&play, it can be possible i guess.

No0ne, 2024-02-17:
Ah, I think thats not possible because touchscreens generally use absolute pointer mode and the PS/2 mouse only can transmit relative movements. So you would need to use the touchscreen like a trackpad which defeats the purpose I think.

Gouster4, 2024-02-17:
I got it. I just seen that touchscreen controller with ps2. But i also thought touchscreen over ps2 isnt possible. That means that its just used as trackpad. Or maybe it is being calibrated to specific resolution, while after connection it will move cursor to one corner of screen, and when touched, it can calculate how much to relative move mouse to touched area. But user would need to set up reaolution, or at least aspect ratio, and then fine-tune are with mouse speed, as it will move mouse to (ideally top left) corner and translate touched area to relative mouse movement.

It would fail with KVM for sure, because raspberry would have no clue that KVM switched input. But for single device it should be possible. It can get unsynchronized when other input is used or if user uses feature to move mouse to pop-up window, but when it would be only input, or rest of inputs would be connected also trought raspberry it should work.

No0ne, 2024-02-17:
I think this isn't very feasible with relative movement. Maybe those touchscreens use a special drive on the OS side which enable absolute movement, like some synaptic trackpads also can do. But this would be very difficult to implement.

Gouster4, 2024-02-17:
So, it can be possible, but most likely unstable with regular driver. And it does need very complex code to at least experiment with it with not guaranteed result. 
It was just an idea. I also think its not worth, as developing special driver would be pain in the a**, and looses whole point of plug&play, and would be unusable with KVM. Without special driver it would be unstable and unusable with multiple inputs connected directly to PC (not trought raspberry).

But interpreting touchscreen as giant touchpad would avoid those problems.
It would work as big touchpad. But im still not sure how difficult would be to implement it anyways, so idk if its worth it.