No0ne/ps2x2pico #26: The firmware for the PC-XT interfaces

Piligrim-77, 2024-02-19:
Hi,
I have a question - will the "ps2pico-XT.uf2" firmware work on this adapter with a logic converter board?

No0ne, 2024-02-19:
No, you can not mix firmware between ps2pico and ps2x2pico.
The transistors on ps2pico invert the signal where as the logic level shifter on ps2x2pico does not.
So the firmware sends inverted bits to correct for that.

Piligrim-77, 2024-02-19:
> No, you can not mix firmware between ps2pico and ps2x2pico.
> The transistors on ps2pico invert the signal where as the logic level shifter on ps2x2pico does not.
> So the firmware sends inverted bits to correct for that.

Is it possible to modify the firmware so that it works on this version of the adapter?