No0ne/ps2x2pico #27: [Feature Request] Mac 128 keyboard support

aaronnewcomb, 2024-03-13:
First of all, let me just say that this is an awesome project!! I have been able to get my modern mechanical keyboard working on many old systems. Since there are many extra GPIO pins on the Pico, I was thinking it might be cool to be able to get this to work on the Mac 128 - Mac Plus that have those reversed 4 pin RJ-9 ports. Possible???

I will be covering this project on an upcoming episode of the Retro Hack Shack once I have done some more testing. Keep up the good work!

No0ne, 2024-03-13:
Cool thanks, I already knew your channel!
Please note: I've still one last major issue with mouse support left before I release version 1.0.

For the pre-ADB Mac support I found a similar adapter: http://www.synack.net/~bbraun/mackbd/
The hardware with clock/data looks identical, only the connector is different and it should be not too hard to implement this!
Main problem here is I don't have any pre-ADB Macs to test this with, so that makes it difficult again.

Maybe I could test with something like this: https://www.tinkerboy.xyz/product/tinkerboy-m0110-keyboard-to-usb-converter-for-the-apple-macintosh-128k-512k-plus-keyboard/

aaronnewcomb, 2024-03-13:
If you are in the SF Bay area, I can hook you up. 😀 Otherwise, I am happy
to do any testing you might need. I know it's not the same, but I am happy
to help if I can.

Thanks,
Aaron

No0ne, 2024-03-13:
Thanks, that would be very nice but I'm in Austria, so not really nearby 😏.

I'll try to reimplement this: https://github.com/n055/ps22plus
Let you know if there is something new to test!

aaronnewcomb, 2024-03-13:
Great! If this works out, I would be happy to make a little daughter board
for this that includes a few level shifters to make it easier for people to
use.

Thanks,
Aaron Newcomb

Retro Hack Shack <https://www.youtube.com/channel/UCN_u47_wJuhh249H9E8sjtw/>
Benicia Makerspace <http://beniciamakerspace.org>
Linux for Makers <https://amzn.to/2BwPjlj>
TWiT Network <https://twit.tv/people/aaron-newcomb>

No0ne, 2024-03-15:
I've cobbled together a really quick&dirty test to see if I'm on the right path: the only keys working are A-Z, 0-9, enter, backspace, tab, space, -, =, [, ] and \.

https://github.com/No0ne/ps2x2pico/raw/27-mac-128-keyboard-support/hid2mac.uf2

Please try it on a real Mac: connect HV1 to data and HV2 to clock on the Mac side.
LV1 and LV2 on the Pico side should already be GPIO11+12.

I will move this to a new repository since there will be no PS/2 code anymore.

No0ne, 2024-04-07:
Did you have time to try it out?

aaronnewcomb, 2024-04-07:
Perfect timing. Sorry it has taken so long, but when I went to find an
early Mac to test with all of the ones I have were having issues. So I was
actually just fixing up one of them and it's ready to go now. I will give
it a try and let you know how it goes.

Thanks,
Aaron

aaronnewcomb, 2024-04-07:
OK. Just got done making the adaptor. Everything is working perfectly. 😀
One watchout that I found is that there is a warning in the manuals not to
draw more than 200ma through the keyboard port on an early Mac. I tested
the current and it showed 41.5 mA with a basic USB keyboard, but an RGB LED
keyboard with backlit keys would probably be a problem.

Thanks,
Aaron Newcomb

Retro Hack Shack <https://www.youtube.com/channel/UCN_u47_wJuhh249H9E8sjtw/>
Benicia Makerspace <http://beniciamakerspace.org>
Linux for Makers <https://amzn.to/2BwPjlj>
TWiT Network <https://twit.tv/people/aaron-newcomb>