No0ne/ps2x2pico #30: Suggested Resource

ssokolow, 2024-04-19:
To help novices with building this, you might want to consider including a link to https://www-ug.eecg.utoronto.ca/desl/nios_devices_SoC/datasheets/PS2%20Protocol.htm

It's just about the only resource I've found for the pinout of a PS/2 connector that unambiguously shows both the plug and socket views, making it much harder to get confused. (Especially when you've cut apart a cheap male-to-male PS/2 cable off Aliexpress and the colors don't match the cable used in the photos.)

I've confirmed that the "Female (Socket)" diagram in it matches the "Female connector from the front" diagram on Wikipedia and every other pinout diagram I could find, bar one which apparently got them mixed up.

No0ne, 2024-04-19:
True! I'll include that in the readme.

No0ne, 2024-04-24:
Pinout diagram added: https://github.com/No0ne/ps2x2pico?tab=readme-ov-file#pinout-diagram

closed 2024-04-24 by No0ne