No0ne/ps2x2pico #31: Document cautions regarding current consumption

ssokolow, 2024-04-19:
Given that PS/2 ports are rated for 275mA, that older motherboards originally used non-resettable fuses, that there's no spec requirement for newer motherboards to use resettable fuses, and that some USB keyboards draw so much power that they require two 500mA USB plugs to stay within specified limits, it'd be a good idea to document the DOs and DON'Ts in that regard.

(In my case, my solution was to hang the Pi (after programming but before soldering), the keyboard I plan to use, the mouse I plan to use, and the USB/VGA KVM switch this whole endeavour is for (my HP t5530 running Win98SE gives Fatal Error 0E if you unplug a USB HID device while it's running), which doubles as a three-port USB hub, off one of those cheap Chinese [Charger Doctor](https://www.ebay.ca/sch/i.html?_from=R40&_nkw=charger+doctor&_sacat=0&LH_BIN=1&_sop=15) voltmeter/ammeter things and got a maximum observed current draw of 0.17 amps.)

No0ne, 2024-04-19:
True! I'll include that in the readme.

No0ne, 2024-04-19:
Updated README.md!

closed 2024-04-19 by No0ne