No0ne/ps2x2pico #35: Mouse wheel input is ignored on some mice

ssokolow, 2024-04-19:
With `ps2x2pico.uf2` or `ps2x2pico-200hz.uf2` from the [Version 1.0](https://github.com/No0ne/ps2x2pico/releases/tag/1.0) release:

* The mouse wheel registers just fine on the Gateway pack-in mouse (Model: MOAKUO) that I was testing with in case I screwed up my assembly and fried something.
* The mouse wheel on the Acer pack-in mouse (Model: MOLDUO) that I actually use (better sensor) has no effect
* Both mouse wheels work just fine when plugged into a PC with USB support

ssokolow, 2024-04-20:
Geez, I even commented in that thread. How tired *was* I?

Anyway, I just realized that I don't need the Acer so, since it's a known issue, unless you think it'd be helpful to the development effort for me to test it right now, I'd prefer to wait until you have a "Release Candidate. Needs compatibility testing" build based on something at least as new as the v1.0 200Hz that I'm running.

(The mouse I *really* wanted to use is my Logitech G203 Prodigy so I have only *one* mouse on my desk, and its scroll wheel works fine with the ps2x2pico, but it's a bit picky and sometimes becomes non-responsive after toggling my "I emulate enough USB HID to notice when the scroll lock LED blinks" USB/DVI KVM switch. Now that I've got a brainless USB switchbox in front of it to bridge in the ps2x2pico, I can easily reset the mouse without reaching behind the desk if it decides to crash.)

No0ne, 2024-04-20:
Please test your mouse with the non-working wheel with this: https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-report.uf2

ssokolow, 2024-04-20:
Wll do... though probably after I've gone back to sleep.

The ps2x2pico is now installed in place on the underside of my desk with the cables routed, so it's now a hassle to disconnect the PS/2 side so it's safe to reprogram, awkward to disconnect the USB side so I can reprogram it, and then a hassle to hook it back up to the PS2-USB adapter I prefer to use when testing new firmware.

...maybe I should order another YD-RP2040 and another M-M PS/2 cable to cut up and build another ps2x2pico for the next time I need to test a new firmware.

No0ne, 2024-04-20:
PS/2 is a little bit hotpluggable. It works under Windows for example, the Pico sends a 0xAA (BAT successful) on both ports and Windows then reinitializes keyboard an mouse new everytime.

ssokolow, 2024-04-20:
"Not hot-pluggable" in the sense of "multiple guides I've read tell you not to do it because it's sufficiently 'not in the spec' that, depending on the device, you run the risk of blowing that potentially non-resettable fuse".

For example, this passage from the [document](https://www-ug.eecg.utoronto.ca/desl/nios_devices_SoC/datasheets/PS2%20Protocol.htm) I recommended for its pinout diagrams:

> Vcc/Ground provide power to the keyboard/mouse.  The keyboard or mouse should not draw more than 275 mA from the host and care must be taken to avoid transient surges.  Such surges can be caused by "hot-plugging" a keyboard/mouse (ie, connect/disconnect the device while the computer's power is on.)  Older motherboards had a surface-mounted fuse protecting the keyboard and mouse ports.  When this fuse blew, the motherboard was useless to the consumer, and non-fixable to the average technician.  Most newer motherboards use auto-reset "Poly" fuses that go a long way to remedy this problem.  However, this is not a standard and there's still plenty of older motherboards in use.  Therefore, I recommend against hot-plugging a PS/2 mouse or keyboard.

ssokolow, 2024-04-20:
> Please test your mouse with the non-working wheel with this: https://github.com/No0ne/ps2x2pico/raw/24-ps2-input/ps2x2pico-report.uf2

Tested.

The first time I plugged it in, the cursor drifted to the right edge of the screen at a fixed rate, ignored horizontal movement, and then when i moved the mouse up and down, it appeared to inverse-scroll the scrollbar that happened to be below the cursor but I forgot to examine what it was doing more closely.

The second time, it drifted right for a few hundred pixels, then started to function in a way that appears normal and fully functional (including the wheel) except for the cursor movement being a bit glitchy. (It's hard to be sure, but I think it's that 'counter overflows when moving faster than a certain threshold' bug I remember being mentioned previously.)

No0ne, 2024-04-20:
Ok thanks, yeah I'll have to do the HID parser.

ssokolow, 2024-04-20:
I'll also need to get that test clip and USB-TTL Serial adapter out again once it's not so late in my day. It seems that something about testing and then flashing back to the firmware I remembered having on it before has it crapping out when I KVM switch away and back.

At the moment, the behaviour is:

1. If I want it to work, I have to set the KVM switch to make the input devices visible before I power on the PS/2 side. (It was always like this)
2. If I KVM switch away and back after it's started, the Logitech G203 Prodigy stops responding but the keyboard comes back fine.
3. If I KVM switch away and back again, there's at least a chance (I only tried it once) that I'll also lose the keyboard.

In any of those failure modes, un-plugging and re-plugging the USB side doesn't fix things and never has.

No0ne, 2024-04-20:
If you want to debug this I can make you a version which also prints out PS/2 bytes on the serial debug.

ssokolow, 2024-04-20:
I'd appreciate having access to it, but, given my degree of mental fatigue right now, I can't say whether I'm willing to commit to actually using it until I've had some rest.

...especially since I tried it again when I forgot to start `miniserve` on my Linux machine *before* booting up Win98SE and, this time, it worked fine... which means there's some variable I'm going to need to pin down in order to reproduce it reliably.

(eg. Maybe the success cases were all ones where I waited until the mouse began the "I've booted up. Let me celebrate before I apply your 'LEDs off' saved setting" animation cycle before I tried touching the keyboard or mouse, while the failure cases had me jumping the gun in some way? That might explain the mouse going first and then the keyboard the next time. Without the mouse celebrating the dawn, it's hard to tell when the keyboard will have finished initializing without counting seconds.)

...I suspect I'll probably say "yes" though. Half the reason the ps2x2pico interests me is that it's more idiot-proof than the USB4VC (the other half is because it's, to quote Douglas Adams, "slightly cheaper") and I've already ordered a USB Type C Male to Type B Male cable to take the USB OTG adapter out of the equation as a potential point of failure. Diagnosing this would fit with that.

ssokolow, 2024-04-22:
Yeah, I'll do it. I'll have to fit it in around whatever else is going on, but I'll do it.

ssokolow, 2024-04-23:
I just realized that might be unclear. It's a "Yes, please offer up the build at your earliest convenience." I just meant that, depending on what comes up, there might be 12-24 hour gaps in my testing."

No0ne, 2024-04-24:
Lets continue that in #36 and leave this for the HID parser so that all mouse wheels are working.

No0ne, 2024-11-05:
The latest pre-release should fix this: https://github.com/No0ne/ps2x2pico/releases/tag/pre-1.3

hax0rbana-adam, 2024-11-07:
I can confirm, this pre-release did fix the scroll wheel issue with my mouse (which didn't work properly before). I tested both the compiled release and compiled it myself from source. Well done. :clap:

No0ne, 2024-11-08:
Thanks for testing!

closed 2024-11-08 by No0ne