No0ne/ps2x2pico #36: mouse/keyboard becomes unresponsive on KVM switch

No0ne, 2024-04-24:
I'll also need to get that test clip and USB-TTL Serial adapter out again once it's not so late in my day. It seems that something about testing and then flashing back to the v1.0 firmware I remembered having on it before has it crapping out when I KVM switch away and back.

At the moment, the behaviour is:

1. If I want it to work, I have to set the KVM switch to make the input devices visible before I power on the PS/2 side. (It was always like this)
2. If I KVM switch away and back after it's started, the Logitech G203 Prodigy stops responding but the keyboard comes back fine.
3. If I KVM switch away and back again, there's at least a chance (I only tried it once) that I'll also lose the keyboard.

In any of those failure modes, un-plugging and re-plugging the USB side doesn't fix things and never has.

**EDIT:** Maybe I should look up which pin is RESET and solder on an alternative to the reset pushbutton buried under the level shifter.

_Originally posted by @ssokolow in https://github.com/No0ne/ps2x2pico/issues/35#issuecomment-2067650032_

ssokolow, 2024-04-24:
I mentioned several comments later that further testing seems to indicate that the actual behaviour is simply "If I press a key or move the mouse before the USB device's initialization sequence completes, the ps2x2pico will ignore that device until I power-cycle the ps2x2pico" and the three cases listed above were simply due to moving the mouse or pressing a key prematurely.

That is, all three listed cases appear to just be symptoms of the same "Oops. I touched the keyboard/mouse too soon and now the ps2x2pico's handling of that type of input device is broken until I power cycle it".

No0ne, 2024-04-24:
Please post a diagram which devices are connected how, thanks!

ssokolow, 2024-04-24:
Here you go. The KVM switch is one of those ultra-simple $10 USB/VGA KVM switches off eBay where the USB side is just a "device sharing hub" (i.e. a USB hub with a pushbutton that unplugs the host-side cable from one of its physical inputs and plugs it into the other)

[image]

No0ne, 2024-04-24:
Ah I always thought this was a PS/2 KVM switch :D so debugging the PS/2 side won't do much here.
This seems to be an issue with the TinyUSB library used by the Pico-SDK. Just checked in the Pico-SDK the library isn't updated yet and we are still on the same version as in ps2x2pico-1.0.

Please record the serial output of GPIO0 while switching the KVM and/or pressing keys too early.
Currently I only have debugging messages for mount/unmount USB devices, I can add more if I don't have a clue after your posting.

ssokolow, 2024-04-24:
As I suspected, I think there's some kind of race condition involved, because it took several tries before I managed to get it to happen.

(On a related note, do you know if a Pico's GPIO pins would go through any potentially problematic states (eg. ones that would be hazardous to the connected PS/2 device's keyboard/mouse fuse) if I were to wire up a Reset pushbutton between RUN and GND and use that instead of holding down the PC's power button for 5 seconds to recover from "neither keyboard nor mouse is responding"?)

Under normal operating conditions, this is what I get from KVM-switching to and then away from it:

```
HID device address = 1, instance = 0 is mounted
HID Interface Protocol = Keyboard
HID device address = 1, instance = 1 is mounted
HID Interface Protocol = Mouse
HID device address = 2, instance = 0 is mounted
HID Interface Protocol = Mouse
HID device address = 2, instance = 1 is mounted
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 1 is unmounted
HID device address = 2, instance = 0 is unmounted
HID device address = 2, instance = 1 is unmounted
```

When I managed to get the mouse to fail to come up, switching to it and then away looked like this:

```
HID device address = 1, instance = 0 is mounted
HID Interface Protocol = Keyboard
HID device address = 1, instance = 1 is mounted
HID Interface Protocol = Mouse
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 1 is unmounted
```

I then decided to double-check, so I quit and then restarted the terminal to clear it so I couldn't get confused about where the cut line was, and then got this from the next switch to and from which also didn't work (exact same hardware and procedure):

```
ps2x2pico-1.0
HID device address = 1, instance = 0 is mounted
HID Interface Protocol = Keyboard
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 1 is unmounted
```

...but, this time, a third attempt brought it back to normal function and GPIO0 output:

```
HID device address = 1, instance = 0 is mounted
HID Interface Protocol = Keyboard
HID device address = 1, instance = 1 is mounted
HID Interface Protocol = Mouse
HID device address = 2, instance = 0 is mounted
HID Interface Protocol = Mouse
HID device address = 2, instance = 1 is mounted
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 1 is unmounted
HID device address = 2, instance = 0 is unmounted
HID device address = 2, instance = 1 is unmounted
```

I'll try for just the keyboard now.

ssokolow, 2024-04-24:
OK, it looks like the keyboard depends on something I'm consistently doing in the non-problem-causing way when I try for a regimented test routine.

It's about time for me to go to bed, but I'll leave GPIO0 hooked up and just use the thing for a few days and see what the log says when I stumble over it again.

ssokolow, 2024-04-24:
I wound up getting side-tracked and delayed, and, as a side-effect, managed to capture the something that will hopefully be of use.

When it stops responding for both keyboard and mouse, it re-displays the "blank line, `ps2x2pico-1.0, blank line` header and then stops outputting stuff on GPIO0.

No0ne, 2024-04-26:
Ok thats definitily a TinyUSB issue. I don't know how to upgrade to the latest version within the Pico-SDK, maybe this would fix it.

The initialization only appears after a [powercycle](https://github.com/No0ne/ps2x2pico/blob/main/src/ps2x2pico.c#L95) or a [panic](https://github.com/No0ne/ps2x2pico/blob/main/src/ps2x2pico.c#L113) (triggered by TinyUSB).

You can always wire a reset button to RUN/GND. Electrically this does nothing to the PS/2 lines as they can only be shorted to GND or not (aka pulled high).

bstrobel, 2024-04-27:
For me the adapter works perfectly when I connect a keyboard directly to the USB port or when I use a passive USB 2.0 Hub and connect mouse and keyboard. So far so good.

I also have a 4x1 4K HDMI/USB 3.0 KVM Switch (China brand, bought from Amazon). When I connect the adapter to this switch no keyboard or mouse input is registered by the PS/2 PC. No matter what I try. But the adapter is not dead. The PS/2 PC recognizes a keyboard (but no mouse) when booting. It prints the infamous "Keyboard not found. Press <F1> to continue" message if there is no keyboard connected.

I attached a serial terminal to GPIO0 and all I get is a line with "ps2x2pico-1.0" and nothing else. Looks like the adapter hangs.

So this sounds like another or the same TinyUSB issue. What do you think? Is there any hope that this can be fixed in the near future? Because the KVM switch is what I need to connect it to.

Just to let you know, I'm happy to help you debug and fix this issue.

No0ne, 2024-04-27:
Yes sounds like the same TinyUSB issue, I'll check if I can update this in Pico-SDK.

bstrobel, 2024-04-27:
I checked out tinyusb from github and configured the project to use it (master branch). There is definitely progress but it doesn't really solve all the problems.

Now I get debug messages when switching back and forth and after some switching the keyboard suddenly works. But not the mouse. The mouse is an old Logitech M705, so nothing fancy btw.

bstrobel, 2024-04-27:
And more news. The mouse is not working in Norton Commander but it is working in Windows 98 on the same PC (a Socket 7 PC with an Intel 430VX chipset and a VIA VT82C42N keyboard controller btw, in case you want to know). And it was working in NC when connected using a passive hub. Not sure what happens here but I will investigate more.

Once it is working on my KVM switch it keeps working not matter how often I switch back and forth. So to me it seems we are 80% there and I'm quite happy with the progress. I will call it a makers day for now but will investigate more in the next days.

No0ne, 2024-04-27:
Can you please provide the uf2 for @ssokolow to test with his KVM switch, I don't have the time right now to compile it, thanks!

bstrobel, 2024-04-27:
@ssokolow Here it is: [ps2x2pico.zip](https://github.com/No0ne/ps2x2pico/files/15138461/ps2x2pico.zip)

ssokolow, 2024-04-29:
Thanks. A bunch of things came up over the last couple of days, so it may take me a couple more before I can reasonably justify testing it.

(On Saturday, I broke my Win98SE install and then discovered that my copy of PowerQuest Drive Image 2.0 would hang mid-way through restoring it. This morning, I woke up to find the KVM switch being non-responsive. etc.)

On the plus side, if the KVM switch *has* gone bad, I at least know it shouldn't have been the cause of this, since I've never experienced this behaviour with the other PC connected to it, and I do have a 4-input unit of the same type I can replace it. (It's a 2-input unit.)

(And, Re: #37, the corrupted Windows 98 SE shouldn't be a problem since I shouldn't need to be able to boot all the way into it to test Ctrl+Alt+Delete with F12 held down.)

...I just can't reasonably justify putting testing this at the top of the priority list for another day or two.

ssokolow, 2024-05-04:
Sorry for going silent. I'm still here, I just had a bunch of stuff blindside me. (Mostly spring cleaning/home maintenance stuff I'd forgotten I set reminders for, but also a bit of .)

I'll get to this as soon as I can spare the time to diagnose/replace the cheap Chinese KVM switch and re-verify the problem behaviour with whatever I boot the t5530 into before flashing the new firmware.

No0ne, 2024-05-11:
No stress, this is after all a voluntary freetime project. I'm also currently out of time to work on it, to be continued

No0ne, 2024-05-19:
please also retry in this issue with the latest fix from here: https://github.com/No0ne/ps2x2pico/issues/37#issuecomment-2119247095

ssokolow, 2024-05-27:
OK, I've already managed to induce the mouse to become non-responsive, so it looks like it's not solved.

ssokolow, 2024-05-28:
I think it might actually be *worse* now. I'm finding the Logitech mouse non-responsive after a KVM switch away and back, even when I *do* wait for the "I've initialized" LED animation to finish.

(I haven't tested with other mice yet. I was just doing some netbooting to re-partitiong the drive using my PartitionMagic boot disk.)

ssokolow, 2024-05-28:
Speaking of which, if you want to test it locally like you did with the t5530, [this](https://www.ebay.ca/itm/315387447779) is the $15 CAD USB/VGA KVM switch I'm currently using and the one I was using before was the 2-port version.

There's a ton of them on eBay, Aliexpress, Amazon Marketplace, and anywhere else Chinese drop-shippers congregate.

[image]

ssokolow, 2024-05-29:
I just noticed something that may be relevant. The first time I move the mouse after netbooting into Damn Small Linux 4.4.10 with the ps2x2pico and the `t5530-fix.zip` firmware , it thinks I pressed the right mouse button at the instant the first motion is received.

It's getting late, but I'll try to do some tests tomorrow to see if I can narrow in on what conditions cause that effect.

ssokolow, 2024-05-29:
Interesting. The mouse refusing to come back after a KVM switch away and back, even if I wait before touching it, is specific to Damn Small Linux. Windows 98 SE appears to be behaving *better*, in that, so far, I haven't managed to get either input device to become unresponsive due to over-eager use of them after hitting the switch button.

No0ne, 2024-05-29:
Thanks for testing, did you also observe the serial debug output?
I also ordered the same KVM switch for testing, to be continued.

ssokolow, 2024-05-29:
I haven't hooked the serial up yet, because I didn't want to be doing too many things at once and I was a bit excited about the progress I was making to identify the ideal driver versions and configurations to solve some blue-screen issues on Windows 98SE. (2.5 down, 0.5 to go... as in there's a simple fix for the third but I need to figure out how to keep it from breaking something else.)

I'm not sure if I'll get to it tomorrow, given that I have some other "within the next few days" things I have to give a timeslice to, but I'll let you know as soon as I do.

No0ne, 2024-06-14:
I'm still waiting for the KVM switch to arrive, its already in shipping.

No0ne, 2024-06-29:
I've received the KVM switch and can confirm its a TinyUSB library issue, to be continued...

No0ne, 2024-07-26:
I don't have a solution yet. I've tried resetting TinyUSB if a device disconnects and also tried resetting the whole Pico itself, but this does not fix the problem :( I'll need to check with a logic analyzer now.

No0ne, 2024-07-26:
Apperently the deinit(), init() trick does work if there are two USB devices connected (keyboard AND mouse).
@ssokolow please try this:
[usbreset.zip](https://github.com/user-attachments/files/16396829/usbreset.zip)

ssokolow, 2024-07-28:
Will do.

This weekend was already spoken for, and the next couple of days are probably going to be too, but I'll let you know how to goes as soon as I fit it in.

ssokolow, 2024-11-15:
Sorry for going silent for half a year. "Life happened" and I wound up not touching the t5530 at all during that time.

I just tested the firmware version from `usbreset.zip` and the mouse does indeed now come back up under Damn Small Linux when KVM-switching away and back.

If you need any more tests, I should be able to do them promptly now.

No0ne, 2024-11-15:
🤔 pre-1.3 also seems to work with exactly two USB devices connected.
It won't work with only one or three, and also not if the mouse is on the raspberry keyboards hub.

ssokolow, 2024-11-15:
OK. I'll try to get to that tomorrow.

I lent the Acer mouse that I referred to and didn't get it back, which means I can't just budget X amount of time for testing since I may need to dig through my other mice for an uncertain amount of time to find a suitable substitute... assuming I even have one..

ssokolow, 2024-11-19:
The replacement for the Acer MOLDUO I lent just arrived. I'll try to get pre-1.3 tested tomorrow.

No0ne, 2024-11-20:
I fixed something related to two mice, please try: https://github.com/No0ne/ps2x2pico/issues/61#issuecomment-2488690026

ssokolow, 2024-11-20:
So far, so good. I haven't tested thoroughly yet (just the most basic operations under Windows 98SE), but both mice appear to have working left-click, right-click, middle-click, mouse wheel, and basic motion.

Is there anything else you might want to tweak/fix before I do a full test run of everything I can think of?

No0ne, 2024-11-20:
I'm currently working on keyboard report mode. I haven't figured out the root cause why the KVM switch only works with exactly two devices.

ssokolow, 2024-11-20:
I was about to say that [reportmode-test.zip](https://github.com/user-attachments/files/17831873/reportmode-test.zip) seems to have fixed it. I can no longer reproduce any problems with two mice and a keyboard.

In fact, I was about to say the following:

> Wouldja look at that. It appears to have also fixed the hotplug issue with Damn Small Linux.
> 
> At least on the HDMI KVM switch I'm using now (which I'm pretty sure I re-confirmed the problem with), I can KVM away from Damn Small Linux and back and not have the input devices become unresponsive.

No0ne, 2024-11-20:
Cool!

No0ne, 2024-11-21:
So every issue except switching the KVM with only one USB device attached should be fixed, right?
I also looked into the one-device issue but this is one I'm not capable of fixing my self since it sits deep within the TinyUSB library. If you are ok with that I will close this issue.

ssokolow, 2024-11-21:
> If you are ok with that I will close this issue.

Sure. I'm happy enough with the current state of things on the `reportmode-test.zip` firmware that I'm feeling the urge to procrastinate testing the new pre-1.3.

ssokolow, 2024-11-21:
No problem. :)