No0ne/ps2x2pico #37: ps2x2pico initializes too slowly to be able to enter BIOS

ssokolow, 2024-04-25:
I just messed up the Win98SE install on my HP t5530 and, when I held F12 for "force netboot to the top of the boot order" to access my homebrew "Apple Internet Restore for DOS/Win9x PCs" so I could restore my backup, I discovered the hard way that none of those BIOS "hold X to Y" keys work, whether on a cold boot or if I warm boot via Ctrl+Alt+Del.

I'm not sure if this is fixable, but maybe there could be some kind of workaround where the ps2x2pico remembers whether a function key or Delete was being held down before a warm reboot and persists that state on the PS/2 side to the PC until USB device initialization completes and it has fresher data?

ssokolow, 2024-04-26:
There's only about a 1-second window when the BIOS is listening for keystrokes, if that. It's an early example of the "Don't worry about stuck key warnings. Just hold the key down while powering on to avoid missing the window." design you tend to see with UEFI.

No0ne, 2024-04-26:
Ok yes, but this should only matter on cold boots and not reboots. I don't think we can get this any faster. You will need to supply power otherwise so that its on before the PC.

ssokolow, 2024-04-26:
Holding F12 and then clicking to reboot from inside Windows 98 SE works to pull up my netboot environment with a genuine PS/2 keyboard but not with the ps2x2pico.

No0ne, 2024-04-26:
Ok hm 🤔 that should work though. I'll build you a debug version for this problem.

No0ne, 2024-04-26:
Yes, command 0xff and response 0xaa is implemented, I'll check if I can get a t5530 for debugging.

No0ne, 2024-04-26:
Got one for 35 euros so I can check for my self 🥳

bstrobel, 2024-04-26:
Your right. I just checked the diff and refreshed my memory. The project is 10y old and I did the fix 5y ago. How time flies... Anyway the bug was that I didn't send 0xfa (ACK) before sending the device id. I checked your code and you are doing that correctly. So no difference to what I did at the time.

Just wanted to say Thank you for this great little project. I always wanted to do something like that and never found the time. Great Work! :+1:

No0ne, 2024-04-26:
Should be the same: https://github.com/No0ne/ps2x2pico/blob/486a5950fcf86650e9b1b41b5542cac5ca1523e7/src/ps2kb.c#L238
I'll check when the t5530 arrives next week.

ssokolow, 2024-04-26:
> Got one for 35 euros so I can check for my self 🥳

Enjoy. It makes an excellent fanless late-Win98SE-era machine if you eBay a bigger ATA disk-on-module, install the appropriate drivers off VIA's website, and either use a USB CD-ROM drive or install a period-appropriate copy of DAEMON Tools. 

Heck, the only reason it's not the greatest for earlier stuff is that the onboard audio doesn't have native Soundblaster compatibility and it doesn't have an ISA or PCI slot to stick a genuine SoundBlaster into.

It's even got VESA mounting holes so you can hang it off the back of a monitor.

No0ne, 2024-05-11:
Still waiting on the t5530 order, but I haven't had time anyway at the moment.

bstrobel, 2024-05-16:
@ssokolow When working on issue #38 I refactored the code a bit and streamlined the host to keyboard communication. I don't think that the Pico initializes too slowly to be honest. If you want you could try the new version I created. I have attached it here:
[ps2x2pico.zip](https://github.com/No0ne/ps2x2pico/files/15332544/ps2x2pico.zip)

If you can it would be good if you would connect a serial terminal to the debug output and post the logs here. I would clarify what is going wrong. Each host has its own way of communicating and configuring the ps/2 keyboard. I tried and traced with a few and each one was different. Maybe yours is especially picky?

ssokolow, 2024-05-17:
I'll give it a look as soon as I've made time for retro-hobby stuff again. The last week or so has been pretty busy with spring cleaning and the like.

No0ne, 2024-05-19:
Finally got the t5530 and I found two potenial issues:
If the key is held down before tinyusb is running it may not be detected,
and (I think thats the real problem) if the host resets the keyboard (while rebooting) the repeating key is also resetted.
It works if you don't hold the key but mash it multiple times, but I think it should also work holding it down, yes!

No0ne, 2024-05-19:
Finally got the t5530 and I found two potenial issues:
If the key is held down before tinyusb is running it may not be detected,
and (I think thats the real problem) if the host resets the keyboard (while rebooting) the repeating key is also resetted.

It works if you don't hold the key but mash it multiple times, but I think it should also work holding it down!

And yes, the bios routing of the t5530 is really fast, did not see any bios this fast before.
As an alternative if you set 5sec POST delay in the bios it also works well.

I'll look into it that it works with holding the key down!

No0ne, 2024-05-19:
I think I got this, its working by holding down F10 or F12 on my t5530 without POST delay and quickpost enabled.
@ssokolow please try this: [t5530-fix.zip](https://github.com/No0ne/ps2x2pico/files/15369643/t5530-fix.zip)

ssokolow, 2024-05-21:
Will do. The last few days have been busy but I should have free time starting on Wednesday.

ssokolow, 2024-05-24:
Sorry I went silent. I bruised a rib the afternoon after I said "Will do" and it took me until now to be physically able to get under my desk to access the ps2x2pico.

It's now sitting on top of the desk where I can comfortably work on it and I should be able to make time to test it either today or within the next couple of days.

ssokolow, 2024-05-27:
So far, I literally haven't tested anything but the F10 and F12 keys the Ctrl+Alt+Delete key combo yet, but `t5530-fix.zip` does let me access the BIOS and force netboot on both a cold boot and a Ctrl+Alt+Delete warm boot by holding down the relevant key.

No0ne, 2024-05-29:
Cool, thanks for testing. So is this issue fixed now?

ssokolow, 2024-05-29:
Yes. As far as I can tell, `t5530-fix.zip` fixes it perfectly.

Thanks. :)

closed 2024-05-29 by ssokolow