No0ne/ps2x2pico #46: Use this with Mouse Only?

lnxsrt, 2024-06-22:
Is it possible to use this only for the Mouse? I have a PS/2 keyboard and would like to use this for the Mouse only. 

I'm seeing it register my USB mouse in the debug messages on GPIO0, but the machine does not recognize it.

No0ne, 2024-06-22:
Yes should work with only the mouse! Please post the debug messages.

lnxsrt, 2024-06-22:
[image]

No0ne, 2024-06-22:
Looks ok yes, please post a picture of the hardware and connections.

lnxsrt, 2024-06-22:
[image]

No0ne, 2024-06-22:
Also seems ok, which motherboard and OS do you have?

lnxsrt, 2024-06-22:
I've tried on my 2 socket 7 boards (P/I-P55T2P4 and a GMB something). DOS with ctmouse and Windows 98SE. Neither work. Both work fine with a PS/2 mouse.

lnxsrt, 2024-06-22:
Also just continuity checked again and everything matches the pinouts, no shorts, etc.

No0ne, 2024-06-22:
Huh, thats interessting!
I just checked with my HP t5530, this works with only the mouse.
But on my Gigabyte GA-5AX Socket 7 the mouse does not work without a ps/2 keyboard connected 🤔 

So I think this is more like a motherboard problem?

lnxsrt, 2024-06-22:
It doesn't work without a ps2x2pico PS/2 keyboard (ie USB)?

I have a PS/2 keyboard connected, just not through the ps2x2pico.

No0ne, 2024-06-22:
Ok strange, I think thats a different problem then. I'll send you a more verbose debug version.

No0ne, 2024-06-22:
Please post the serial debug of this version: 
[mousedebug.zip](https://github.com/user-attachments/files/15938215/mousedebug.zip)

lnxsrt, 2024-06-22:
[image]

No0ne, 2024-06-22:
Ok theres something wrong with the hardware, we don't get any bytes from the PC.
Should look like this:
```
HID(1,2,NONE) mounted
 ID: 1532:0098
 Manufacturer: Razer
 Product:      Razer DeathAdder Essential

host > ms ff
ms > host fa
ms > host aa
ms > host 00
** ps2in reset sm 01
host > ms ff
ms > host fa
ms > host aa
ms > host 00
** ps2in reset sm 01
host > ms ff
ms > host fa
ms > host aa
ms > host 00
** ps2in reset sm 01
host > ms e8
ms > host fa
host > ms 00
ms > host fa
host > ms e6
ms > host fa
host > ms e6
ms > host fa
host > ms e6
ms > host fa
...
```

Is clock and data mixed up?

lnxsrt, 2024-06-22:
Ok, so I re-checked continuity on all of the HV side levelshifter connections to the PS/2 socket. Everything is good there. But, this time I also checked continuity from the resistors on the levelshifter to the pins on the levelshifter. Turns out the mouse data LV side was not connecting the the resistor on the levelshifter board. The trace had somehow lifted or something. Reflowed that connection, confirmed continuity, and it is working now.

Thanks for the help! Never thought the levelshifter would have a funky trace.

No0ne, 2024-06-22:
Cool! Thanks for debugging :) Now I'll need to check my Socket 7 board 😆

closed 2024-06-22 by No0ne