No0ne/ps2x2pico #47: Debug press key

destroyer-dcf, 2024-07-23:
Hello, great project. Please could you indicate more specifically the configurations, connections and hardware necessary to be able to debug when a key is pressed. Thank you

No0ne, 2024-07-25:
You'll need a serial connection from GPIO0 (and GND) to another computer, either through a USB adapter cable or directly.

You can use a Debug Probe like this: https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#serial-connections

Or you can use Raspberry Pi and connect to GPIO15. Then enable the serial port via raspi-config, reboot and start `screen /dev/ttyAMA0 115200`

Or you can use another Pi Pico connected to GPIO17 and running this: https://github.com/Noltari/pico-uart-bridge

destroyer-dcf, 2024-08-05:
Thank you!!!.. debug working.... Another question, would it be possible for it to work on rp2040 Zero board

No0ne, 2024-08-05:
Cool! Yes, this should work with any rp2040 board. Maybe you have to compile it with your own different GPIOs.

closed 2024-08-05 by No0ne