No0ne/ps2x2pico #48: Not Found board_api.h

Chandler-Kluser, 2024-07-30:
Build process breaks in commit `404aaf02949d5bee8013e3b5d0b3239abf6e13bd`:

```
(...)
[ 26%] Building C object CMakeFiles/ps2x2pico.dir/src/ps2x2pico.c.obj
/home/chandler/Documentos/ps2x2pico/src/ps2x2pico.c:32:10: fatal error: bsp/board_api.h: No such file or directory
   32 | #include "bsp/board_api.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
(...)
```

No0ne, 2024-07-30:
Please update to the latest TinyUSB version first:
```
export PICO_SDK_PATH=/path/to/pico-sdk
cd $PICO_SDK_PATH/lib/tinyusb
git checkout 0.16.0
```

Chandler-Kluser, 2024-07-30:
Thanks, I was using `0.15.0`

but now I have got this build issue on [ps2x2pico.c](https://github.com/No0ne/ps2x2pico/blob/404aaf02949d5bee8013e3b5d0b3239abf6e13bd/src/ps2x2pico.c#L129):

```
/home/chandler/Documentos/ps2x2pico/src/ps2x2pico.c:129:3: error: implicit declaration of function 'tuh_deinit'; did you mean 'tuh_init'? [-Wimplicit-function-declaration]
  129 |   tuh_deinit(TUH_OPT_RHPORT);
      |   ^~~~~~~~~~
      |   tuh_init
make[2]: *** [CMakeFiles/ps2x2pico.dir/build.make:76: CMakeFiles/ps2x2pico.dir/src/ps2x2pico.c.obj] Error 1
```

No0ne, 2024-07-30:
Hm 🤔 I didn't get that error before, but this was an experiement.
Just comment out the tuh_deinit line an the tuh_init line below.

Chandler-Kluser, 2024-07-30:
commenting that line fully built the object and linked the binary, thank you!!

closing

closed 2024-07-30 by Chandler-Kluser