No0ne/ps2x2pico #49: error compilation

davidcf, 2024-08-07:
Hello, after installing the necessary dependencies, I am compiling following the steps indicated on an Ubuntu arm64. I have updated the tinyusb library to tag 0.16.0 but I get the following result....

-- Configuring incomplete, errors occurred!
parallels@ubuntu-linux-22-04-02-desktop:~/ps2x2pico/build$ export PICO_SDK_PATH=/home/parallels/pico-sdk
parallels@ubuntu-linux-22-04-02-desktop:~/ps2x2pico/build$ cmake ..
Using PICO_SDK_PATH from environment ('/home/parallels/pico-sdk')
PICO_SDK_PATH is /home/parallels/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
PICO target board is pico.
Using board configuration from /home/parallels/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
TinyUSB available at /home/parallels/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/parallels/pico-sdk/lib/btstack
cyw43-driver available at /home/parallels/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/parallels/pico-sdk/lib/lwip
mbedtls available at /home/parallels/pico-sdk/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parallels/ps2x2pico/build
parallels@ubuntu-linux-22-04-02-desktop:~/ps2x2pico/build$ make
Scanning dependencies of target bs2_default
[  1%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  2%] Linking ASM executable bs2_default.elf
[  2%] Built target bs2_default
[  3%] Generating bs2_default.bin
[  4%] Generating bs2_default_padded_checksummed.S
[  4%] Built target bs2_default_padded_checksummed_asm
[  5%] Creating directories for 'PioasmBuild'
[  6%] No download step for 'PioasmBuild'
[  7%] No update step for 'PioasmBuild'
[  9%] No patch step for 'PioasmBuild'
[ 10%] Performing configure step for 'PioasmBuild'
loading initial cache file /home/parallels/ps2x2pico/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/PioasmBuild-cache-Release.cmake
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parallels/ps2x2pico/build/pioasm
[ 11%] Performing build step for 'PioasmBuild'
[ 10%] Building CXX object CMakeFiles/pioasm.dir/main.cpp.o
[ 20%] Building CXX object CMakeFiles/pioasm.dir/pio_assembler.cpp.o
[ 30%] Building CXX object CMakeFiles/pioasm.dir/pio_disassembler.cpp.o
[ 40%] Building CXX object CMakeFiles/pioasm.dir/gen/lexer.cpp.o
[ 50%] Building CXX object CMakeFiles/pioasm.dir/gen/parser.cpp.o
[ 60%] Building CXX object CMakeFiles/pioasm.dir/c_sdk_output.cpp.o
[ 70%] Building CXX object CMakeFiles/pioasm.dir/python_output.cpp.o
[ 80%] Building CXX object CMakeFiles/pioasm.dir/hex_output.cpp.o
[... 90 lines cut ...]
[ 79%] Building CXX object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj
[ 80%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/src/rp2_common/pico_standard_link/binary_info.c.obj
[ 81%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/src/rp2_common/pico_stdio/stdio.c.obj
[ 82%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/src/rp2_common/pico_stdio_uart/stdio_uart.c.obj
[ 84%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/src/rp2_common/hardware_pio/pio.c.obj
[ 85%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c.obj
[ 86%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj
[ 87%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/host/usbh.c.obj
[ 88%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/host/hub.c.obj
[ 89%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/class/cdc/cdc_host.c.obj
[ 90%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/class/hid/hid_host.c.obj
[ 92%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/class/msc/msc_host.c.obj
[ 93%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/class/vendor/vendor_host.c.obj
[ 94%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/tusb.c.obj
[ 95%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/src/common/tusb_fifo.c.obj
[ 96%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/lib/tinyusb/hw/bsp/rp2040/family.c.obj
[ 97%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/src/rp2_common/pico_unique_id/unique_id.c.obj
[ 98%] Building C object CMakeFiles/ps2x2pico.dir/home/parallels/pico-sdk/src/rp2_common/hardware_flash/flash.c.obj
[100%] Linking CXX executable ps2x2pico.elf
**/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/ps2x2pico.dir/src/ps2x2pico.c.obj: in function `tuh_hid_umount_cb':
ps2x2pico.c:(.text.tuh_hid_umount_cb+0x1e): undefined reference to `tuh_deinit'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/ps2x2pico.dir/build.make:1111: ps2x2pico.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1513: CMakeFiles/ps2x2pico.dir/all] Error 2
make: *** [Makefile:91: all] Error 2**

davidcf, 2024-08-07:
Hello, solved... I was compiling with the main branch. By changing to tag 1.2, it has worked. Sorry and thank you.

closed 2024-08-07 by davidcf

No0ne, 2024-08-07:
If you want to compile the main branch you also need to checkout tinyusb to master:
```
export PICO_SDK_PATH=/path/to/pico-sdk
cd $PICO_SDK_PATH/lib/tinyusb
git checkout master
```