No0ne/ps2x2pico #50: The scroll wheel data still doesn’t appear on the PC.

zoushiyin, 2024-08-16:
Thank you for sharing. Even with the latest version, the scroll wheel data still doesn’t appear on the PC. I’ve tried multiple wired and wireless USB mice. I’m wondering if your new version has any improvements.

zoushiyin, 2024-08-18:
always report[3] is 0x0

No0ne, 2024-08-18:
In #35 im currently working on HID report mode instead of boot mode. It depends on the vendor if the wheel works in boot mode. It works on Razor mice and the Logitech Unifying receiver, but not on mice using the PixArt chip like the RaspberryPi mouse and the Microsoft Basic mouse.

You can try to enable report mode by enabling this line: https://github.com/No0ne/ps2x2pico/blob/f661daca5008d9cd7f7c2d683cfb6eb35f914751/src/ps2x2pico.c#L78

But until I finished a full HID parse in #35 your results may vary!

No0ne, 2024-11-05:
The latest pre-release should fix this: https://github.com/No0ne/ps2x2pico/releases/tag/pre-1.3

No0ne, 2024-11-08:
Confirmed working: https://github.com/No0ne/ps2x2pico/issues/35#issuecomment-2463402091

closed 2024-11-08 by No0ne