No0ne/ps2x2pico #51: 5V PS/2 side connection with separate keyboard and mouse port; is this correct?

nistvan86, 2024-08-20:
There's no wiring schematics which shows how the two PS/2 mini DIN jacks are wired when both the keyboard and mouse is being emulated. But I assume the 5V line powering the Pico (and the level shifter) should be only connected to one of the plugs in this case.
However the GND must be connected with both PS/2 jacks.

Is this correct?

No0ne, 2024-08-20:
Correct! I also connected both 5V lines together since they are connected on the motherboard anyway.
But you should not connect them together if keyboard and mouse are not connected to the same PC!

closed 2024-08-20 by No0ne

nistvan86, 2024-08-20:
Thanks!
Yeah, probably on the same machine it's fine to connect them together. But generally this is not a good idea.