No0ne/ps2x2pico #52: Mouse not working on 486, is working on Pentium 3

rmpel, 2024-09-20:
Hi. First off; fantastic project!

I have a question though; I don't seem to get the mouse working on my 486 computer.
It is working perfectly on my Pentium 3 laptop, however, the intended use case is on my 486 and 386 computers. On both, the mouse is not recognised.

Keyboard works perfectly.

I checked the wiring, it is all correct (of course, as it works on the P3 computer), I also tried the two alternate firmwares with fixed polling rates.

Tested with 4 mice, the mouse in the debug log below is a Dell branded Logitech mouse, and I also tested a Commodore Tank Mouse imitation (The one intended for the "The C64"), a Logitech MX something or other wireless and my Razor Viper Ultimate.
I have not used the serial debugging with the latter 3, but they respond the same on both computers as the Logitech. (they all work on the P3, none of them on the 486)

Debugging output on the 486 (dedicated mouse port, according to the case labeling);
```
ps2x2pico-1.2
Setting defaults for keyboard
scancodeset set to 2
** ps2in reset sm 00
kb > host aa
** ps2in reset sm 01
Blinking keyboard LEDs
HID(1,0,MOUSE) mounted
 ID: 046d:c077
 Manufacturer: Logitech
 Product:      USB Optical Mouse

HID(1,0,MOUSE) registered for reports
```

And on the pentium 3 (combined mouse/keyboard port);
```
ps2x2pico-1.2
Setting defaults for keyboard
scancodeset set to 2
** ps2in reset sm 00
kb > host aa
** ps2in reset sm 01
Blinking keyboard LEDs
HID(1,0,MOUSE) mounted
 ID: 046d:c077
 Manufacturer: Logitech
 Product:      USB Optical Mouse

HID(1,0,MOUSE) registered for reports
** ps2in reset sm 01
```

The only difference is the line;
```
** ps2in reset sm 01
```
at the end.

In both cases, there is no further output. I don't know if it should, but there is no indication of mouse movement or button clicking in the serial log.

Unfortunately, this knowledge doesn't help me solve the problem, so I hope you can give some pointers.

Thank you in advance :)

Remon.

rmpel, 2024-09-20:
Hi, thanks for responding :)

OS: MS-Dos 6.22, Win 3.11, DR DOS 6.0
Driver: any. (Logitech, cutemouse, microsoft mouse)
Mainboard: [Acer A1GX](https://theretroweb.com/motherboards/s/acer-a1gx).

I am quite sure OS and Drivers are not the issue, as the BIOS upon init already tells me "No pointing device" and when I have a PS/2 mouse connected "Pointing device installed".

No0ne, 2024-09-21:
[pre-1.3.zip](https://github.com/user-attachments/files/17083570/pre-1.3.zip)
Please post the serial output of the pre-1.3 version, it has more mouse debugging, thanks!

rmpel, 2024-09-21:
Hi.

On the 486;

```
ps2x2pico-1.3
Setting defaults for keyboard
scancodeset set to 2
** ps2in reset sm 00
kb > host aa
ms > host aa
ms > host 00
** ps2in reset sm 01
Blinking keyboard LEDs
HID(1,0,MOUSE) mounted
 ID: 046d:c077
 Manufacturer: Logitech
 Product:      USB Optical Mouse

HID(1,0,MOUSE) registered for reports
```

Bios reports "point device none", driver says, no mouse, 100% like before.

On the Pentium 3;

```
ps2x2pico-1.3
Setting defaults for keyboard
scancodeset set to 2
** ps2in reset sm 00
kb > host aa
ms > host aa
ms > host 00
** ps2in reset sm 01
Blinking keyboard LEDs
HID(1,0,MOUSE) mounted
 ID: 046d:c077
 Manufacturer: Logitech
 Product:      USB Optical Mouse

HID(1,0,MOUSE) registered for reports
host > ms f5
ms > host fa
host > ms f2
ms > host fa
ms > host 00
host > ms ee
ms > host fa
host > ms ff
ms > host fa
ms > host aa
ms > host 00
** ps2in reset sm 01
```
followed by a lot of 
```
ms > host 04
host > ms f3
ms > host fa
host > ms 64
ms > host fa
host > ms e8
ms > host fa
host > ms 03
ms > host fa
host > ms f4
```

No0ne, 2024-09-21:
hm, thats odd, has to do with timings 😟 
do you have a logic analyzer at hand?

No0ne, 2024-09-21:
I made the output clock slower by 50%, just a shot in the dark.
[slowclk.zip](https://github.com/user-attachments/files/17083888/slowclk.zip)

rmpel, 2024-09-21:
Well, it does more, but no solution yet :(

```
ps2x2pico-1.3-slow
Setting defaults for keyboard
scancodeset set to 2
** ps2in reset sm 00
kb > host aa
ms > host aa
ms > host 00
** ps2in reset sm 01
Blinking keyboard LEDs
HID(1,0,MOUSE) mounted
 ID: 046d:c077
 Manufacturer: Logitech
 Product:      USB Optical Mouse

HID(1,0,MOUSE) registered for reports
host > kb ff
KBHOSTCMD_RESET_FF
Setting defaults for keyboard
scancodeset set to 2
** ps2in reset sm 00
kb > host fa
kb > host aa
Blinking keyboard LEDs
host > kb f2
KBHOSTCMD_READ_ID_F2
kb > host fa
kb > host ab
kb > host 83
host > kb ed
KBHOSTCMD_SET_LEDS_ED
kb > host fa
host > kb 02
kb > host fa
```

Reg a logic analyzer; no, I don't, sorry. I have a very simple handheld scope (multimeter combo), but I don't think that will help much.

rmpel, 2024-09-21:
I connected the kb here, let me do a new test with just the mouse port connected, i think most in above is not mouse related.

rmpel, 2024-09-21:
Yeah, as expected. I will cleanup my responses for future reference and avoidance of confusion

rmpel, 2024-09-21:
I see five things of note in that topic;
1. Question if the Data and Clock are swapped; they aren't (confirmed by multimeter and by working on P3 laptop)
2. Question about continuity; continuity is tested and approved (this also by multimeter and working P3)
3. Suggestion that a KB present or not makes a difference; it does not, both a real PS/2 keyboard as well as a ps2x2pico keyboard
4. Concern on correct function of the level shifter; considering the P3 works, I think it safe to assume the level shifter is working properly (as mouse is working on the P3).
5. A different firmware; just to avoid the "should I try, yes, try" delay; tried that fw, no difference.

To confirm; I hope to keep the information here as on-topic as possible, but the 386 has been tested in each case as well and responds the same as the 486, so no need to duplicate all that data, but just know that the 486 is not the only one I test :)

rmpel, 2024-09-21:
Oh, just remembered, I should also have a 486 laptop somewhere, I haven't tested on that one, I'll have to dig it out, will do so if you need the 4th test-case ;)

No0ne, 2024-09-21:
Sorry, don't know whats wrong here :( Could you create some scope traces? Maybe I can see something.

rmpel, 2024-09-21:
I'm sorry, I tried, but unfortunately, I can't. 
The record function of my scope seems to only record for 10 frames at 100ns interval, and I cannot get it to record at the exact time the computer powers on. 

I havent had my scope for that long, I don't know how to operate those functions. My scope skills are practically non-existent.

I've also dug through the code, added more debug output, but all I can say is that the 486 is not sending the same commands as the P3 seems to do. Could the 486 use an older variant ps/2 protocol? Is there a way to fake the handshake and just go ;) 

i tried faking receipt of 0xFF and 0xF4, but that doesn't help. I haven't found out yet how to fake the entire sequence I see on the p3, being 0xF5 0xF2 0xEE 0xFF

the reason I tried F4 is because the code then should "Enable data reporting" according to the comments, but maybe I don't know what that means ;) ( Is "data reporting" mouse movement, or config data...)

I also tried to use commands to the ps/2 port using debug to try and get the mouse to respond, but the instructions I get from ChatGPT don't work well (instead of mouse being disabled and reenabled, my keyboard gets disabled, with, well, unhelpful results ;)

I haven't found a bios upgrade for my computer (not the 486 and also not the 386). If only I was more skilled in assembly, perhaps I could debug the BIOS to find out why the initialisation sequence isn't running. (I assume, even a 486 will want to know what mouse is present, so it should send F5 F2 at least...)

I appreciate the time and effort you have put into this and I'm sorry I'm not of much help.

No0ne, 2024-09-21:
As far as I know there are no protocol variants. The only thing which comes to mind is that I pushed the clock to the upper limit to achieve 200Hz polling under WinXP and up.

F4 Enable data reporting means mouse movement, right.

You can search for 8 channel logic analyzer, you can get one for under 10 euros.
If you want to debug this I think this is the only way. I used [Logic2](https://www.saleae.com/pages/downloads) and a cheap anlayzer for development.

rmpel, 2024-09-21:
Aight, I have a plan, but it will take some time
step 1, get a logic analyser
step 2, build an interposer (i have all the parts, just needs some soldering) to hook up de LA
step 3, learn how to use the LA and decode the PS/2 serial messages
step 4, capture and document the boot up sequence of the P3 (working) and the 486 (non working) ps2x2pico as well as a real mouse on both.

Step 1; I ordered this one `
https://www.amazon.nl/-/en/Youmile-Analyzer-Device-Channel-Arduino/dp/B09BTT94PC`
Says on the page it is compatible with saleae software, which is the program you linked above.
Will be here next week, but it will probably not be before next weekend until I get any kind of result ;)

Thanks again!

To be continued.

rmpel, 2024-09-22:
Step 2 done; interposer ready. (Cables not shown in photo ;)
[image]

( The PS/2 breakout boards are wired as Keyboard from factory, cut traces, rewired to the mouse pins )

No0ne, 2024-09-22:
Ah you have nice breakout boards, I just cut PS/2 extension cords 😂

rmpel, 2024-09-22:
Yeah, I got them to create dummy ISA card. The idea was (and still is) to expose 2 USB and 2 PS/2 ports at the back of the 486, allowing plugin in USB KB and Mouse and short PS/2 cables from the ports to the mainboard (maybe in future, if I feel brave enough, connect the PS/2 signalling directly to the mainboard.).
The mouse not working is a real spanner in the works. But I am hopeful we can get this fixed.

rmpel, 2024-09-27:
maybe found something. I think I interpreted the wiring diagram wrong

pins 2 and 6 are Mouse ... ON DUALPORT and normally not, I guess...

rmpel, 2024-09-27:
Confirmed. My own stupidity was causing the problems.

Turns out, the pentium 3 listens for mouse signals on both the normal and the alternate pins ... I will have to check, but it's probably a dualport without advertising it.

The 486 now has a USB keyboard and Mouse ;)

Very sorry for all the time wasted.

No0ne, 2024-09-27:
Ha, thats cool, I never thought that normal motherboards are wired for that. Only saw the dualport on notebooks and embedded systems.

Hope you'll find the logic analyzer useful anyway!

closed 2024-09-27 by No0ne