No0ne/ps2x2pico #53: Possibility of porting this for use in pi zero 2 W

WanderingNullPtr, 2024-09-30:
Hi,
Would it be realistic to port this for the Zero 2 W system too? I was what kind of pitfalls would one face for that.
It will definitely need some form of realtime priority, so perhaps set it to a single core with high priority?

The usecase I have for this is, if achievable, then a single Zero 2 W board would be enough to control systems with PS/2 ports using PiKVM.

Thanks for any input you can provide.

No0ne, 2024-09-30:
This is not possible as only the Raspberry Pico RP2040 and RP2350 have the necessary hardware PIO blocks.

closed 2024-09-30 by No0ne

WanderingNullPtr, 2024-09-30:
Right, the dependence on PIO blocks does make it exclusive to pico. On the zero sw bitbanging over GPIO seems like the only option. DMA accessing the GPIOs might help. I plan on investigating this once my zero arrives.