No0ne/ps2x2pico #55: missing Raspberry Pi Pico hat

hax0rbana-adam, 2024-10-13:
This project requires finding a male PS/2 cable and soldering it onto a level shifter board. Many users would prefer to just put a hat on their pico which has all the required electronics and call it a day. No soldering required.

This feature request also comes with a free solution! I made a hat in KiCAD, MIT licensed. Ready to be pulled in.

```
git remote add hax0r https://gitlab.hax0rbana.org/public-repos/ps2x2pico.git
git pull hax0r main
git push
```

I've already made a few of them. If anyone is interested in trying one of my prototypes, hit me up. I'm willing to ship them anywhere in America for the cost of shipping. Because what am I going to do with 9 of these hats? lol

No0ne, 2024-10-14:
Wow cool, thanks! Do you have a photo or render? I haven't done anything with KiCAD so far. Also pcbway contacted me for sponsorship some weeks ago. Maybe we could arrange something? As for shipping: I live in Austria.

hax0rbana-adam, 2024-10-15:
Sure, here are some photos:

[image]
[image]

Yeah, if PCBWay wants to sponsor you, you should go for it!

Shipping would be about $30 to Europe. Building 3 hats costs about that same price, so I'd suggest just making your own (and doubly so with a sponsorship).

I'd be happy to ship a prototype off to anyone in the USA that happens to stumble across this thread in the future until the extras are gone.

hax0rbana-adam, 2024-10-15:
I also just added the model for a 3D printable case to the repo. Feel free to scoop that up into this repo as well. I don't really want to maintain a fork. :rofl:

hax0rbana-adam, 2024-10-15:
Also, would it be okay if I submitted the hardware to be certified as open source by the Open Source Hardware Association (OSHWA)? The hardware is my own design, but it is entirely built based on the work done in this project, so it seems like it'd be rude for me to submit it for certification without coordinating with you first.

Note: When OSHWA certifies hardware they also look at the licenses for the software, libraries/dependencies, and documentation. MIT is fine, of course, but I just wanted to let you know what to expect if you give me a :+1:

No0ne, 2024-10-17:
Sure, go ahead! I assume the PS/2 is wired as a combo port? So you need a Y-cable if you want keyboard and mouse?
Is there space for another PS/2 port so the Y-cable is not needed?

I always thought of doing two hardware version. One not much bigger than the passive adapters, so it can only do keyboard or mouse. And a second one with an integrated USB hub like [this one](https://commons.wikimedia.org/wiki/File:Active_USB_to_PS2_Adatper_(keyboard%2Bmouse).jpg).

No0ne, 2024-10-19:
I found out that the PIO USB implementation only needs one PIO now: https://github.com/sekigon-gonnoc/Pico-PIO-USB/releases/tag/0.6.0
So maybe the HUB chip is not necessary any more for two devices.

hax0rbana-adam, 2024-10-19:
# Y-cable
Yeah, the PS/2 is wired as a combo port. It's funny, because once I realized that the PS/2 port on the back of my computer does keyboard AND mouse (thanks for that pinout diagram, BTW), I was excited to not have to buy a Y adapter.

I would need to move some things around, but I think I could get a second PS/2 port on there, and I guess if we just wired them both up as combo ports, they'd be interchangeable. As long as people don't do something wacky like plugging them into two different computers, it should be fine (and even then it might be okay).

# Integrated USB hub
I don't see any USB pins on the pico, and I'm not exactly sure what's going on with that Pico-PIO-USB project you linked to. It doesn't seem to have any electronics schematics and the video shows multiple picos being connected together instead of multiple devices plugged into a single host. I did see `pio_usb_host.c` though, so I'm guessing it'll make two GPIO pins into the data pins for a USB port (host mode).

If that's correct, I agree that it'd be very cool to use 4 GPIO pins to give us two female USB-A connectors without having any micro -> A adapter nor USB hub.

If you can integrate the code and make a prototype on a breadboard/protoboard, I should be able to implement it in KiCAD.

# Smaller design
I wanted to put the PS/2 port between the pico and the hat, but unfortunately it doesn't fit. The PS/2 port is 13 mm tall ([datasheet](https://www.mouser.com/datasheet/2/222/KMDGX-334716.pdf)), and female headers are 8.5 mm tall.

If we can find any 1x20 female headers with an extra 5 mm spacer, that would give us enough room. I could find them in 2x20, but not 1x20. The last picture [in this listing](https://www.amazon.com/Extra-Female-Stacking-Header-Raspberry/dp/B01IRRCEBK) shows the total length above the PCB to be 13.5 mm, and [this other listing](https://www.amazon.com/2x20-pin-Female-Stacking-Header-Raspberry/dp/B0756KM3GB/133-7665531-1269736) has the right size tails and is actually available. But spending $10 on headers seems like a bit much, doubly so considering they're twice as wide as we actually want. :grimacing: 

The other option that comes to mind is to just use some 1x20 [stackable headers](https://www.sparkfun.com/products/14311) in there as a spacer, but I'm not sure that'd be much shorter than the current design.

In summary, I think it's going to be tough to make it any shorter.

# Final thoughts
FWIW, I'm personally more excited about having a functional scroll wheel #35 on any of my mice than prioritizing improvements on the hat. However, I also understand how open source development works and I'm not ready to drop other things and write it myself. Just figured I'd get my vote in. :slightly_smiling_face:

No0ne, 2024-10-19:
True! I also want #35 / #50 to be resolved as it is the last "big" bug right now. But its on the edge of being too complicated for me, but I'll try again.

I'm simultaneously working on the ITX-Llama's code as they basically integrated a ps2x2pico on board: https://github.com/eivindbohler/itxllama?tab=readme-ov-file#ps2-or-usb-hid
There I discoverd the second USB port via GPIO.

Yes you are right, the built-in USB port has no header pins, only testpoints under the micro usb plug. Some other projects contacted them with springloaded pogopins.

But for the custom "small adapter plug" and "full featured board" actually thought of not using the Pico but rather the RP2040 it self to got them really small. Is that something you can work with?

No0ne, 2024-10-19:
Also the "extra full featured board" could also incorporate the PS/2 input passthru for #24 / https://github.com/pikvm/pikvm/issues/476 and an extra USB plug to make it work for https://docs.pikvm.org/pico_hid_bridge/ or https://docs.pikvm.org/pico_hid/

No0ne, 2024-10-19:
Now the Feature creep is kind of getting started 😳

No0ne, 2024-10-19:
Ha! That totally works!! Just tested with 4 GPIOs, got a keyboard and mouse working without a HUB and the built-in USB port. 🥳

hax0rbana-adam, 2024-10-19:
I've done USB-C plugs that use USB 2.0, so that's certainly something I could do with the standard 4-wire USB ports. That just requires a 56K ohm resistor. Easy. :smiley: 

As for using the RP2040 directly, as long as you can make a prototype and give me the schematics of what needs to connect where, absolutely. If not, maybe we can leverage another open source hardware project that already has a RP2040 symbol and footprint and any supporting components (resistors, capacitors, etc.) to avoid having to do extra work and make the same mistakes over again.

I tried loading the KiCAD files from itxllama, but it was created with an unstable version of KiCAD that is not compatible with the stable version. So I can't open the file right now. I can spin up another VM with Debian testing or unstable at some point to see if I can at least see it and understand what that project is. There's no description and the README just jumps right into "how to build your own" without explaining "build your own what?" :laughing:

No0ne, 2024-10-19:
Maybe the picogus has something useful? https://github.com/polpo/picogus/tree/main/hw-chipdown

hax0rbana-adam, 2024-10-19:
Apparently everyone likes to use bleeding edge versions of KiCAD that the stable version can't open. Seeing RP2040.kicad_pro sure looks promising though. :grin:

No0ne, 2024-10-19:
Maybe consider the RP2350 as I need three PIOs for PS/2 out, in and USB.

hax0rbana-adam, 2024-10-22:
The Pico Hat is now certified Open Source Hardware! https://certification.oshwa.org/us002692.html I can update the Project Website link on that page to point to this GitHub repo once the hardware files get merged in.

hax0rbana-adam, 2024-10-22:
I created #57 for a v1.3 of the hardware/software. That'll split off the additional features from what already has been done.

No0ne, 2024-10-23:
Hardware files are merged, thanks!

closed 2024-10-23 by No0ne

No0ne, 2024-11-05:
> True! I also want #35 / #50 to be resolved as it is the last "big" bug right now. But its on the edge of being too complicated for me, but I'll try again.
> 

The latest pre-release should fix this: https://github.com/No0ne/ps2x2pico/releases/tag/pre-1.3