No0ne/ps2x2pico #57: v1.3 of the Raspberry Pi Pico hat

hax0rbana-adam, 2024-10-22:
This ticket came out of a conversation on #55.

I propose a v1.3 of the hardware/software which is a hat for the Raspberry Pi Pico 2 (RP2350) that includes 2 PS/2 ports (both dual wired), 2 USB-A ports, and (if possible) 2 USB-C ports. That should eliminate the need for all adapters regardless of what type of connectors your HIDs have and what port(s) your computer has.

What I need is a code base that I can build and flash onto a Raspberry Pi Pico 2. I'll try to build a prototype that can demonstrate that the electronics work as expected and then design a circuit in KiCAD and a PCB to match. Once that's done, I'll have some boards made, assemble them and find out if I made any mistakes. Then possibly iterate on this process several times until all the kinks are worked out.

Once we have a solid version 1.3 board, I design a new case, and we commit all the new files to this repo.

No0ne, 2024-10-23:
That sounds great! I'll have to get a Pico 2 for myself too.

hax0rbana-adam, 2024-12-05:
I now have a pico2 and am prepared to test when it's ready. A few questions & comments:

1. Will we need another repo, or another branch for the software? Or can we just handle it with something like `#ifdef`s?
2. If we don't need another branch/repo, can we combine the ps2pico with this repo? Seems like it'd be nice to reduce the duplicated code if possible.
3. Do you expect it will be pin-compatible? It'd be nice if the existing hat could also work with either the pico or the pico2. And the v1.3 hat could work with either as well, but just have extra features with the pico2

I see some exciting progress over there on #65 and WOW that RP2040-Zero by Waveshare is **tiny**!

No0ne, 2024-12-05:
3. the Pico-SDK 2.0 and 2.1 still seems a bit buggy with TinyUSB, replugging devices doesn't work and too much serial output locks up my PS/2 code 🤷🏻

javavi, 2024-12-05:
I think it is not worth wasting the power of Pico2 (RP2350) for the adapter. If RP2040 is enough for this task, then we should continue to do it on it, especially since they have become cheaper and will be sold out.

javavi, 2024-12-05:
Here are two examples of adapters I made for the GP2040-Zero.
Yes, it would be great to develop a compact and elegant PCB design for such an adapter that includes several connection options.
[image]