No0ne/ps2x2pico #59: Cursor Movement and Button Functionality Limited （only move up and down）for Certain Mice

zoushiyin, 2024-11-13:
We have noticed that some wireless mice, such as the Microsoft Ergonomic Mouse (which uses a Bluetooth receiver) and the SANWA SUPPLY mouse (which uses a 2.4G receiver), are experiencing issues with the PS2 to USB adapter. Specifically, the cursor can only move up and down, and the mouse buttons do not function at all. However, wired mice work normally with the adapter. Additionally, keyboards are functioning correctly.

No0ne, 2024-11-13:
Did you try it with version 1.2 or the 1.3-pre release?

No0ne, 2024-11-14:
I discoverd a bug in the 1.3-pre release, I'll make another one soon.

No0ne, 2024-11-14:
Please try the new pre-release: https://github.com/No0ne/ps2x2pico/releases/tag/pre-1.3

No0ne, 2024-11-20:
Ah I've got a mouse now so I can reproduce this, I'll get back to you!

No0ne, 2024-11-20:
Please try this: https://github.com/No0ne/ps2x2pico/issues/61#issuecomment-2488690026

ssokolow, 2024-11-20:
For the record, that sounds exactly like the behaviour I was getting out of the second mouse when I plugged in two mice with 1.3-pre and the `reportmode-test.zip` file at the linked comment fixed it... so definitely test it and see if it fixes it for you too.

No0ne, 2024-11-21:
Should be fixed in the latest pre-release: https://github.com/No0ne/ps2x2pico/releases/tag/pre-1.3
If not please leave a comment.

closed 2024-11-21 by No0ne