No0ne/ps2x2pico #60: Mouse not working on combination PS/2 port

Justasic, 2024-11-14:
Hello,

I have a combination PS/2 Mouse/Keyboard port on my motherboard and as such I have just soldered a single male connector to the pico. I plugged a keyboard into a USB hub which uses an OTG cable on the pico. The keyboard works perfectly as expected but when I try to use a USB mouse, it does not. If I plug a USB mouse directly into the OTG connector, it still does not function.

I connected to the serial console and I can see the keyboard is recognized and sends bytes on button presses but when I do the same with a mouse (even without a USB hub) I see it recognize the mouse as a USB device but it doesn't specify the manufacturer or product, it also does not send any data to the host when I move the mouse or click it's buttons. 

I have tried at least 5 different brands/models of mice to confirm that this isn't a model-specific issue but all of them report no manufacturer, no product, and don't function or show any information about being recognized outside of stating it's registered a new USB device. 

Is there something I'm doing wrong or can test to debug further? I tried using the ps2x2pic version 1.2 100 Hz, the standard version 1.2 and the pre-release version 1.3 to see if that changed anything and unfortunately it did not.  

Thanks!

No0ne, 2024-11-14:
Please try the new 1.3-pre release: https://github.com/No0ne/ps2x2pico/releases/tag/pre-1.3

No0ne, 2024-11-14:
Which exact models have you tested?

Justasic, 2024-11-14:
Tested mice:
- Lenovo MOJUUO
- Microsoft "Optical Mouse Blue USB and PS/2 Compatible" (no model, has a P/N of X08 - 72983)
- HP gaming keyboard and mouse (mouse only was tested) km100
- Microsoft basic optical mouse v2.0 model 1113
- "Optical Mouse" M700 (seems like a generic no-name brand)

I work in a PC shop so I have access to many more mice if necessary. I used the 1.3-pre for testing all of these. 

The serial console output looks like this for each of them.
```
HID(1,0,KEYBOARD) mounted
 ID: 045e:0752
 Manufacturer: Microsoft
 Product:      Wired Keyboard 400

HID(1,0,KEYBOARD) regystered for reports
HID has 1 reports
HID(2,1,MOUSE) mounted
 ID: 045e:00cb
 Manufacturer: 
 Product:      

HID(2,1,MOUSE) registered for reports
HID(2,1) unmounted
HID has 1 reports
HID(2,1,MOUSE) mounted
 ID: 1c4f:0034
 Manufacturer: 
 Product:      

HID(2,1,MOUSE) registered for reports
HID(2,1) unmounted
HID has 1 reports
HID(2,1,MOUSE) mounted
 ID: 1c4f:0048
 Manufacturer: 
 Product:      

HID(2,1,MOUSE) registered for reports
HID(2,1) unmounted

```

No0ne, 2024-11-14:
🤔 I have a mouse with the same IDs which works:
```
HID has 1 reports
HID(1,0,MOUSE) mounted
 ID: 045e:00cb
 Manufacturer: PixArt
 Product:      Microsoft USB Optical Mouse

HID(1,0,MOUSE) registered for reports
```
Which Pico board do you have?

No0ne, 2024-11-14:
Did you try the new pre-release I posted minutes ago?

Justasic, 2024-11-14:
Alright, I installed the pre-release you just made and tested it, it still seems non-functional and I used the OTG cable itself to test. It seems to be able to detect the mice models/manufacturers now but still not send any data to the host.

Output:
```
Blinking keyboard LEDs
HID(1,0,KEYBOARD) mounted
 ID: 045e:0752
 Manufa,W���� Microsoft
 Product:      Wired Keyboard 400

HID(1,0,KEYBOARD) registered for reports
HID has 1 reports
HID(2,1,MOUSE) mounted
 ID: 17ef:608d
 Manufacturer: PixArt
 Product:      Lenovo USB Optical Mouse

HID(2,1,MOUSE) registered for reports
HID(2,1) unmounted
HID(1,0) unmounted
HID has 1 reports
HID(1,0,MOUSE) mounted
 ID: 17ef:608d
 Manufacturer: PixArt
 Product:      Lenovo USB Optical Mouse

HID(1,0,MOUSE) registered for reports
HID(1,0) unmounted
HID has 1 reports
HID(1,0,MOUSE) mounted
 ID: 045e:0053
 Manufacturer: Microsoft
 Product:      Microsoft 3-Button Mouse with IntelliEye(TM)

HID(1,0,MOUSE) registered for reports
HID(1,0) unmounted
```

No0ne, 2024-11-14:
The mouse data to the host is only sent if the host enables mouse streaming mode (0xf4) via the PS/2 port.
So you have to see the following first in the debug log, try rebooting your host:
```
host > ms ff
ms > host fa
kb > host f0
kb > host 5a
ms > host aa
ms > host 00
host > ms e8
ms > host fa
host > ms 03
ms > host fa
host > ms f2
ms > host fa
ms > host 00
host > ms f3
ms > host fa
host > ms c8
ms > host fa
host > ms f3
ms > host fa
host > ms 64
ms > host fa
host > ms f3
ms > host fa
host > ms 50
ms > host fa
host > ms f2
ms > host fa
ms > host 03
host > ms ff
ms > host fa
ms > host aa
ms > host 00
host > ms f3
ms > host fa
host > ms c8
ms > host fa
host > ms f3
ms > host fa
host > ms 64
ms > host fa
host > ms f3
ms > host fa
host > ms 50
ms > host fa
host > ms f4
```

Justasic, 2024-11-14:
I've rebooted my host and the mouse is not working, keyboard still functions though. I wonder if maybe I've swapped the pins in the PS/2 port. I have pin 2 (based on the plug pin diagram in the readme) going to GPIO 11 and pin 6 going to GPIO 12. I would assume these are correct. I checked my BIOS for any PS/2-related settings but it didn't seem to have any. 

Actually, this may be something else, I tried with a real PS/2 mouse in the PS/2 port and it's not working as well. *sigh* maybe this is some form of user error :sweat_smile: I'll try rebooting again and see if the PS/2 mouse is recognized.

No0ne, 2024-11-15:
A mouse is normally not wired for a dual-port, try a PS/2 Y-splitter if you have one.
Does it work now?

Justasic, 2024-11-15:
Ok after some testing, the port works with a PS/2 mouse but only after I reboot the machine. When I try to plug it in after having a keyboard plugged in or if I try to use a splitter cable, the PS/2 mouse stops working. If I try to plug the pico pi into the port, I get no mouse even after a reboot and only having a mouse plugged in. I understand that PS/2 is not hot-swap, I just find it interesting it doesn't seem to be initializing a mouse correctly even at boot.

For more information: I have an ASUS X670-p wifi motherboard and I've read online that there seems to be some issues with ASUS's support of PS/2 hardware. The rational behind me using PS/2 is because I use QubesOS.

No0ne, 2024-11-15:
Do you see any **host > ms** messages in the debug log?

Justasic, 2024-11-20:
Apologies for the delay, things got busy. I did a few things, first being I plugged the device in and it gave me this output when I plugged it in:
```
bash-5.2# cat /dev/ttyUSB0 
�

ps2x2pico-1.3

Setting defaults for keyboard

scancodeset set to 2

kb > host aa

ms > host aa

ms > host 00

Blinking keyboard LEDs

HID has 1 reports

HID(1,0,MOUSE) mounted

 ID: 17ef:608d

 Manufacturer: PixArt

 Product:      Lenovo USB Optical Mouse

HID(1,0,MOUSE) registered for reports
```
Then this output when I rebooted the host and it was in the BIOS:
```
HID(1,0,MOUSE) registered for reports

host > kb f2

KBHOSTCMD_READ_ID_F2

kb > host fa

kb > host ab

kb > host 83

host > kb ed

KBHOSTCMD_SET_LEDS_ED

kb > host fa

host > kb 02

kb > host fa
```
Then after the OS kernel was executed, I got more output:
```
host > kb f2

KBHOSTCMD_READ_ID_F2

kb > host fa

kb > host ab

kb > host 83

host > kb f5

KBHOSTCMD_DISABLE_F5

kb > host fa

host > kb ed

KBHOSTCMD_SET_LEDS_ED

kb > host fa

host > kb 00

kb > host fa

host > kb f3

KBHOSTCMD_SET_TYPEMATIC_PARAMS_F3

kb > host fa

host > kb 00

kb > host fa

host > kb f4

KBHOSTCMD_ENABLE_F4

kb > host fa
```

However, the mouse did not function.

No0ne, 2024-11-20:
As I excepted, there is communication from the host, *host > ms* is missing in your log.
This sounds similar to https://github.com/No0ne/ps2x2pico/issues/52 and most certainly is a hardware/wiring problem I think.

No0ne, 2024-11-20:
As I excepted, there is no communication from the host, **host > ms** is missing in your log.
This sounds similar to https://github.com/No0ne/ps2x2pico/issues/52 and most certainly is a hardware/wiring problem I think.

Justasic, 2024-11-20:
hmmm okay, that is a bit puzzling as I tested the wires with a multimeter. I'll review everything and see if maybe I swapped pins or something weird. Thanks for your help and patience! I'll update this issue with what I find (for future folks who may have the same issue)

hax0rbana-adam, 2024-11-30:
Hi there. I'm here to add some fuel to this fire. I have the same issue with both v1.3 and the previous version. 

I've varied enough things, one at a time, to narrow it down to some kind of incompatibility between the motherboard and the ps2x2pico. The ps2x2pico isn't broken, nor is the motherboard.

Here's what I'm seeing:

- Computer 1: USB keyboard & mouse (I tried multiple, but the one I actually want to use is a Reaper ZMS-1000)
  - Motherboard: ASRock B450M Pro4
  - Keyboard and mouse work fine in Qubes
- Computer 2: same keyboard, mouse, pico, hat, and PS/2 cable as before
  - Asus ProArt B550-creator
  - Mouse doesn't work. It's as if it isn't plugged in at all
  - Keyboard works fine
  - Tried it with Qubes, Tails, and in the BIOS
  - Plugging in an actual PS/2 mouse works in the BIOS
    - The PS/2 mouse is a Logitech M-SBJ96

I understand it's totally unreasonable to expect anyone to buy a $240 motherboard and the a CPU and some RAM to go into it just to test a bug report. My hope is that we can collect a list of motherboards which exhibit this behavior and maybe some pattern will emerge.

No0ne, 2024-12-01:
Can you try a PS/2 splitter an plug in a real keyboard an mouse at the same time on the B550?

No0ne, 2024-12-01:
I suspect your port and @Justasic can switch between mouse an keyboard but can't do both simultaneously, just a theory!

hax0rbana-adam, 2024-12-01:
I've ordered a splitter to test that theory.

I'm skeptical since using a USB keyboad directly and the ps2x2pico for the mouse doesn't work in the BIOS. It really seems like the ps2x2pico can't output PS/2 mouse signals in a way this motherboard is willing to accept.

Either way, the splitter test should sort this out in short order. I will report back in about a week.

No0ne, 2024-12-02:
If two real PS/2 devices are indeed working and you want to debug this further we also need a cheap 8-channel logic analyzer: https://github.com/No0ne/ps2x2pico/issues/52#issuecomment-2365244262

Justasic, 2024-12-02:
Some initial testing, I have ordered 2 separate splitters and using the keyboard and mouse (both of which worked as expected independently) it seems only the keyboard wants to work. I plugged the mouse in first and then the keyboard but only the keyboard functioned. I have not rebooted my system yet (I'll do that later during lunch) to confirm whether or not it's a motherboard issue. 

It will take me a bit of time to get the logic analyzer (likely after Christmas) so if someone else can do that testing, it'd be appreciated :sweat_smile:

Justasic, 2024-12-02:
Yeah, it seems I can't use the keyboard and mouse with a splitter. I've tried two different splitters and I know the keyboard and mouse work independently. Not sure what the point is of putting a dual-port on when it only supports single device communication :/

Anyway, thank you for your help! It may be helpful to add some kind of note to the readme that some motherboards may not support both devices at the same time so future folks aren't confused about why their device doesn't work. I'll let you close this issue in case you want to keep it open to track similar issues others are having in this thread.

I'll see if I can do some more research and maybe find another solution but I doubt it. :grin:

No0ne, 2024-12-03:
Thanks for your research. Does the mouse work on either of the two splitter ports? (so they wired all 6 pins) Does the keyboard also work on both splitter ports?

hax0rbana-adam, 2024-12-05:
My splitter arrived and I have test results: my motherboard only supports a PS/2 keyboard XOR a PS/2 mouse.

I'm offended by this and I hold Asus responsible for my feelings of shock and horror. They really couldn't spend the extra few bucks to support both concurrently? Sheesh.

The splitter I have doesn't label the keyboard side and the mouse side, but it seems to matter. I didn't bust out a multimeter to check for continuity, but it seems that not all 6 pins are wired to both sides. That'd make a lot more sense if at least side one was labeled or color coded, but it was cheap so I guess I got what I paid for.

I'm not sure why I couldn't get my mouse to work in the BIOS with the ps2x2pico (while using a USB keyboard), but I'm going to say it was probably testing error. I'm too discouraged to reboot and tinker with it now that I know the horrible truth about my motherboard's limitations.

I agree with @Justasic that this ticket can be closed by adding "some kind of note to the readme that some motherboards may not support both devices at the same time". I'd also suggest mentioning that this doesn't seem to be correlated to price so people don't fall into the mental trap of "oh, I have a really nice motherboard, of course it'll support both at the same time". Maybe mention the Asus ProArt B550-creator as an example?

At any rate, I am glad we got to the bottom of this. It's probably more frustrating to have these unsolved mysteries and open tickets that just hang around forever.

No0ne, 2024-12-05:
Yeah thats bad :( I'll add that to the readme.

Normally a splitter should be wired like this:
| Plug    | Keyboard Socket | Mouse Socket |
|---------|-----------------|--------------|
| 1 Data KB  | 1 Data          | -            |
| 2 Data MS | -               | 1 Data       |
| 3 GND   | 3 GND           | 3 GND        |
| 4 +5V   | 4 +5V           | 4 +5V        |
| 5 Clock KB | 5 Clock         | -            |
| 6 Clock MS | -               | 5 Clock      |

closed 2024-12-05 by No0ne