No0ne/ps2x2pico #62: Support for Beirui Kone A2 Remote Control Hardware

zoushiyin, 2024-11-22:
I'm interested in trying out a remote control hardware: Beirui Kone A2. Does this device have support? If not, are there any plans to add support for it?

Thank you!

No0ne, 2024-11-22:
Please connect a serial cable to GPIO0 and post the debug output while plugging the device in.

0x504B0304, 2024-11-22:
here is it's hid report descriptor:
protocol:2
0x05 0x01 0x09 0x02 0xa1 0x01 0x85 0x01 0x09 0x01 0xa1 0x00 0x05 0x09 0x19 0x01 0x29 0x03 0x15 0x00 0x25 0x01 0x95 0x03 0x75 0x01 0x81 0x02 0x95 0x01 0x75 0x05 0x81 0x03 0x05 0x01 0x09 0x38 0x15 0x81 0x25 0x7f 0x75 0x08 0x95 0x01 0x81 0x06 0x09 0x30 0x09 0x31 0x15 0x00 0x26 0xff 0x7f 0x35 0x00 0x46 0xff 0x7f 0x75 0x10 0x95 0x02 0x81 0x02 0xc0 0xc0 
pid 572b,vid 0483
HID device address = 1, instance = 1 is mounted
protocol:2
0x05 0x01 0x09 0x02 0xa1 0x01 0x85 0x02 0x09 0x01 0xa1 0x00 0x05 0x09 0x19 0x01 0x29 0x03 0x15 0x00 0x25 0x01 0x95 0x03 0x75 0x01 0x81 0x02 0x95 0x01 0x75 0x05 0x81 0x03 0x05 0x01 0x09 0x38 0x15 0x81 0x25 0x7f 0x75 0x08 0x95 0x01 0x81 0x06 0x09 0x30 0x09 0x31 0x16 0x01 0x80 0x26 0xff 0x7f 0x75 0x10 0x95 0x02 0x81 0x06 0xc0 0xc0 
pid 572b,vid 0483
HID device address = 1, instance = 2 is mounted
protocol:1
0x05 0x01 0x09 0x06 0xa1 0x01 0x05 0x07 0x19 0xe0 0x29 0xe7 0x15 0x00 0x25 0x01 0x95 0x08 0x75 0x01 0x81 0x02 0x95 0x01 0x75 0x08 0x81 0x03 0x95 0x06 0x75 0x08 0x15 0x00 0x25 0xff 0x05 0x07 0x19 0x00 0x29 0x65 0x81 0x00 0x25 0x01 0x95 0x05 0x75 0x01 0x05 0x08 0x19 0x01 0x29 0x05 0x91 0x02 0x95 0x01 0x75 0x03 0x91 0x03 0xc0

No0ne, 2024-11-22:
Using http://eleccelerator.com/usbdescreqparser/ this looks like a normal mouse+keyboard combo.
What does it do with the latest 1.3 pre-release?

No0ne, 2024-11-22:
No, absolute mode in PS/2 is implemented in proprietary drivers like for older laptop touchpads.

zoushiyin, 2024-11-23:
I only see the report for the keyboard, not the mouse. However, when I move the mouse, the cursor jumps to the bottom-right corner. It seems that the keyboard is working normally. Here is part of the report:

HID(1,0,MOUSE) mounted
 ID: 0483:572b
HID has 1 reports
 Manufacturer: STMicroelectronics
 Product:      OrayKVM Mouse Keyboard

HID(1,0,MOUSE) registered for reports
HID(1,1,MOUSE) mounted
 ID: 0483:572b
HID has 1 reports
 Manufacturer: STMicroelectronics
 Product:      OrayKVM Mouse Keyboard

HID(1,1,MOUSE) registered for reports
HID(1,2,KEYBOARD) mounted
 ID: 0483:572b
HID has 1 reports
 Manufacturer: STMicroelectronics
 Product:      OrayKVM Mouse Keyboard

HID(1,2,KEYBOARD) registered for reports
kb > host e0
kb > host 1f
kb > host e0
kb > host f0
kb > host 1f
kb > host e0
kb > host 1f
kb > host e0
kb > host f0
kb > host 1f
kb > host e0
kb > host 1f
kb > host e0
kb > host f0
kb > host 1f
kb > host e0
kb > host 1f
kb > host e0
kb > host f0
kb > host 1f
kb > host e0
kb > host 1f
kb > host e0
kb > host f0
kb > host 1f
kb > host e0
kb > host 1f
kb > host e0
kb > host f0
kb > host 1f
kb > host e0
kb > host 1f
kb > host e0
kb > host f0
kb > host 1f
kb > host e0
kb > host 1f
kb > host 23
kb > host f0
kb > host 23

Thank you for your help!

No0ne, 2024-11-23:
This log output only says you pressed the Win+D shortcut.
As I see in the second HID report descriptor there is also a relative mode. You will need to use this, absolute mode is not going to work.

zoushiyin, 2024-11-23:
I modified ps2ms.c to set ms_streaming to true in an attempt to get the mouse pointer position feedback. However, when I recompile, I encounter the following error:

[ 25%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (0.4s)
-- Generating done (0.0s)
-- Build files have been written to: /home/zoushiyin/pico/ps2x2pico/build/elf2uf2
[ 26%] Performing build step for 'ELF2UF2Build'
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[ 27%] No install step for 'ELF2UF2Build'
[ 28%] Completed 'ELF2UF2Build'
[ 28%] Built target ELF2UF2Build
[ 30%] Building C object CMakeFiles/ps2x2pico.dir/src/ps2x2pico.c.obj
/home/zoushiyin/pico/ps2x2pico/src/ps2x2pico.c:32:27: fatal error: bsp/board_api.h: No such file or directory
 #include "bsp/board_api.h"
                           ^
compilation terminated.
CMakeFiles/ps2x2pico.dir/build.make:75: recipe for target 'CMakeFiles/ps2x2pico.dir/src/ps2x2pico.c.obj' failed
make[2]: *** [CMakeFiles/ps2x2pico.dir/src/ps2x2pico.c.obj] Error 1
CMakeFiles/Makefile2:1496: recipe for target 'CMakeFiles/ps2x2pico.dir/all' failed
make[1]: *** [CMakeFiles/ps2x2pico.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

Could you please help me understand what I might be doing wrong? Thank you!

No0ne, 2024-11-23:
Make shure you are on Pico-SDK 1.5.1 and TinyUSB 0.17.0, see: https://github.com/No0ne/ps2x2pico?tab=readme-ov-file#build

zoushiyin, 2024-11-27:
This is the data reported by my device, and all the data is showing up in the bottom right corner.
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 254, Y = 254, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x01
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00

No0ne, 2024-11-27:
What kind of device is this? If I search for "Beirui Kone A2" nothing shows up.

0x504B0304, 2024-11-27:
> What kind of device is this? If I search for "Beirui Kone A2" nothing shows up.

https://sunlogin.oray.com/hardware/kongkong2
It's a commercial kvm hardware, just like pi kvm.

No0ne, 2024-11-27:
Ahh, ok! Does it support relative mouse movements?

zoushiyin, 2024-11-27:
This A2 device is used for forwarding keyboard and mouse data, meaning it forwards the data from the controlling end to the controlled end. For the ps2x2pico, the A2 acts as an input device. The data you see above is being sent from the A2 to the A2.
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 254, Y = 254, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 254, Y = 254, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 127, Y = 127, Z = 0, Buttons = 0x00
Mouse Move: X = 254, Y = 254, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
keyboard  len: 08  id: 00 items: 15
RPT8 00 00 09 00 00 00 00 00
kb > host 2b
keyboard  len: 08  id: 00 items: 15
RPT8 00 00 00 00 00 00 00 00

No0ne, 2024-11-28:
I will try it with PiKVM (not using the Pico HID bridge) which does absolute and relative mouse on USB.

No0ne, 2024-11-28:
[image]
If I switch to relative mode, it is indeed working. Does your KVM have such a switch?

zoushiyin, 2024-11-28:
I don't see support for absolute and relative mouse movements in this device. Where in the code can I print out the raw received data? I will send you the code for you to take a look. Thank you.

0x504B0304, 2024-11-28:
> I don't see support for absolute and relative mouse movements in this device. Where in the code can I print out the raw received data? I will send you the code for you to take a look. Thank you.

there do have a relative mode switch in sunlogin client software:
[image]

No0ne, 2024-11-28:
Without relative mode it will not work, as I said PS/2 doesn't support absolute mode without special drivers.

The raw data is coming in here: https://github.com/No0ne/ps2x2pico/blob/main/src/usbin.c#L645
You can printf the report bytes [0] to [x], the length is in the parameter len.

zoushiyin, 2024-11-28:
I'm experiencing a similar issue with my A2 device. When I quickly move the cursor back and forth on the controlling end, I can see the cursor moving on the controlled end, but it quickly moves to the corner. I was thinking of using this approach to modify the code, but I'm having trouble getting it to compile without errors. If you could help me with this, it would be greatly appreciated.

Here is the pseudocode for the idea:

// Pseudocode: Convert mouse absolute coordinates to relative coordinates
int x_prev = 0, y_prev = 0;

void process_mouse_data(int x_abs, int y_abs) {
    // Calculate the relative increments
    int dx = x_abs - x_prev;
    int dy = y_abs - y_prev;

    // Update the previous coordinates
    x_prev = x_abs;
    y_prev = y_abs;

    // Convert to PS/2 protocol format and send
    send_ps2_mouse_data(dx, dy);
}
Thank you in advance for your help!

No0ne, 2024-11-28:
Try this, replace:
```c
void ms_send_movement(u8 buttons, s8 x, s8 y, s8 z) {
  ms_db = buttons;
  ms_dx += x;
  ms_dy += y;
  ms_dz += z;
}
```

With this:
```c
s8 x_prev = 0;
s8 y_prev = 0;

void ms_send_movement(u8 buttons, s8 x, s8 y, s8 z) {
  ms_db = buttons;
  ms_dx += x - x_prev;
  ms_dy += y - y_prev;
  ms_dz += z;
  x_prev = x;
  y_prev = y;
}
```

zoushiyin, 2024-11-28:
This works, but it seems that the network speed is too slow, causing the cursor to occasionally move straight up and down or left and right. I'm not sure if there is a better way to verify this.

0x504B0304, 2024-11-28:
So，why not just use the relative mode？

No0ne, 2024-11-28:
Maybe the conversion to 8-bit signed cuts of the larger absolute values: https://github.com/No0ne/ps2x2pico/blob/main/src/usbin.c#L166

zoushiyin, 2024-11-29:
> So，why not just use the relative mode？

The "controlled end mouse" cannot be set to relative mode.

zoushiyin, 2024-11-30:
I still haven't managed to get it working, and I think I'm going to give up.

0x504B0304, 2024-11-30:
> I still haven't managed to get it working, and I think I'm going to give up.

留个联系方式，也许可以再讨论下

No0ne, 2024-11-30:
How much does this KVM cost? Can you ship one to Europe?

No0ne, 2025-01-09:
I'm closing this issue because it became inactive.

closed 2025-01-09 by No0ne

zoushiyin, 2025-03-01:
hi, @No0ne ,This is the HIDDUMP (Human Interface Device dump) of the OrayKVM Mouse Keyboard.
PS F:\ps2_usb\winhiddump> .\winhiddump.exe
17EF:6099: LiteOn - Lenovo Traditional USB Keyboard
PATH:\\?\hid#vid_17ef&pid_6099&mi_01&col02#7&3b84c28e&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  85  02  09  E2  15  00  25  01  35  00
  45  01  65  00  55  00  75  01  95  01  81  02  09  EA  81  02
  09  E9  81  02  05  09  09  11  81  02  05  0C  09  CD  81  02
  09  B7  81  02  09  B6  81  02  09  B5  81  02  05  09  09  26
  81  02  09  27  81  02  09  21  81  02  09  25  81  02  05  0C
  0A  94  01  81  02  0A  92  01  81  02  95  02  81  03  C1  00

  (96 bytes)
13BA:0018: (unnamed manufacturer) -
PATH:\\?\hid#vid_13ba&pid_0018&mi_01&col01#9&3a1961ef&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  85  01  09  01  A1  00  05  09  19  01
  29  05  15  00  25  01  35  00  45  01  65  00  55  00  75  01
  95  05  81  02  95  03  81  03  05  01  09  30  25  7F  45  00
  75  08  95  01  81  06  09  31  81  06  09  38  81  06  C1  00
  05  FF  09  02  15  00  25  01  45  01  75  01  B1  22  95  07
  B1  03  C1  00
  (84 bytes)
17EF:6099: LiteOn - Lenovo Traditional USB Keyboard
PATH:\\?\hid#vid_17ef&pid_6099&mi_00#7&2854a650&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  03  95  03  91  02  95  05  91  03
  C1  00
  (50 bytes)
13BA:0018: (unnamed manufacturer) -
PATH:\\?\hid#vid_13ba&pid_0018&mi_01&col02#9&3a1961ef&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  80  A1  01  85  02  19  81  29  83  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  03  81  06  95  05
  81  03  C1  00
  (36 bytes)
0483:572B: STMicroelectronics - OrayKVM Mouse Keyboard
PATH:\\?\hid#vid_0483&pid_572b&mi_01#7&392459f7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  85  02  09  01  A1  00  05  09  19  01
  29  03  15  00  25  01  35  00  45  01  65  00  55  00  75  01
  95  03  81  02  95  05  81  03  05  01  09  38  25  7F  45  00
  75  08  95  01  81  06  09  30  26  FF  7F  75  10  81  06  09
  31  81  06  C1  00  C1  00
  (71 bytes)
17EF:6099: LiteOn - Lenovo Traditional USB Keyboard
PATH:\\?\hid#vid_17ef&pid_6099&mi_01&col01#7&3b84c28e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  80  A1  01  85  01  19  81  29  83  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  03  81  02  95  05
  81  03  C1  00
  (36 bytes)
046D:C232: (unnamed manufacturer) - (unnamed product)
PATH:\\?\hid#vid_046d&pid_c232#2&357f35a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  05  95  05  91  02  95  03  91  03
  C1  00
  (50 bytes)
0483:572B: STMicroelectronics - OrayKVM Mouse Keyboard
PATH:\\?\hid#vid_0483&pid_572b&mi_00#7&154d1c35&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  85  01  09  01  A1  00  05  09  19  01
  29  03  15  00  25  01  35  00  45  01  65  00  55  00  75  01
  95  03  81  02  95  05  81  03  05  01  09  38  25  7F  45  00
  75  08  95  01  81  06  09  30  15  00  26  FF  7F  46  FF  7F
  75  10  81  02  09  31  81  02  C1  00  C1  00
  (76 bytes)
0483:572B: STMicroelectronics - OrayKVM Mouse Keyboard
PATH:\\?\hid#vid_0483&pid_572b&mi_02#7&2bced439&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  05  95  05  91  02  95  03  91  03
  C1  00
  (50 bytes)
093A:2510: PixArt - USB Optical Mouse
PATH:\\?\hid#vid_093a&pid_2510#7&1c167874&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  01  09  02  A1  01  09  01  A1  00  05  09  19  01  29  03
  15  00  25  01  35  00  45  01  65  00  55  00  75  01  95  03
  81  02  95  05  81  03  05  01  09  30  25  7F  45  00  75  08
  95  01  81  06  09  31  81  06  09  38  81  06  C1  00  C1  00

  (64 bytes)
13BA:0018: (unnamed manufacturer) -
PATH:\\?\hid#vid_13ba&pid_0018&mi_01&col03#9&3a1961ef&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  85  03  15  00  25  01  35  00  45  01
  65  00  55  00  75  01  95  10  81  03  C1  00
  (28 bytes)
13BA:0018: (unnamed manufacturer) -
PATH:\\?\hid#vid_13ba&pid_0018&mi_00#9&26e945b1&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  38
  81  03  05  08  19  01  29  03  95  03  91  02  95  05  91  03
  C1  00
  (50 bytes)