No0ne/ps2x2pico #63: Adding Common Mouse Shortcut Keys to the Tool

zoushiyin, 2024-11-24:
Hi,

We often use shortcut keys on common mice, such as copy, paste, cut, previous page, next page, etc. Would it be possible to add support for these shortcut keys to this tool?

zoushiyin, 2024-11-25:
I want to map the "Previous Page" and "Next Page" buttons to "Copy" and "Paste" actions. The ms_db values for these buttons are 0x10 and 0x08, respectively. Thanks.

No0ne, 2024-11-25:
I don't know what "Copy" and "Paste" actions are on mouse buttons. Do you mean the mouse button should send a keyboard key?

No0ne, 2024-11-25:
Regarding to https://download.microsoft.com/download/1/6/1/161ba512-40e2-4cc9-843a-923143f3456c/translate.pdf the keyboard copy/paste keys don't have PS/2 equivalents.

zoushiyin, 2024-11-25:
I have the following code to detect and handle mouse shortcuts:

// Detect and handle mouse shortcuts
if (buttons == 0x10) {  // Copy shortcut (Ctrl+C)
    printf("Copy (Ctrl+C) triggered\n");
    // Simulate pressing the Ctrl key
    kb_send_key(0x14, true, 0);
    // Simulate pressing the 'C' key
    kb_send_key(0x21, true, 0);
    // Simulate releasing the 'C' key
    kb_send_key(0x21, false, 0);
    // Simulate releasing the Ctrl key
    kb_send_key(0x14, false, 0);
} else if (buttons == 0x08) {  // Paste shortcut (Ctrl+V)
    printf("Paste (Ctrl+V) triggered\n");
    // Simulate pressing the Ctrl key
    kb_send_key(0x14, true, 0);
    // Simulate pressing the 'V' key
    kb_send_key(0x2a, true, 0);
    // Simulate releasing the 'V' key
    kb_send_key(0x2a, false, 0);
    // Simulate releasing the Ctrl key
    kb_send_key(0x14, false, 0);
}
My code is as follows, but the serial output shows this, and it does not execute the correct commands:

Paste (Ctrl+V) triggered 
kb > host 15
kb > host 66
kb > host f0
kb > host 66
kb > host f0
kb > host 15
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x08
Paste (Ctrl+V) triggered
kb > host 15
kb > host 66
kb > host f0
kb > host 66
kb > host f0
kb > host 15
Why is it not executing the correct commands?

No0ne, 2024-11-25:
Try this:
```C
kb_send_key(0xe0, true, 0);
kb_send_key(0x06, true, 0);
kb_send_key(0x06, false, 0);
kb_send_key(0xe0, false, 0);
```
```C
kb_send_key(0xe0, true, 0);
kb_send_key(0x19, true, 0);
kb_send_key(0x19, false, 0);
kb_send_key(0xe0, false, 0);
```

No0ne, 2024-11-25:
Try this:
```C
kb_send_key(KEYBOARD_MODIFIER_LEFTCTRL, true, 0);
kb_send_key(HID_KEY_C, true, 0);
kb_send_key(HID_KEY_C, false, 0);
kb_send_key(KEYBOARD_MODIFIER_LEFTCTRL, false, 0);
```
```C
kb_send_key(KEYBOARD_MODIFIER_LEFTCTRL, true, 0);
kb_send_key(HID_KEY_V, true, 0);
kb_send_key(HID_KEY_V, false, 0);
kb_send_key(KEYBOARD_MODIFIER_LEFTCTRL, false, 0);
```

zoushiyin, 2024-11-26:
The keys are being sent multiple times each time they are triggered. I'm not sure where to control the number of times they are sent. Thank you for your help.
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x08
Paste (Ctrl+V) triggered
kb > host 14
kb > host 2a
kb > host f0
kb > host 2a
kb > host f0
kb > host 14
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x08
Paste (Ctrl+V) triggered
kb > host 14
kb > host 2a
kb > host f0
kb > host 2a
kb > host f0
kb > host 14
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x08
Paste (Ctrl+V) triggered
kb > host 14
kb > host 2a
kb > host f0
kb > host 2a
kb > host f0
kb > host 14
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x08
Paste (Ctrl+V) triggered
kb > host 14
kb > host 2a
kb > host f0
kb > host 2a
kb > host f0
kb > host 14
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x08
Paste (Ctrl+V) triggered
kb > host 14
kb > host 2a
kb > host f0
kb > host 2a
kb > host f0
kb > host 14
Mouse Move: X = 0, Y = 0, Z = 0, Buttons = 0x00

No0ne, 2025-01-09:
I'm closing this issue because it became inactive.

closed 2025-01-09 by No0ne