No0ne/ps2x2pico #65: Built-in software USB Hub on PIO SM

javavi, 2024-12-01:
When using the ps2x2pico adapter to simultaneously connect a USB keyboard and mouse, you have to connect them via an external USB hub, which creates additional inconveniences and not all of them are suitable.
Perhaps a good solution would be to use the Pico-PIO-USB library for software implementation of two separate USB-A ports for separate connection of a USB keyboard and USB mouse?
At the same time, the hardware USB port of Pico will remain free for other applications... for example, connecting gamepads with mapping to keyboard buttons.
[https://github.com/sekigon-gonnoc/Pico-PIO-USB](url)

No0ne, 2024-12-01:
Hi, I'm actually using this already for the ITX-Llama integration: https://github.com/eivindbohler/itxllama/pull/2
Here a patch for TinyUSB is needed to activate both native and PIO USB ports.

I think this would be most suitable with a custom PCB discussed here: https://github.com/No0ne/ps2x2pico/issues/57

javavi, 2024-12-01:
Isn't it possible to run both PS/2 on one PIO, and USB on another PIO?

No0ne, 2024-12-02:
Yes thats what I'm doing on the ITX-Llama. There is no PS/2 input/passthru on the ITX-Llama, but the normal ps2x2pico has that feature enabled.

No0ne, 2024-12-03:
Yes, its already working, one PIO is used for output and one for input. If the input is disabled it can be used for USB as seen on the ITX-Llama.

Do you want to try this out?

javavi, 2024-12-03:
> Do you want to try this out?

Yes, it would be interesting, where can I see the schematic and firmware?

javavi, 2024-12-03:
> I don't have any schematic as I don't have any KiCAD skills, but we can agree on using GPIO16/17 and GPIO18/19 as D+/D- and I'll send you a testing firmware build.

OK! I'll try to assemble a prototype of such an adapter device and let you know... and also, will it be possible to use arbitrary pairs of ports for USB interfaces, for example GP0, GP1 and GP2, GP3?

No0ne, 2024-12-03:
Correct, any GPIO pairs are working (without space between D+/D- and none of the PS/2 pins ;)

javavi, 2024-12-03:
[image]
I plan to do it according to this scheme

No0ne, 2024-12-03:
Try this: [pio-usb-test.zip](https://github.com/user-attachments/files/17998787/pio-usb-test.zip)

javavi, 2024-12-04:
Today, I quickly assembled this circuit, flashed your firmware... And yes! It works!
[image]

javavi, 2024-12-05:
Just a question out of curiosity:
Is it possible to do this so that in addition to the ports of the software USB Hub, the built-in USB in Pico also works?

javavi, 2024-12-05:
I also have an adapter made for standard firmware, but I can solder two USB ports to it.
Could you rebuild your software for this connection scheme?
[image]
* It would be good to have a port configurator for freedom of schematic creativity.

No0ne, 2024-12-05:
Now all three ports should work: [pio+native-usb-test.zip](https://github.com/user-attachments/files/18022122/pio%2Bnative-usb-test.zip)

hax0rbana-adam, 2024-12-05:
:wave: I'd be happy to put together a KiCAD schematic and PCB for this. I have a few questions first:

1. Can we slap two PS/2 ports on there in parallel for people who don't have a combo PS/2 port? (just wire all 6 pins to both ports and expect that people will not do something weird like connect them to two computers at once)
2. Do you have a part number for the ZD 3V3 so I don't mess it up?
3. Are those really just 22 ohm resistors?
4. For the PCB, do we want to stick with the Raspberry Pi Pico or Pico 2, or switch to the Waveshare RP2040-zero?

Personally, I like the smaller form factor of the Waveshare and the built-in USB-C port. I'm not super keen on making boards for both the rPi & Waveshare footprints since I'd then feel obligated to build, test and stock them for people who want a ps2x2pico without any soldering.

Oh, and I guess maybe we should move this conversation over to #57 since it's about the hardware and this ticket is about the software?

ssokolow, 2024-12-05:
> Can we slap two PS/2 ports on there in parallel for people who don't have a combo PS/2 port? (just wire all 6 pins to both ports and expect that people will not do something weird like connect them to two computers at once)

I'd second this. Properly wired PS/2 combo-port breakout cables are surprisingly difficult to find cheaply in Canada so, for the ps2x2pico I built, I just bought a dirt-cheap PS/2 male-to-male cable (Seriously. Why?) from China, cut it in half, and made some little printer-paper-and-scotch/sellotape flags to say which one to plug into the keyboard port and which one to plug into the mouse port.

> For the PCB, do we want to stick with the Raspberry Pi Pico or Pico 2, or switch to the Waveshare RP2040-zero?

As a data point, I like the USB-C port on my YD-RP2040, but a big part of the reason I chose it was that it was the most inexpensive thing I could find. I didn't want to be paying more for the PS/2-to-USB adapter than I paid for the thin client I was repurposing into a fanless retro PC.

zoushiyin, 2024-12-09:
> Now all three ports should work: [pio+native-usb-test.zip](https://github.com/user-attachments/files/18022122/pio%2Bnative-usb-test.zip)

Which version of the source code does this file correspond to? I would like to make a version. Is that possible?

No0ne, 2024-12-09:
Its currently in a separate branch: https://github.com/No0ne/ps2x2pico/tree/65-built-in-software-usb-hub-on-pio-sm

javavi, 2024-12-09:
Look, people are already making adapter boards using a simplified scheme. :))
[image]
[image]
[image]
[image]
https://github.com/xtremespb/usb2ps2-zero

zoushiyin, 2025-01-04:
> Its currently in a separate branch: https://github.com/No0ne/ps2x2pico/tree/65-built-in-software-usb-hub-on-pio-sm
Hello, @No0ne 

I hope this message finds you well. I have followed the schematic from [this comment](https://github.com/No0ne/ps2x2pico/issues/65#issuecomment-2519971280) to design my PCB for the Pico Zero version. After flashing the firmware, I noticed that there is no response from the inputs and outputs. Additionally, I am unsure where to set the GPIO pins for the USB port, and the LED configuration has not been updated for the Zero version, so the LEDs do not light up.

However, when I flash the latest code, the USB host input for keyboard and mouse works correctly, and the PS/2 output also functions as expected.

javavi, 2025-07-16:
@No0ne
Everything worked out!
It was just necessary to apply a patch to the library TinyUSB 0.17.0