No0ne/ps2x2pico #66: Is ps2x2pico Compatible with DIY Keyboards (Such as Those Using QMK Firmware)?

zoushiyin, 2024-12-07:
Hi @No0ne,

I would like to confirm if ps2x2pico is compatible with DIY keyboards (such as those using QMK firmware).
Thank you for your assistance!

No0ne, 2024-12-07:
Just plug it in and test I guess?

nvllsvm, 2024-12-08:
Works fine on my various qmk boards.

No0ne, 2024-12-08:
Does QMK support NKRO? Does it work?

nvllsvm, 2024-12-08:
I'm glad I tested this when I did because there appears to be a regression in ps2x2pico v2.0. I'm using a Keychron S1 running vanilla qmk with a custom layout that allows for toggling NKRO with a hotkey. 
- in v1.3, both NKRO enabled and disabled works albeit with the PC only ever seeing a maximum of 6 keys
- in v2.0, enabling NKRO mode results in no input to the PC at all

No0ne, 2024-12-09:
Thanks for testing, I'll look into it!

No0ne, 2024-12-09:
I tried a spare Teensy and mapped the NKRO toggle:  [image]

But it doesn't work connected directly to my Mac. I always stops send keys after the 6th. @nvllsvm Did I forget something?

nvllsvm, 2024-12-09:
I verified that my keyboard does report multiple keys on a mac. Here's how I have NKRO configured on my board:
- NKRO is enabled by default each time the keyboard is powered on https://github.com/nvllsvm/dotfiles/blob/master/qmk_firmware/keychron/s1/ansi/rgb/keymaps/nvllsvm/config.h#L1
- some other configuration  https://github.com/nvllsvm/dotfiles/blob/master/qmk_firmware/keychron/s1/ansi/rgb/keymaps/nvllsvm/rules.mk#L9

On Linux, my keyboard reports three separate input devices. Using `evtest`, I noticed that different devices are used depending on whether NKRO is enabled. It appears that the first keyboard USB interface is NKRO-disabled and the second interface is NKRO-enabled. 
| Device | Path |  Notes |
| ---- | --- | --- | 
| `/dev/input/by-id/usb-Keychron_Keychron_S1-event-kbd` | `pci-0000:0d:00.3-usbv2-0:4.1.1.3:1.0-event-kbd` | reports when NKRO is disabled |
| `/dev/input/by-id/usb-Keychron_Keychron_S1-if01-event-kbd` | `pci-0000:0d:00.3-usbv2-0:4.1.1.3:1.1-event-kbd` | reports when NKRO is enabled |
| `/dev/input/by-id/usb-Keychron_Keychron_S1-event-if01` | | not reporting in either NKRO mode |

nvllsvm, 2024-12-12:
Output from GPIO0
```
ps2x2pico-2.0
Setting defaults for keyboard
scancodeset set to 2
kb > host aa
ms > host aa
ms > host 00
Blinking keyboard LEDs

HID(1,0,KEYBOARD) mounted
 ID: 3434:0410
 HID has 1 reports
 HID(1,0,KEYBOARD) registered for reports

HID(1,1,NONE) mounted
 ID: 3434:0410
 HID has 3 reports
host > kb f2
KBHOSTCMD_READ_ID_F2
kb > host fa
kb > host ab
kb > host 83
host > kb f2
KBHOSTCMD_READ_ID_F2
kb > host fa
kb > host ab
kb > host 83
host > kb ff
KBHOSTCMD_RESET_FF
Setting defaults for keyboard
scancodeset set to 2
kb > host fa
kb > host aa
Blinking keyboard LEDs
host > ms ff
ms > host fa
ms > host aa
ms > host 00
host > kb f2
KBHOSTCMD_READ_ID_F2
kb > host fa
kb > host ab
kb > host 83
host > kb ed
KBHOSTCMD_SET_LEDS_ED
kb > host fa
host > kb 00
kb > host fa
host > kb ed
KBHOSTCMD_SET_LEDS_ED
kb > host fa
host > kb 02
kb > host fa
kb > host e0
kb > host 72
kb > host e0
kb > host f0
kb > host 72
HID(1,0) unmounted
HID(1,1) unmounted
```

No0ne, 2024-12-12:
Thanks for the hint with the QMK configuration. The ez_maker/directpins config has NKRO completely disabled, I used another keyboard with a teensy and NKRO enabled and it worked.

QMK does send NKRO a little different than the 8BitDo that I initially tested with.
Please try this testbuilt: [nkro-fix.zip](https://github.com/user-attachments/files/18109965/nkro-fix.zip)

nvllsvm, 2024-12-12:
> QMK does send NKRO a little different than the 8BitDo that I initially tested with. Please try this testbuilt: [nkro-fix.zip](https://github.com/user-attachments/files/18109965/nkro-fix.zip)

That build fixes input in BIOS and Windows 98. However, neither mouse nor keyboard function in Windows XP after a fresh boot. If I boot into Windows 98 and then reboot into Windows XP, both mouse and keyboard work. NKRO also functions as expected (saw over 14 concurrent keypresses).

I tested with both your build and a build configured with `-DMS_RATE_DEFAULT=100 -DMS_RATE_HOST_CONTROL=OFF`.

- [boot_to_winxp.log](https://github.com/user-attachments/files/18113977/boot_to_winxp.log)
- [boot_to_win98_then_reboot_winxp.log](https://github.com/user-attachments/files/18113975/boot_to_win98_then_reboot_winxp.log)

No0ne, 2024-12-12:
Does this bug also exist in 2.0 and/or 1.2?

No0ne, 2024-12-12:
Also please try compiling with `pico_enable_stdio_uart(ps2x2pico 0)` set to 0 ind CMakeLists.txt

nvllsvm, 2024-12-12:
I was wrong as I found that Windows XP will sometimes detect input. This is regardless of any version+configuration of ps2x2pico I used. It seemed to be most consistent when I would press a keyboard key before reaching my boot menu.

I'm thinking that the issue lies elsewhere in my setup.  
- kb+m -> Extron USB switcher -> external hub to inject power -> Retrocution Tiny USB hub -> ps2x2pico -> Abit AV8 motherboard

I removed the external hub and Tiny USB hub and it is so far working flawlessly. The system also boots slightly faster as well 🤷. Anyway, thanks for fixing QMK+NKRO support!

---

> Also please try compiling with `pico_enable_stdio_uart(ps2x2pico 0)` set to 0 in CMakeLists.txt (there will be no serial output)

Curious - why did you suggest this; were you thinking there was a timing issue?

No0ne, 2024-12-13:
I discovered that too much serial output crashes something in the Pico and the PIOs stop outputting PS/2 bytes. I don't why this is but its a lot worse on SDK 2.0 and 2.1. Thats why I stick to 1.5.1 at the moment.

No0ne, 2024-12-13:
Also TinyUSB seems more buggy regarding replugging devices.

closed 2024-12-13 by No0ne