No0ne/ps2x2pico #67: Idea: Switch Output Between Two PCs (PS/2 and USB) Using Hotkeys or Automatic Cursor Detection

zoushiyin, 2024-12-07:
The proposed features are:

Hotkey-based switching: Use a hotkey combination to toggle the output between the two PCs.
Automatic cursor detection: Automatically switch the output when the mouse cursor reaches the edge of one PC's screen, similar to how multi-monitor setups work.
This could be particularly useful in dual-computer setups where users might want to switch control between a legacy system and a modern system seamlessly.

Thank you very much for your time and feedback. I look forward to hearing your thoughts!

No0ne, 2024-12-07:
This sounds like: https://github.com/hrvach/deskhop

I think monitor edge detection won't be possible on PS/2.

zoushiyin, 2024-12-09:
Yes, it's similar to this. However, in this idea, the switching is done using defined hotkeys, and an additional pair of IO pins is used for USB output.

0x504B0304, 2024-12-09:
It's similar to this. However, in this idea, the switching is done using defined hotkeys, and an additional pair of IO pins is used for USB output.

If use hot keys, pico must be able to repeat the usb signal to another usb pins, however there's no more pio in rp2040. it's conflict with ps/2 pass through or "internal hub".

zoushiyin, 2024-12-09:
Could the switching be done before the PS/2 processing takes place? Ideally, I would like a single Pico to handle time-shared output for both data streams.

zoushiyin, 2024-12-20:
> This sounds like: https://github.com/hrvach/deskhop
> 
> I think monitor edge detection won't be possible on PS/2.

I have connected the PS/2 interface using a ps2x2pico adapter to a DeskHop device, but I am not getting any response from it. The DeskHop device uses absolute coordinates for mouse control. Additionally, the keyboard is also not responding.