No0ne/ps2x2pico #68: Compilation issues with pico-sdk 1.5.1 when used with latest code

zoushiyin, 2024-12-12:
Is it necessary to upgrade to Pico-SDK version 2.0 to resolve these compilation issues? If so, are there any specific steps or considerations I should be aware of when upgrading?

Below are the exact error messages I receive during compilation:

Thank you for your help!
[ 22%] Performing build step for 'ELF2UF2Build'
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[ 23%] No install step for 'ELF2UF2Build'
[ 24%] Completed 'ELF2UF2Build'
[ 24%] Built target ELF2UF2Build
[ 25%] Building C object CMakeFiles/ps2x2pico.dir/src/ps2x2pico.c.obj
[ 26%] Building C object CMakeFiles/ps2x2pico.dir/src/usbin.c.obj
/home/zoushiyin/pico/ps2x2pico/src/usbin.c: In function 'hid_parse_report_descriptor':
/home/zoushiyin/pico/ps2x2pico/src/usbin.c:225:13: error: a label can only be part of a statement and a declaration is not a statement
             u16 offset = (info->num_items == 0) ? 0 : (info->item[info->num_items - 1].bit_offset + info->item[info->num_items - 1].bit_size);
             ^~~
CMakeFiles/ps2x2pico.dir/build.make:89: recipe for target 'CMakeFiles/ps2x2pico.dir/src/usbin.c.obj' failed
make[2]: *** [CMakeFiles/ps2x2pico.dir/src/usbin.c.obj] Error 1
CMakeFiles/Makefile2:1512: recipe for target 'CMakeFiles/ps2x2pico.dir/all' failed
make[1]: *** [CMakeFiles/ps2x2pico.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

zoushiyin, 2024-12-12:
I have moved the declaration of offset to the top of the function as follows:

_u8 ri_report_usage_count = 0;
u8 ri_collection_depth = 0;
u16 offset; // Move the declaration of offset here_

And within the switch-case structure, I replaced the inline declaration with an assignment:

_switch(type) {
  case RI_TYPE_MAIN:
    switch(tag) {
      case RI_MAIN_INPUT:
      case RI_MAIN_OUTPUT:
      case RI_MAIN_FEATURE:
        // u16 offset = (info->num_items == 0) ? 0 : (info->item[info->num_items - 1].bit_offset + info->item[info->num_items - 1].bit_size);
        offset = (info->num_items == 0) ? 0 : (info->item[info->num_items - 1].bit_offset + info->item[info->num_items - 1].bit_size);_
        ...
Is this modification correct?

Thank you!

No0ne, 2024-12-12:
Currently SDK 2.0 and 2.1 have strange bugs but it should compile normally with all versions.

How did you install your ARM compiler?

Chandler-Kluser, 2024-12-12:
Exatcly same thing have just happened to me here, my gcc is:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Built from https://archlinux.org/packages/extra/x86_64/arm-none-eabi-gcc/ (I use arch, btw)

I have built with older versions, but the latest one really gives that error

No0ne, 2024-12-12:
This line is origibally from https://github.com/pdaxrom/usb2n64-adapter/blob/master/hid_parser.c#L71

There was an additional statement, but I wonder why I never got this error 🤔.
I use the homebrew arm compiler on macOS.

No0ne, 2024-12-14:
I did finally reproduce it with the Pico SDK for Windows, this should fix it: 00067cb67a44facbedb0a90e7388502f83479a0e

zoushiyin, 2024-12-14:
When comparing the current version to version 1.3, I have observed several instability issues:
• Random GUI Window Switching: The graphical user interface windows occasionally switch unexpectedly.
• Slower Device Recognition: It feels like devices take longer to be recognized by the system.
• Multiple Mouse Connections Not Recognized: When multiple mice are connected, they are not being recognized properly.
The specific reasons for these issues are unclear, and they are also difficult to reproduce consistently. Any insights or suggestions on how to diagnose and resolve these problems would be greatly appreciated.
Thank you for your attention and assistance!

Chandler-Kluser, 2024-12-14:
I still got the same issue:

```
$ cmake ..         
PICO_SDK_PATH is /home/chandler/pico-sdk
Target board (PICO_BOARD) is 'pico'.
Using board configuration from /home/chandler/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
CMake Error at CMakeLists.txt:15 (message):
  Raspberry Pi Pico SDK version 1.5.1 required.  Your version is 2.0.0

-- Configuring incomplete, errors occurred!
```

Additional info about my compiler:

```
$ which arm-none-eabi-gcc
/usr/bin/arm-none-eabi-gcc
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

No0ne, 2024-12-15:
What @zoushiyin described where symptoms I also saw with SDK 2.0 and 2.1, thats why I currently stick to 1.5.1.
@Chandler-Kluser your error messages says: `Raspberry Pi Pico SDK version 1.5.1 required.  Your version is 2.0.0`

No0ne, 2024-12-15:
Please install SDK 1.5.1 and TinyUSB 0.17.0 according to https://github.com/No0ne/ps2x2pico?tab=readme-ov-file#build

Chandler-Kluser, 2024-12-15:
sorry, I not paid attention to that 😅

zoushiyin, 2024-12-15:
I am working on integrating UART1 communication into the ps2x2pico project and need some help. Specifically, I want to enable UART1 using GPIO8 (TX) and GPIO9 (RX) to receive data from another development board. However, I have not been able to establish successful communication so far.

Could someone provide a code snippet or guidance on how to properly set up UART1 within the ps2x2pico framework? A simple example demonstrating the initialization and handling of incoming data would be extremely helpful.

Here is what I have tried so far:

#include "hardware/uart.h"
#include "hardware/irq.h"

#define UART_ID uart1
#define BAUD_RATE 9600
#define UART_TX_PIN 8 // UART TX 
#define UART_RX_PIN 9 // UART RX 

void init_uart1() {
    uart_init(UART_ID, BAUD_RATE);
    gpio_set_function(UART_TX_PIN, GPIO_FUNC_UART);
    gpio_set_function(UART_RX_PIN, GPIO_FUNC_UART);

    uart_set_irq_enables(UART_ID, true, false);
    irq_set_exclusive_handler(UART1_IRQ, on_uart_rx);
    irq_set_enabled(UART1_IRQ, true);
}

void on_uart_rx() {
    while (uart_is_readable(UART_ID)) {
        char c = uart_getc(UART_ID);
        printf("Received: %c\n", c); // 
    }
}
Despite this setup, I am still encountering issues. Any insights or corrections would be greatly appreciated!

Thank you very much for your help.

Best regards,

No0ne, 2024-12-15:
You need to disable FIFOs as seen in the example: https://github.com/raspberrypi/pico-examples/blob/master/uart/uart_advanced/uart_advanced.c
```
uart_set_fifo_enabled(UART_ID, false);
```