No0ne/ps2x2pico #70: Mouse misbehaving on old laptop when connected during power up

depau, 2024-12-22:
Hi, thanks for sharing this project, it's really useful!

I tried it to connect a modern mouse to my old Toshiba Satellite 310CDS running Windows 98.

I noticed the mouse works fine but only if I reset the microcontroller (or reconnect the PS/2 port). If I leave it connected during power-up it moves incorrectly and it randomly clicks around.

I tried flashing the 100Hz build (haven't tried the 200Hz one) but the behavior is the same.

Do you have any suggestions for troubleshooting? Thanks!

No0ne, 2024-12-22:
Hi, do you have a serial cable? Can you post the debug output from GPIO0?

depau, 2024-12-22:
Hi, yes!

You should be able to see where I reset the pico; after that it started working correctly.

```
ps2x2pico-2.1
Setting defaults for keyboard
scancodeset set to 2
kb > host aa
ms > host aa
ms > host 00
Blinking keyboard LEDs

HID(1,0,keyboard) mounted
 VID: 046d  PID: c52b
 HID has 1 reports
 HID(1,0,keyboard) registered for reports

HID(1,1,mouse) mounted
 VID: 046d  PID: c52b
 HID has 4 reports
 HID(1,1,mouse) registered for reports

HID(1,2,none) mounted
 VID: 046d  PID: c52b
 HID has 3 reports
 HID(1,2,none) registered for reports
host > ms ff
ms > host fa
ms > host aa
ms > host 00
host > ms f2
ms > host fa
ms > host 00
host > ms ff
ms > host fa
ms > host aa
ms > host 00
keyboard unknown  usage_page: ff00  usage: 02
keyboard unknown  usage_page: ff00  usage: 02
host > ms ff
ms > host fa
ms > host aa
ms > host 00
host > ms ff
ms > host fa
ms > host aa
ms > host 00
host > ms f3
ms > host fa
host > ms c8
ms > host fa
host > ms f3
ms > host fa
host > ms 64
ms > host fa
host > ms f3
ms > host fa
host > ms 50
ms > host fa
host > ms f3
ms > host fa
host > ms 0a
ms > host fa
host > ms e8
ms > host fa
host > ms 03
ms > host fa
host > ms e6
ms > host fa
host > ms f3
ms > host fa
host > ms 28
ms > host fa
host > ms f4
�
ps2x2pico-2.1
Setting defaults for keyboard
scancodeset set to 2
kb > host aa
ms > host aa
ms > host 00
host > ms f5
ms > host fa
host > ms f4
host > ms ff
ms > host fa
Blinking keyboard LEDs
ms > host aa
ms > host 00

HID(1,0,keyboard) mounted
 VID: 046d  PID: c52b
 HID has 1 reports
 HID(1,0,keyboard) registered for reports
host > ms ff
ms > host fa

HID(1,1,mouse) mounted
 VID: 046d  PID: c52b
 HID has 4 reports
 HID(1,1,mouse) registered for reports

HID(1,2,none) mounted
 VID: 046d  PID: c52b
 HID has 3 reports
 HID(1,2,none) registered for reports
ms > host aa
ms > host 00
host > ms f3
ms > host fa
host > ms c8
ms > host fa
host > ms f3
ms > host fa
host > ms 64
ms > host fa
host > ms f3
ms > host fa
host > ms 50
ms > host fa
host > ms ff
ms > host fa
ms > host aa
ms > host 00
host > ms ff
ms > host fa
ms > host aa
ms > host 00
host > ms f3
ms > host fa
host > ms 0a
ms > host fa
host > ms e8
ms > host fa
host > ms 03
ms > host fa
host > ms e6
ms > host fa
host > ms f3
ms > host fa
host > ms 28
ms > host fa
host > ms f4
```

No0ne, 2024-12-22:
Thanks! What OS and mouse driver are you running?

No0ne, 2024-12-23:
Thanks! I'll check that after christmas.