No0ne/ps2x2pico #73: SHIFT not working with version 2.0 and 2.1

fingerr, 2025-01-07:
Hi,

I have situation that SHIFT not working using v2.0 or v2.1, with v1.2 works fine, but with v1.2 sometimes when I type it happens that key is being locked and is repeated until I press any key (i.e. 'helpppppppppp').

My keyboard is 'x-kom Alu Keys Black' 
https://www.x-kom.pl/p/713562-klawiatura-bezprzewodowa-x-kom-alu-keys-black.html
and I'm using it with TerribleFire TF330 for Amiga CD32.

Anyone faced same situation too?

Thanks in advance,
Mariusz.

No0ne, 2025-01-08:
Hi, can you send me a HID dump of your keyboard using this:
https://github.com/todbot/win-hid-dump or
https://github.com/DIGImend/usbhid-dump

fingerr, 2025-01-08:
> Hi, can you send me a HID dump of your keyboard using this: https://github.com/todbot/win-hid-dump or https://github.com/DIGImend/usbhid-dump

Here you are, nokb.txt is keyboard disconnected and kb.txt is connected.
[kb.txt](https://github.com/user-attachments/files/18350031/kb.txt)
[nokb.txt](https://github.com/user-attachments/files/18350033/nokb.txt)

fingerr, 2025-01-09:
> Thanks, do the other modifiers work? (CTRL, ALT, WIN)

Unfortunately not, none of them work.

fingerr, 2025-01-09:
> Can you attach a serial usb cable to GPIO0 and post the debug output when pressing a modifier, thanks!

Sadly don't have it atm. Any examples of wiring USB->COM to PiPico for debug? And examples of such coverter?

fingerr, 2025-01-09:
> Can you attach a serial usb cable to GPIO0 and post the debug output when pressing a modifier, thanks!

Ok, got it working using RPI2.

Nothing happens on console when I press ALT, CTRL or SHIFT keys (for sure it shows pressing other keys)

`ps2x2pico-2.1
Setting defaults for keyboard
scancodeset set to 2
kb > host aa
ms > host aa
ms > host 00
Blinking keyboard LEDs

HID(1,0,keyboard) mounted
 VID: 25a7  PID: fa10
 HID has 1 reports
 HID(1,0,keyboard) registered for reports

HID(1,1,mouse) mounted
 VID: 25a7  PID: fa10
 HID has 4 reports
 HID(1,1,mouse) registered for reports
kb > host e0
kb > host 72
`

No0ne, 2025-01-09:
Thank you! I'll need to send you a version with more debug output.

No0ne, 2025-01-09:
Please post the same output using this: [moredebug.zip](https://github.com/user-attachments/files/18367294/moredebug.zip)

fingerr, 2025-01-09:
> Please post the same output using this: [moredebug.zip](https://github.com/user-attachments/files/18367294/moredebug.zip)

Here you go:

```
Welcome to minicom 2.8

OPTIONS: I18n
Port /dev/serial0, 20:52:43

Press CTRL-A Z for help on special keys

ps2x2pico-2.1
Setting defaults for keyboard
scancodeset set to 2
kb > host aa
ms > host aa
ms > host 00
Blinking keyboard LEDs

HID(1,0,keyboard) mounted
 VID: 25a7  PID: fa10
 HID has 1 reports
 HID(1,0,keyboard) registered for reports

HID(1,1,mouse) mounted
 VID: 25a7  PID: fa10
 HID has 4 reports
 HID(1,1,mouse) registered for reports

 02  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00
 01  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00
 08  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00
 04  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00
 40  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00
 80  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00
 10  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00
 20  00  00  00  00  00  00  00
 00  00  00  00  00  00  00  00
```

No0ne, 2025-01-10:
🤔 thats odd, should be working, please try again: [evenmoredebug.zip](https://github.com/user-attachments/files/18371974/evenmoredebug.zip)
Also post the log while pressing normal keys, thanks!

fingerr, 2025-01-10:
> 🤔 thats odd, should be working, please try again: [evenmoredebug.zip](https://github.com/user-attachments/files/18371974/evenmoredebug.zip) Also post the log while pressing normal keys, thanks!

Here you are: pressed keys 1 2 3 4 and  all modifiers 

```
HID(1,0,keyboard) mounted
 VID: 25a7  PID: fa10
 HID has 1 reports
 HID(1,0,keyboard) registered for reports
 00  00  28  00  00  00  00  00
mod 00
kb > host 5a

HID(1,1,mouse) mounted
 VID: 25a7  PID: fa10
 HID has 4 reports
 HID(1,1,mouse) registered for reports
 00  00  00  00  00  00  00  00
mod 00
kb > host f0
kb > host 5a
 00  00  28  00  00  00  00  00
mod 00
kb > host 5a
 00  00  00  00  00  00  00  00
mod 00
kb > host f0
kb > host 5a
 00  00  1e  00  00  00  00  00
mod 00
kb > host 16
 00  00  00  00  00  00  00  00
mod 00
kb > host f0
kb > host 16
 00  00  1f  00  00  00  00  00
mod 00
kb > host 1e
 00  00  00  00  00  00  00  00
mod 00
kb > host f0
kb > host 1e
 00  00  20  00  00  00  00  00
mod 00
kb > host 26
 00  00  00  00  00  00  00  00
mod 00
kb > host f0
kb > host 26
 00  00  21  00  00  00  00  00
mod 00
kb > host 25
 00  00  00  00  00  00  00  00
mod 00
kb > host f0
kb > host 25
 02  00  00  00  00  00  00  00
mod 00
 00  00  00  00  00  00  00  00
mod 00
 01  00  00  00  00  00  00  00
mod 00
 00  00  00  00  00  00  00  00
mod 00
 08  00  00  00  00  00  00  00
mod 00
 00  00  00  00  00  00  00  00
mod 00
 04  00  00  00  00  00  00  00
mod 00
 00  00  00  00  00  00  00  00
mod 00
 40  00  00  00  00  00  00  00
mod 00
 00  00  00  00  00  00  00  00
mod 00
 80  00  00  00  00  00  00  00
mod 00
 00  00  00  00  00  00  00  00
mod 00
 10  00  00  00  00  00  00  00
mod 00
 00  00  00  00  00  00  00  00
mod 00
 20  00  00  00  00  00  00  00
mod 00
 00  00  00  00  00  00  00  00
mod 00

```

No0ne, 2025-01-10:
Ah thats what I thought, theres a bug in the modifier find function. I hardcoded the first byte in the last debug version, do the modifiers work now?

fingerr, 2025-01-10:
> Ah no, they don't, theres also a bug with the NKRO stuff.

I have no idea what are you talking to me, but I trust you :)

And yes, modifiers are not working :(

No0ne, 2025-01-10:
Hm this keyboard confuses me 😵
Modifiers should work now: [shouldwork.zip](https://github.com/user-attachments/files/18373556/shouldwork.zip)
Also please send a debug log while holding down more than six keys at once (excl. modifiers).

fingerr, 2025-01-10:
> Hm this keyboard confuses me 😵 Modifiers should work now: [shouldwork.zip](https://github.com/user-attachments/files/18373556/shouldwork.zip) Also please send a debug log while holding down more than six keys at once (excl. modifiers).

Brilliant! It works like a charm right now! Thank you very much! 

Here comes debug log, 1 2 3 4 and all modifiers incl caps lock:

```
 00  00  1e  00  00  00  00  00
kb > host 16
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 16
 00  00  1f  00  00  00  00  00
kb > host 1e
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 1e
 00  00  20  00  00  00  00  00
kb > host 26
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 26
 00  00  21  00  00  00  00  00
kb > host 25
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 25
 00  00  39  00  00  00  00  00
kb > host 58
host > kb ed
KBHOSTCMD_SET_LEDS_ED
kb > host fa
host > kb 04
kb > host fa
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 58
 00  00  39  00  00  00  00  00
kb > host 58
host > kb ed
KBHOSTCMD_SET_LEDS_ED
kb > host fa
host > kb 00
kb > host fa
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 58
 02  00  00  00  00  00  00  00
kb > host 12
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 12
 01  00  00  00  00  00  00  00
kb > host 14
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 14
 08  00  00  00  00  00  00  00
kb > host e0
kb > host 1f
 00  00  00  00  00  00  00  00
kb > host e0
kb > host f0
kb > host 1f
 04  00  00  00  00  00  00  00
kb > host 11
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 11
 40  00  00  00  00  00  00  00
kb > host e0
kb > host 11
 00  00  00  00  00  00  00  00
kb > host e0
kb > host f0
kb > host 11
 00  00  65  00  00  00  00  00
kb > host e0
kb > host 2f
 00  00  00  00  00  00  00  00
kb > host e0
kb > host f0
kb > host 2f
 00  00  65  00  00  00  00  00
kb > host e0
kb > host 2f
 00  00  00  00  00  00  00  00
kb > host e0
kb > host f0
kb > host 2f

```

fingerr, 2025-01-10:
anad pressing and holding  10 keys
```

 00  00  04  00  00  00  00  00
kb > host 1c
 00  00  04  16  00  00  00  00
kb > host 1b
 00  00  16  04  07  00  00  00
kb > host 23
 00  00  07  16  04  09  00  00
kb > host 2b
 00  00  09  16  04  00  00  00
kb > host f0
kb > host 23
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
kb > host 2b
 00  00  00  00  00  00  00  00
kb > host f0
kb > host 2b
kb > host f0
kb > host 1b
kb > host f0
kb > host 1c

```

No0ne, 2025-01-10:
Ok thank you, it seems the keyboard doesn't have NKRO despite saying so in the HID descriptor?
But this fix will be in the 2.2 release, thank you for all the testing!

closed 2025-01-10 by No0ne

fingerr, 2025-01-10:
> Ok thank you, it seems the keyboard doesn't have NKRO despite saying so in the HID descriptor? But this fix will be in the 2.2 release, thank you for all the testing!

Thank you for help.

If you need more debugging, I'm here.

Cheers!

zoushiyin, 2025-02-13:
> Ok thank you, it seems the keyboard doesn't have NKRO despite saying so in the HID descriptor? But this fix will be in the 2.2 release, thank you for all the testing!

Hello, @No0ne 

This version can solve the issue of non-responsive function keys. Could you please let me know when the latest v2.2 will be released? Looking forward to your update. Thank you.