No0ne/ps2x2pico #75: Powering the Pico from external source

Enig02, 2025-01-10:
Really usefull project you have here - gonna have to make one of these :)

I know a similar question has been asked before, but I hope you can help with my additional questions.
If I want to power the Pico from an external source (ie. a USB charger):

A. should I power it through VSYS or VBUS (and GND obviously)?

B. Should I simply not connect 5V from HV -> VBUS or also not connect 5V from PS/2 -> HV?

And Thanks again for a cool project :)

No0ne, 2025-01-10:
You have to power it through VBUS, otherwise the micro USB port and attached keyboard/mouse/hub won't get powered.
If you use a powered hub you can use VSYS. This just powers the Pico and not the micro USB port.

You must leave HV->VBUS disconnected otherwise you would power your retro computers 5V rail from the USB charger.
So you only need to connect the PS/2 5V to HV of the levelshift and nothing more.

Hope this clears it up!

Enig02, 2025-01-10:
I think that should do it. Now to source the parts and give it a shot :)
Thank you!

closed 2025-01-10 by No0ne