No0ne/ps2x2pico #76: MCHOSE Keyboard: Function Keys (Ctrl, Shift, Alt, Win) Not Recognized

zoushiyin, 2025-01-22:
I currently have a MCHOSE keyboard that is experiencing issues with function keys. The Ctrl, Shift, Alt, and Win keys are not being recognized by the system. However, regular character keys are recognized correctly, although they also produce notifications.

kb > host 1a
kb > host f0
kb > host 1a
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
kb > host 21
kb > host f0
kb > host 21
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
kb > host 1b
kb > host f0
kb > host 1b
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
kb > host 23
kb > host f0
kb > host 23
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!
WARNING: Unmapped HID key 0x1 in set 2, ignoring it!

No0ne, 2025-01-22:
Please try this: https://github.com/No0ne/ps2x2pico/issues/73#issuecomment-2582142634

zoushiyin, 2025-01-22:
Thank you for help.

zoushiyin, 2025-02-17:
> Please try this: [#73 (comment)](https://github.com/No0ne/ps2x2pico/issues/73#issuecomment-2582142634)

Looking forward to your update. Thank you.

zoushiyin, 2025-03-01:
Hi @No0ne ,

Hello! I have been using the https://github.com/No0ne/ps2x2pico/issues/73#issuecomment-2582142634 version you provided, and it works very well with support for Ctrl, Alt, and Shift keys. Thank you for your excellent work!

However, I am interested in some of the new features and improvements available in version v2.2. Could you please share a preview build or the modified files from this version? My project requires some secondary development based on your code, and having access to the latest changes would be extremely helpful.

Specifically:

Is it possible to get a pre-release version or the specific files that have been updated in v2.2?
If not, could you provide any guidance or documentation on how to implement these changes myself?
Are there any particular instructions or considerations I should be aware of when using these updated files?
Your assistance and guidance would be greatly appreciated. Thank you very much for your help and support.

No0ne, 2025-03-01:
I'm sorry, I am sick at the moment. I commit when can get out of bed.

zoushiyin, 2025-03-02:
Hi @No0ne,

I hope you're feeling better soon. Please take all the time you need to rest and recover. Your health is the most important thing right now.

Wishing you a speedy recovery, and thank you for all your hard work on this project. We'll be here waiting for your return when you're ready.

Take care!

No0ne, 2025-03-04:
This is the diff which fixes your modifiers: https://github.com/No0ne/ps2x2pico/commit/0a14314dd0a740725ab040ae067ba1d42ac45867
Will be included in v2.2 when ready.

closed 2025-03-04 by No0ne

zoushiyin, 2025-03-08:
Hi @No0ne,

After the recent update, I noticed that the LED light is no longer functioning. Previously, it worked fine, but now it remains off regardless of the operation. However, when using the shouldwork.zip package, the LED functions correctly.