No0ne/ps2x2pico #77: Looking for help debugging

Reletiv, 2025-02-05:
Here i am building a usb to ps/2 converter because my HP KVM switch does not like my NanoKVM usb keyboard mouse. I am looking for help in debugging since i have no idea where to start

Currently trying to get it to work on an old HP elitedesk, does not work, neither does it on the kvm.

here are some images of my assembled prototype, maybe its hardware, maybe not:

[image]
[image]

i do get the following serial output with a hp keyboard attached:
`19:16:19.929 -> ps2x2pico-2.1
19:16:19.929 -> Setting defaults for keyboard
19:16:19.964 -> scancodeset set to 2
19:16:19.964 -> kb > host aa
19:16:19.964 -> ms > host aa
19:16:19.964 -> ms > host 00
19:16:20.056 -> Blinking keyboard LEDs
19:16:20.641 -> 
19:16:20.641 -> HID(1,0,keyboard) mounted
19:16:20.641 ->  VID: 0d62  PID: 913e
19:16:20.641 ->  HID has 1 reports
19:16:20.641 ->  HID(1,0,keyboard) registered for reports
19:16:20.641 -> 
19:16:20.641 -> HID(1,1,none) mounted
19:16:20.641 ->  VID: 0d62  PID: 913e
19:16:20.641 ->  HID has 2 reports
19:16:20.641 ->  HID(1,1,none) registered for reports
`
Also clicking the space button etc results in some output so the keyboard seems to be detected.
With the logic analyser i can also see some data coming out of the gpio pins
`19:16:25.299 -> kb > host f0
19:16:25.299 -> kb > host 11
19:16:25.345 -> kb > host f0
19:16:25.345 -> kb > host 29
19:16:25.527 -> kb > host 29
19:16:25.616 -> kb > host f0
19:16:25.616 -> kb > host 29
19:16:25.877 -> kb > host 29
19:16:26.286 -> kb > host f0
19:16:26.286 -> kb > host 29
19:16:26.597 -> kb > host 29
19:16:26.686 -> kb > host f0
19:16:26.686 -> kb > host 29
19:16:26.804 -> kb > host 29
`

No0ne, 2025-02-05:
I've not tested the Pico W, but I think its ok?

Can you post a screenshot of the logic analyzer?

Reletiv, 2025-02-07:
I bought a normal pico today, but i still have the same issues. I will record some signals this weekend