No0ne/ps2x2pico #78: Two bugs found. Delay and keyboard LEDs

yugokoral, 2025-02-06:
I just got pico from my coworker, to test it. Actually the device itself works pretty good. But i found two bugs.

First one: When I have Y usb splitter connected to pico (and USB keyboard to the splitter, so i have another usb port for mouse), the NUM, CAPS and SCROL lock LEDs dont work. Their intended function works, but led-s don't. I tried splitter with HIDman, and it works, so there is no problem in splitter.

Another problem is probably a delay problem. The PC wont boot, when pico is connected to my KVM switch as input device. It freezes before loading primary and secondary IDE ports. I think it's the same problem as in HIDman's case, which rasteri just solved a day ago. The pico responds to "led reset ??" command way too fast. Usually keyboards responds back with BAT code (0xAA) after cca 300ms. So rasteri made a delay of 500ms and it works in HIDmans case. Here is a link of discussion:
(https://www.vogons.org/viewtopic.php?t=89777&start=460)

No0ne, 2025-02-06:
I ordered the USB splitter to test it my self. The delay should be fixable.

Please give me some time as I'm busy at the moment.

yugokoral, 2025-02-10:
Take your time! I just have one more question... 
Have you been thinking of implementing RS232 port for a mouse too? It would be a nice package. I also found two similar projects on vogons forum and one is based on raspberry pico too. But it would be nice to implement it together with your ps2x2pico, because alone rs232 can not provide enough power to the board and you need additional power supply, which in your case, while using PS2 port of keyboard, is reachable and you don't need to supply any additional power :)).