No0ne/ps2x2pico #83: PS2IN Not Working After Latest UF2 Flash; Circuit Follows Diagram

zoushiyin, 2025-05-12:
Hi @No0ne,
I am experiencing an issue where the ps2in functionality is not working after flashing the latest .uf2 file onto my board. I have followed the wiring diagram provided, but the setup does not seem to work as expected.

[image]

No0ne, 2025-05-12:
Hi there, I'm trying to get back on track with fixing issues. You did solder a purple jack to the mouse port. Do you intend to use this with a keyboard or mouse?

zoushiyin, 2025-05-13:
Thank you for your response and for getting back to addressing the issues.

My goal is to have both PS/2 and USB keyboard/mouse inputs working simultaneously.

No0ne, 2025-05-13:
Please post the debug output from GPIO0, there you should see activity when you have keyboard and/or mouse connected.