No0ne/ps2x2pico #84: USB Keyboard and Mouse not working after cold boot on ITX-Llama

flewid, 2025-05-24:
Hello, 

I've been playing with the ITX Llama and noticed an issue where if I have the computer off, then turn it on, the mouse and keyboard do not work until I unplug/replug them then works fine. Using the HID ports. 

This seems to continue working over power off, and, resets no problem, but if I leave the computer for any period of time and turn it back on the same issue occurs. 

I've also ensured the firmware is the latest, which I found from the Discord [linked here ](https://github.com/eivindbohler/itxllama/blob/main/rp2040/FLASH.md)

Thanks!

No0ne, 2025-05-24:
Hi, I'm in the process of upgrading to the latest Pico-SDK and TinyUSB library. This sounds like a library issue, I'll come back to you when its done.

No0ne, 2025-07-27:
I've updated to the latest pico-sdk, tinyusb and pio-usb version!
If you want you can test this: https://github.com/No0ne/itxllama/tree/main/rp2040/build