No0ne/ps2x2pico #87: Pico 2 support

gmiranda, 2025-08-01:
Hi,

Do you plan to support the Pico 2?

I see that you've tried to support pico sdk 2.1 but found issues when re-plugging the USB device.

I've built against SDK 2.2 and the pico2 (with `cmake .. -DPICO_BOARD=pico2`) and both mouse and keyboard work. Re-plugging the USB receiver works as well.  That said, I haven't thoroughly tested it.

The only issue I can see is that the LED is always on.

gmiranda, 2025-08-03:
That's great news! I hope resoldering is not required.

Would it be useful for you if I switch to the pre-3.0 branch?

No0ne, 2025-08-03:
No, resoldering should not be required.
The pre-3.0 branch is a complete mess right now as I do a restructure and import code from ps2pico.
I'll get back to you when it compiles/works again.